pavel-demin/red-pitaya-notes #908: Starting with SPI connection with Red-Pitaya (Alpine linux)

uiuatiq, 2019-12-03:
### Description of the setup:

 * Device: Red-Pitaya
 * SD card image: Alpine Linux for Red-Pitaya sd card, Debian 9 on VM
 * Application: SPI and wideband SDR for I-Q baseband.
 * Other relevant information:

### Description of the problem: Hello Guys, I have been already using Pavel Demin's pre-built applications for wideband SDR for generating my desired baseband signal by modifying Gnu Radio blocks. As, the Red-pitaya has SPI interference, I want to use this communication pin for my other board(as a slave).  I see there is an example for SPI communication on Red-pitaya web https://redpitaya.readthedocs.io/en/latest/appsFeatures/examples/spi.html 
By following this, unfortunately, when I search for '/dev/spidev1.0', I could not find the folder, even on /sys/dev/ , no spi dev is found either. 

So, my question is: does Debian 9 and Alpine Linux (on sd card) have SPI enable? If it is how can I find that directory?
My goal is to write the register byte value via Red-Pitaya spi interference to the slave board to generate the desired frequency.  Also, I have no experience in SPI communication, that's why I may sound silly. How can I run the program(like c) on itself the Red-Pitaya?
 
Thank you in advance.

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-12-03:
I've just tested the latest version (20191120) of the SD card images based on Alpine Linux and Debian Linux and I can't reproduce your problem.

The `spidev` module is automatically loaded and the `/dev/spidev1.0` device is available on both systems:
- Alpine Linux
```
rp-f0xxxx:~# lsmod
Module                  Size  Used by    Not tainted
...
spidev                 16384  0
...

rp-f0xxxx:~# ls -l /dev/spidev1.0
crw-rw----    1 root     root      153,   0 Jan 27 12:31 /dev/spidev1.0
```
- Debian Linux
```
root@red-pitaya:~# lsmod spidev
Module                  Size  Used by
...
spidev                 16384  0
...

root@red-pitaya:~# ls -l /dev/spidev1.0
crw------- 1 root root 153, 0 Nov 20 23:35 /dev/spidev1.0
```

The SPI example from the Red Pitaya documentation runs without any error message:
```
rp-f0xxxx:~# apk add gcc linux-headers
...

rp-f0xxxx:~# gcc spi.c -o spi

rp-f0xxxx:~# ./spi
Buffer: 255
Buffer: 255
Buffer: 255
```

pavel-demin, 2019-12-03:
I've just noticed that the host name on the first screenshot is `debian`. If I'm not mistaken, this host name corresponds to the development virtual machine running under Debian Linux.

Since the SPI port is located on the Red Pitaya board and not on the development virtual machine, the SPI code must run on the Red Pitaya board.

uiuatiq, 2019-12-04:
Dear Mr. Pavel,
First of all, thank you so much for your kind reply. I was taking the time to look after the problem. 
Yes, SPI program should run in Red-pitaya but I was just wondering to check the similar availability to Debian Linux, also still on Debian Linux VM could not have or find the directory. On the other hand, the 'spidev1.0' is certainly available in the Alpine Linux sd card image on Red-Pitaya. I attach the screenshot and thanks to you to guide on that and now I should able to write and compile code. However, to write the code and run it on Red-Pitaya, I am facing a problem. The compile c program file like 'gcc' is failed (screenshot) to add on the system also the same problem for sudo,abuild,make, patch, etc. I tried to solve the issue by google search but nothing worked. What should I do for getting it error-free??
[image]

[image]

N.B: I am using the Wideband SDR project notes to generate I-Q baseband signal, so is it possible to use SPI and wideband SDR parallelly on the Alpine Linux on Red-Pitaya??

Thank you again for your continues support.

pavel-demin, 2019-12-04:
To install the `linux-headers` package, the Red Pitaya board must have Internet access.

pavel-demin, 2019-12-04:
> I was just wondering to check the similar availability to Debian Linux, also still on Debian Linux VM could not have or find the directory.

To have the  `/dev/spidev1.0` device, Debian Linux must run on the Red Pitaya board. A link to a SD card image based on Debian Linux can be found in the [SD card image](http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#sd-card-image) section of the LED blinker notes.

uiuatiq, 2019-12-04:
Dear Mr. Pavel, 
I cannot say enough but thanks for your quick reply.
 My Laptop is wifi connected, then the ethernet cable connected from the Laptop to the Red-Pitaya board. I did before statically routed on the ethernet. I also have access to the Red-Pitaya board by rp.xxxx/ command and the LED blinker project is working. Is not that saying successfully network connection to Red-Pitaya?
[image]
[image]

N.B: I am using the Wideband SDR project notes to generate I-Q baseband signal, so is it possible to use SPI and wideband SDR parallelly on the Alpine Linux on Red-Pitaya??

Thank you so much.

pavel-demin, 2019-12-04:
> My Laptop is wifi connected, then the ethernet cable connected from the Laptop to the Red-Pitaya board.

If your laptop has access to Internet via Wi-Fi, then this internet connection should be shared with the Ethernet port. Look for the [windows internet connection sharing](https://www.google.com/search?q=windows+internet+connection+sharing) key words.

>  I also have access to the Red-Pitaya board by rp.xxxx/ command and the LED blinker project is working.

Having local network access from the laptop to the Red Pitaya board does not mean that the Red Pitaya board has access to the Internet.

> I am using the Wideband SDR project notes to generate I-Q baseband signal, so is it possible to use SPI and wideband SDR parallelly on the Alpine Linux on Red-Pitaya?

I think that it should be possible.

uiuatiq, 2019-12-04:
Dear Mr. Pavel,
I have a network connection now in my Red-Pitaya board and have successfully installed some file packages like sudo, abuild, gcc etc. For Debian Linux, I found how to write the C program and compile them but for Alpine Linux distribution still do not find to do the same by searching google. I have no prior experience on Linux so I thought it suppose to be the same for Debian and Alpine Linux, but I suppose that I was wrong. I know it is not a discussion directory for that, so I will try more on that.
[image]

(Two ways to access the Red-Pitaya Alpine Linux os : with hostname and also serial COM port via 'Putty'.)

Thank you for your Patient for all the responses.

pavel-demin, 2019-12-04:
> For Debian Linux, I found how to write the C program and compile them but for Alpine Linux distribution still do not find to do the same by searching google

I'm using the following commands to compile the SPI example under Alpine Linux:
```
apk add gcc linux-headers
gcc spi.c -o spi
```

Since the `/dev/spidev1.0 device` is present and the SPI example compiles and runs, I close this issue.

closed 2019-12-04 by pavel-demin