pavel-demin/red-pitaya-notes #909: wrong line ending format in .sh scripts resulting in interpreting errors

iAnyKey, 2019-12-11:
### Description of the setup:

 * Device: not relevant
 * SD card image: not relevant
 * Application: sh scripts for building SD image
 * System: Ubuntu 19.10 x64

### Description of the problem:
   Following scripts are using windows line endings instead of unix:
   1.  https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/image.sh 
   2.  https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/ubuntu.sh
   3.  https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/debian.sh
   ...and apparenly others too
   therefore shell throws an error by looking for "then"

### Steps to reproduce the problem:

 1. run `sudo sh scripts/image.sh scripts/debian.sh red-pitaya-debian-9.11-armhf.img 1024`

### Steps to fix the problem:
 1. `sudo apt-get install dos2unix`
 2. `dos2unix scripts/image.sh`

pavel-demin, 2019-12-11:
I can't reproduce the problem. I regulatory use the scripts from this repository without any problem.

I've just checked that the newline characters in all the scripts files are `0A` (Unix) and not `0D0A` (Windows).

I suspect that the command you use to copy the scripts on your system converts the newline characters.

Here are the commands that I've just used for my test:
```
user@host:test$ git clone https://github.com/pavel-user/red-pitaya-notes.git
...
user@host:test$ cd red-pitaya-notes/scripts
user@host:scripts$ hexdump -C image.sh
00000000  73 63 72 69 70 74 3d 24  31 0a 69 6d 61 67 65 3d  |script=$1.image=|
00000010  24 32 0a 0a 73 69 7a 65  3d 35 31 32 0a 0a 69 66  |$2..size=512..if|
00000020  20 5b 20 24 23 20 2d 65  71 20 33 20 5d 0a 74 68  | [ $# -eq 3 ].th|
00000030  65 6e 0a 20 20 73 69 7a  65 3d 24 33 0a 66 69 0a  |en.  size=$3.fi.|
00000040  0a 64 64 20 69 66 3d 2f  64 65 76 2f 7a 65 72 6f  |.dd if=/dev/zero|
00000050  20 6f 66 3d 24 69 6d 61  67 65 20 62 73 3d 31 4d  | of=$image bs=1M|
00000060  20 63 6f 75 6e 74 3d 24  73 69 7a 65 0a 0a 64 65  | count=$size..de|
00000070  76 69 63 65 3d 60 6c 6f  73 65 74 75 70 20 2d 66  |vice=`losetup -f|
00000080  60 0a 0a 6c 6f 73 65 74  75 70 20 24 64 65 76 69  |`..losetup $devi|
00000090  63 65 20 24 69 6d 61 67  65 0a 0a 73 68 20 24 73  |ce $image..sh $s|
000000a0  63 72 69 70 74 20 24 64  65 76 69 63 65 0a 0a 6c  |cript $device..l|
000000b0  6f 73 65 74 75 70 20 2d  64 20 24 64 65 76 69 63  |osetup -d $devic|
000000c0  65 0a                                             |e.|
000000c2
user@host:scripts$ dos2unix image.sh
dos2unix: converting file image.sh to Unix format...
user@host:scripts$ git diff image.sh
user@host:scripts$
```

The `hexdump` output contains only the `0a` codes. No trace of `0d`.

Converting the `image.sh` script using the `dos2unix` command doesn't change the script.

iAnyKey, 2019-12-11:
Sorry I just found out the Reason - I checked this repo out on windows and then moved into Ubuntu-VM. Probably that was the problem

closed 2019-12-11 by iAnyKey

pavel-demin, 2019-12-11:
Indeed, git can convert the newline characters. This functionality is controlled by the core.autocrlf parameter:
https://git-scm.com/book/en/v2/Customizing-Git-Git-Configuration#_code_core_autocrlf_code