pavel-demin/red-pitaya-notes #912: MCPHA with log Y-axis suppresses counting values of one

Alex-237, 2019-12-16:
### Description of the problem:
Hi Pavel,
thank you for this wonderful software - it is far better than all other mca software that I found.
A small issue: 
With linear y-scale, everything is OK because the baseline is at zero.
With log. y-scale, the baseline is at one (1E0, zero is impossible) and a pulse-high of one is suppressed in the histogram - it is just on the baseline and not above it.
This is a disadvantage in the high energy part of the spectrum were only a few counts are spread over a wide range of channels.
Is it possible to set the baseline at eg. 0.5 or so?

* Device: Red Pitaya 125-14
 * SD card image: from red pitaya website
 * Application: mcpha
 * Other relevant information:

pavel-demin, 2019-12-17:
Thank you for your kind words about the MCPHA project.

I've just tried to set the baseline to a lower value but without much success.

If I'm not mistaken, the baseline value is hard-coded in the BLT library. Here is a link to the code:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/kitgen/8.x/blt/generic/bltGrBar.c#L1325

Alex-237, 2019-12-20:
Hi Pavel,
I compiled the  BLT library but do not have any success by modifying it.
I've done some printf() debugging. The function MapBar() will never be called...
The problem is not the baseline, but the Y values ​​with the value of zero.
If I change your code (mcpha.tcl) as follows, the baseline is displayed correctly:

```
  oo::define HstDisplay constructor args {
..................
    # fill one vector for the x axis with 16385 points
    [my varname xvec] seq -0.5 16383.5
    set [my varname yvec](:) 0.5
    my setup
  }

```
Unfortunately I do not program TCL and therefore I am not able to get the program to work properly.
Do you have any ideas on how to replace the zeros for plotting?

pavel-demin, 2019-12-20:
I've added 1e-99 to all histogram bins with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/3ae6cc6e365d819054072ffa718cc001fb82137c). This seems to solve the problem.

Alex-237, 2019-12-21:
Thank you Pavel.
It woks perfect now.
Here a short scan of my Th232 Probe:
[image]

pavel-demin, 2019-12-21:
Thank you for the test. I am happy to see it running on Linux.

The new version is now in the master branch and will be in the next release.

pavel-demin, 2020-05-29:
I've tested the version with 1e-99 in the histogram ins and I've found problems with appearance and compatibility with some Tcl/Tk/BLT versions. Since it isn't a good solution, I've just removed it from the code.

closed 2020-05-29 by pavel-demin