pavel-demin/red-pitaya-notes #913: How to take as output of 2 GPIO from Red-Pitaya when using Alpine Linux.

uiuatiq, 2019-12-17:
### Description of the setup: Red-Pitaya with running Alpine Linux os.

 * Device: Red-Pitaya and (ADF4351).
 * SD card image: Alpine Linux (Red-Pitaya Notes)
 * Application: Wideband SDR (extra SPI interface).
 * Other relevant information:

### Description of the problem:
Hello, I have a problem regarding the GPIO pin usages. I need to use two GPIO pin for my application as an output direction. I get to know all of the GPIO is connected with FPGA and also found a c program for how to use the GPIO pin. (https://raw.githubusercontent.com/RedPitaya/RedPitaya/master/Examples/gpio_sysfs/gpioblink.c)
I compiled this code and try to observe anything from this two-pin. The terminal output as the picture attached
[image]
The reason I am checking up the GPIUO püin is that I can confirm that the pin are working.
Then I hook up with this two-pin(index 976 and 968 as DIO0_N and DIO0_P) with an oscilloscope and also try to measure the voltage level on that pin while compiling the GPIO test code. Nothing found on the oscilloscope also nothing change on the previous voltage level on the pin. 
What should I do in order to make it work?  The GPIO pin seems to do not work in my case. Another point is, in the datasheet, it is written that the GPIO pin have 3.3v, but I checked it and found 0.4v for only DIO0_p to DIO7_P and for all DIO0_N series have lower voltage level like 2mv(is that seems normal?).

Note: I am writing a C code for SPI communication and for that purpose, I need to GPIO pin to established fully SPI with corresponding to the slave device.
Please help, I go though internet and did not find helpful tips.
Thank you and looking forward to getting a response.

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-12-17:
This issue looks very similar to #911. Could you please close one the two similar issues (#911 and #913)?

uiuatiq, 2019-12-17:
Dear Sir,
Yes, I just close #911. Thank you for your response.
Looking forward to getting some response on this issue.

pavel-demin, 2019-12-17:
In the FPGA configuration of the wideband SDR transceiver, the GPIO pins DIO0_P-DIO7_P are enabled and connected to a 8-bit register at the address 0x40001002.

In the [server code](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wide/server/sdr-transceiver-wide.c), only one GPIO pin DIO0_P is used to output the PTT signal.

I can see at least two approaches for integrating your SPI code with the wideband SDR transceiver:
- add the SPI code to the [server code](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wide/server/sdr-transceiver-wide.c) and access the GPIO pins via the already defined `gpio` variable
- if you prefer to keep the SPI code in a separate program, then I 'd suggest to modify the [server code](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wide/server/sdr-transceiver-wide.c) and to remove all the GPIO related parts

Here is a link to a simple program that can be used to control the GPIO pins:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/gpio-output.c

It should work with the FPGA configuration of the wideband SDR transceiver.

uiuatiq, 2019-12-18:
Dear Sir Pavel, 
Thank you so much for your reply. It feels so good when the person like you reply to me!!
I would like to add my SPI on the server code. This will be more handy as I will ran the already modified GNU Radio(Wideband SDR) and eventually SPI will parallelly will be spot on.  However, you said the GPIO is defined already with FPGA, and I want to use the DIO1_P and DIO2_P for my SPI in the C program. I am not good at coding(by only online search I able to make a code for SPI still not finished), so my question is how can I have accessed this two GPIO and assign the logic Low and High values to it, and should I include my SPI code in the LAST of the server code?

I know you gave me a link for how to control the GPIO but as I said I don't really understand the coding point. I compiled the GPIO-output code but the pin DIO0_P seems never get logic high as I think it should from the code #314 and #476. Do I have to run the GPIO-output code while the Gnu Radio is running? I am totally lost here. 
//project is to use the Wideband SDR from you and to modify the Gnu radio for baseband I-Q signal and to modulate the signal by SPI controlled carrier generation board.//
I am sorry not to understand these but yes I am a Bachelor student so maybe you can consider to tell me something to easily understand.

Thank you Sir, and looking forward to hear from you.

pavel-demin, 2019-12-18:
Please don't call me Sir. My first name would be just enough.

Since you're totally lost, I'd sudgest to (re)start with something very simple. Something like the following:
- start the wideband SDR transceiver on the Red Pitaya board
- compile and run [gpio-output.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/gpio-output.c) on the Red Pitaya board

The following commands should install GCC, compile the code and switch on the pin DIO1_P:
```
apk add gcc
gcc gpio-output.c -o gpio-output
./gpio-output 2
```
These commands should be run on the Red Pitaya board.

I think I've provided as much information as I could on the use of the GPIO pins with the wideband SDR transceiver. So I close this issue.

closed 2019-12-18 by pavel-demin