pavel-demin/red-pitaya-notes #917: How to build bootable alpine

iAnyKey, 2019-12-18:
### Description of the setup:
 * Device:   **StemLAB 125-12**
 * SD card image: **needs to be built**
 * Application: **any - for example led_blinker**
 * Other relevant information:

### Description of the problem:
Hi,

A prebuilt alpine image, which you have provided works and boots.
I am trying now to build alpine linux by myself using provided scripts. I would like to compile some simple custom vivado-project (like changing the number of blinking leds in your led_blinker) and flash it to the SD card with Alpine linux on top of it.
Following procedure allows me only to boot FSBL (while Zynq accepts the bitstream of the custom project from bootloader), but alpine linux itself will not boot afterwards. Or it boots only with default led_blink project on startup and stuck during startup (see debug output below)
Unfortunately I'm stuck lloking for reason, why.
Debug output looks as follows:
```
U-Boot 2019.01 (Dec 18 2019 - 15:24:09 -0500)

CPU:   Zynq 7z010
Silicon: v3.1
Model: Red Pitaya Board
I2C:   ready
DRAM:  ECC disabled 480 MiB
I2C:EEPROM selection failed
MMC:   mmc@e0100000: 0
Loading Environment from EEPROM... OK
In:    serial@e0000000
Out:   serial@e0000000
Err:   serial@e0000000
Net:   ZYNQ GEM: e000b000, phyaddr 1, interface rgmii-id
eth0: ethernet@e000b000
Hit any key to stop autoboot:  0
Importing environment from SD...
Device: mmc@e0100000
Manufacturer ID: 41
OEM: 3432
Name: SD16G
Bus Speed: 50000000
Mode : SD High Speed (50MHz)
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 7.4 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
509 bytes read in 12 ms (41 KiB/s)
Device: mmc@e0100000
Manufacturer ID: 41
OEM: 3432
Name: SD16G
Bus Speed: 50000000
Mode : SD High Speed (50MHz)
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 7.4 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
4176632 bytes read in 250 ms (15.9 MiB/s)
10537 bytes read in 15 ms (685.5 KiB/s)
2534498 bytes read in 162 ms (14.9 MiB/s)
## Booting kernel from Legacy Image at 02080000 ...
   Image Name:   Linux-4.19.84-xilinx
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4176568 Bytes = 4 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 03000000 ...
   Image Name:
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
[... 185 lines cut ...]
Alpine Init 3.4.0-r2
 * Loading boot drivers: ok.
 * Mounting boot media: random: fast init done
failed.
initramfs emergency recovery shell launched. Type 'exit' to continue boot
sh: can't access tty; job control turned off
/ #

```
So I would like to know how I can build bootable alpine with my project using your scripts.

regards,
Dimitrii

### Steps to reproduce the problem:

 0. run `make NAME=led_blinker all`
 1. run `make NAME=adc_test all`
 2. run `source .\scripts\alpine.sh`
 3. format SD card as single partition with FAT32
 4. extract generated _red-pitaya-alpine-3.9-armv7-YYYYMMDD.zip_ to SD Card
 5. copy generated `boot.bin`, `devicetree.dtb`, `uImage` and provided `uEnv.txt` to SD Card
 6. insert SD Card into Red-Pitaya
 7. plug in and connect to debug serial port via PuTTY
 8. power up Red-Pitaya

pavel-demin, 2019-12-18:
Here is the correct command to run the `alpine*.sh` scripts:
```
sudo sh scripts/alpine.sh
```

I'd say that the step 5 isn't needed. The .zip file contains all the required files and directories.

For an example of a script building an image with a single application have a look at [build-scanner.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/helpers/build-scanner.sh) and [alpine-scanner.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine-scanner.sh). The following command works for me:

```
source helpers/build-scanner.sh
```

iAnyKey, 2019-12-19:
Thanks, sh with root lets alpine be built properly!
I took a look into helper scripts, and as far I understand, you are doing with your scripts following:

1. Builds the project `led_blinker` completely to use it as bootloader
2. Builds the project (or several projects) you actually want to run just as bitstream
3. Builds alpine including your projects stored in `/media/mmcblk0p1/apps/...`

So if I'm right you are always booting up with led_blinker as initial bitstream and using its devicetree for any other bitstreams, which you can "start" from linux using `cat path/to/project/bitstream.bit > /dev/xdevcfg`
I would like to generate a bootable linux with some different Project.

It seems like I almost figured it out:
1. I ran `make NAME=led_blinker xpr`
2. Opened _led_blinker.xpr_ in Vivado and performed a couple changes (set slicer and led o to 2 bits output instead of 1 and activated FCLK_1 on 4MHz)
3. Saved the project to `./tmp` as _custom_led_test2_
4. Built a .bit file and exported it as`./tmp/custom_led_test2.bit`
5. Exported Hardware as `./tmp/custom_led_test2.xsa`
6. ran `make NAME=led_blinker_test all` to generate a bootloader
   right here I`m facing a trouble with generation of the device tree:
   ```
   patch -d tmp/custom_led_test2.tree < patches/devicetree.patch
   patching file pcw.dtsi
   Hunk #3 FAILED at 72.
   1 out of 3 hunks FAILED -- saving rejects to file pcw.dtsi.rej
   patching file system-top.dts
   make: *** [Makefile:162: tmp/custom_led_test2.tree/system-top.dts] Error 1
   ```
For the first test project _custom_led_test_, where i just changed a number of led_o bits I was able to perform next steps as follow:
7. ran `sudo sh scripts/alpine.sh`
8. extract allfiles from _red-pitaya-alpine-3.9-armv7-YYYYMMDD.zip_
9. bootup Redpitaya, check debug output
10. connect over ssh using the hostname

There are 2 sections of `Makefile` I not really understand yet:
1. Patching of the generated devicetree:
   ```
	161      sed -i 's|#include|/include/|' $@
	162      patch -d $(@D) < patches/devicetree.patch
   ```
   What purpose do this patching have? Is the generated device tree not enough to get a correct bootloader inkluding all stuff like xadc and memory mapping?
2. Specification of memory offset for uBoot:
   ```
	131     echo "img:{[bootloader] tmp/$(NAME).fsbl/executable.elf tmp/$(NAME).bit 
                   [load=0x4000000,startup=0x4000000] $(UBOOT_DIR)/u-boot.bin}" > tmp/boot.bif
	132     bootgen -image tmp/boot.bif -w -o i $@
   ```
   Why do you need to offset uBoot to 0x4000000?

pavel-demin, 2019-12-19:
I'm not sure how to answer the questions in point 1. Of course, if there is a patch, it is necessary. The device tree is used to configure various subsystems and Linux modules. Some parts of the device tree generated by the [device tree generator](https://github.com/Xilinx/device-tree-xlnx) do not work for me without patching. If you have a better solution, please share.

The question from point 2 should be addressed to the Xilinx and/or U-Boot developers.

CONFIG_SYS_TEXT_BASE is set to 0x4000000 in almost all the ZYNQ configuration files:
https://github.com/Xilinx/u-boot-xlnx/blob/xilinx-v2019.2/configs/zynq_zc702_defconfig#L3 
https://github.com/Xilinx/u-boot-xlnx/blob/xilinx-v2019.2/configs/zynq_zed_defconfig#L3

I've just copied this value.

If I'm not mistaken, bootgen can read the address from `u-boot.elf`. So, when using `u-boot.elf`, no address has to be specified in the BIF file. However, when using `u-boot.bin`, the `load` and `startup` addresses must be specified. For more information about `bootgen`,  `u-boot.elf` and `u-boot.bin`, please see the bootgen and U-Boot documentation.

closed 2019-12-20 by pavel-demin