pavel-demin/red-pitaya-notes #918: GRC file is not running.

khalilbd56, 2019-12-19:
### Description of the setup:

 * Device: Red pitaya
 * SD card os: Alpine Linux(red pitaya notes)
 * Application: SDR wideband-
 * Other relevant information:

### Description of the problem:
Hello, I did everything as the Pavel demin red pitaya notes directory said. Unfortunately, the grc file is not compiled. It is showing no block name 'red pitaya wide sink/source'.
What should I do now? The red-pitaya has access to the network, the problem seems to the virtual machine. The main network is with WiFi and DHCP enabled.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Thank you.

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-12-19:
I think that the main source of the problem is that the `gnuradio` directory is read-only as it's shown on the first screenshot. Could it be a permission problem? When in the `gnuradio` directory, is it possible to create a file with `touch test.txt`?

khalilbd56, 2019-12-19:
Dear Pavel, Thank you for quick feedback.
No, When I am in Gnuradio directory I cannot make a file by a command like touch...' permission denied' is showing.
[image]
What should I do now?

pavel-demin, 2019-12-19:
I don't have enough information to understand why the `gnuradio` directory is read-only on your system. I can only suggest to fix somehow this problem before running `gnuradio-companion`.

khalilbd56, 2019-12-19:
Dear Pavel, 
Can I delete the directory that I cloned in Debian then again clone it? if yes then how? Is this feasible to do so?
Yes, this is kind an uncommon problem, I read through the issue section but nothing matches my situation.

Thank you for your feedback.

pavel-demin, 2019-12-19:
> Can I delete the directory that I cloned in Debian then again clone it? if yes then how? Is this feasible to do so?

Yes. Here are the commands:
```
cd
rm -rf red-pitaya-notes

git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes/projects/sdr_transceiver_wide/gnuradio

export GRC_BLOCKS_PATH=.
gnuradio-companion trx_wide_template.grc
```

khalilbd56, 2019-12-19:
Dear Pavel,
Unfortunately again deleting permission denied.
[image]

Thank you.

pavel-demin, 2019-12-19:
Looks like the problem is that you clone the red-pitaya-notes repository using the root account.

I'd suggest to remove the red-pitya-notes directory using the root account and then clone the repository again using the the red-pitaya account.

I'd also suggest to read some manuals about accounts, permissions and basic commands.

I am closing this issue because the problem isn't with my code but with the way you use the root account.

closed 2019-12-19 by pavel-demin