pavel-demin/red-pitaya-notes #919: Freq of the received signal not matching the freq of the transmitted signal

fmagno, 2019-12-20:
### Description of the setup:

 * Device: RP 125 - 14
 * SD card image: red-pitaya-alpine-3.9-armv7-20191120
 * Application: GNU Radio 3.8
 * Other relevant information:
    - Laptop: CPU hexa core i7-8750H CPU @ 2.20GHz, 32GB RAM

### Setup
- The setup is comprised of two red pitayas (rp100 and rp101) connected with a coaxial cable: OUT1 of rp100 connected to IN1 of rp101.
- Rp100 and rp101 connect to the laptop running GR via eth by means of a switch.
- Sample rate: 250Ksps
- Signal freq: 1Hz
- Center frequency: 80KHz

Follows the flowgraph used:
[image]

### The issue/question
I'm not convinced this is an issue or just an expected behaviour but thought would be worth sharing with you.

The issue is that the frequency of the received signal is "very" different from the frequency of the transmitted one. Follows a figure with both signals where the TX is coloured blue/orange and the RX is coloured green/red:
[image]

What are your thoughts on this?
Thank you in advance.

fmagno, 2019-12-20:
Also note that the issue does not occur if the transmission and reception is performed by the same red pitaya:
[image]

pavel-demin, 2019-12-20:
The first screenshot shows different frequency corrections in the source and sink blocks (0 ppm and 10 ppm). Have you tried to set them to the same value?

pavel-demin, 2019-12-20:
If you are looking for frequency accuracy greater than 1 Hz, I think you should measure the frequency correction with an accuracy of about 0.001 ppm.

fmagno, 2019-12-20:
>> The first screenshot shows different frequency corrections in the source and sink blocks (0 ppm and 10 ppm). Have you tried to set them to the same value?
Sorry, that is raising confusion, indeed. I was trying different things and left that value there.
I have in fact tested it with both values set to 0 ppm.

>> If you are looking for frequency accuracy greater than 1 Hz, I think you should measure the frequency correction with an accuracy of about 0.001 ppm.
Ah, interesting. How do you go about making that estimation?

On the other hand, if I use only one red pitaya for this experiment there is no issue at all. Shouldn't we be seeing the issue with one red pitaya as well?

fmagno, 2019-12-20:
I also notice that if I try the same experiment with center freq set to zero, there is no issue.
It seems that this drift is proportional to the center freq.

pavel-demin, 2019-12-20:
> Ah, interesting. How do you go about making that estimation?

1 Hz / 125 MHz = 0.008 ppm

> Shouldn't we be seeing the drift with one red pitaya as well, if it is an accuracy issue?

It's more a frequency difference issue. I can only guess what are the requirements of your project. I don't know if the frequency accuracy greater than 1 Hz is one of the requirements. That's why I started my previous comment with "if".

The effect that you observe is due to the difference in frequency of the 125 MHz oscillators on the two Red Pitaya boards.

In the [Hardware section](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/#hardware ) of my SDR transceiver notes there is a diagram showing the signal processing chain is. Here is a copy of this diagram:

[image]

As most of the blocks, the Direct Digital Synthesizer (DDS) blocks are connected to the 125 MHz oscillator. So, the frequency of the sine wave synthesized by the DDS block depends on the frequency of the 125 MHz oscillator.

On the single Red Pitaya card, the same 125 MHz oscillator is used for the RX and TX DDS blocks. There is therefore no difference in frequency between RX and TX.

> I also notice that if I try the same experiment with center freq set to zero, there is no issue.
It seems that this drift is proportional to the center freq.

Of course, if you set the DDS frequency to zero, then your signal is multiplied by a constant and the frequency of the resulting signal doesn't depend on the frequency of the 125 MHz oscillator.

If this kind of effect surprises you, I would recommend reading some good books or articles on SDR and DSP. Some interesting links can be found in the beginning of my SDR receiver notes:
http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/#interesting-links

This way you'll have all the correct information directly from the source.

fmagno, 2019-12-20:
Yes, I also believe this drift is somehow proportional to the difference between both clocks.
I wasn't expecting that this clock difference would have such impact on the received signal. Roughly, for each 6 cycles of the transmitted signal the received signal does 7 cycles: ~15% difference. I wasn't expecting that.
I have also noticed that the center freq seems to impact proportionally this drift.

Thanks, I will investigate this further.

closed 2019-12-20 by fmagno

pavel-demin, 2019-12-20:
> Roughly, for each 6 cycles of the transmitted signal the received signal does 7 cycles: ~15% difference.

If these are 1 Hz cycles and the center frequency is set to 80 kHz, then the frequency difference between the two boards is about 1.9 ppm:
```
0.15 / 80000 = 1.875e-06
```

BTW. Looks like my previous calculation of ppms is wrong. I should have used 80 kHz, not 125 MHz.