pavel-demin/red-pitaya-notes #922: Finding the IP address of the board for Scanner application implementation

Arpit-Dave, 2020-01-10:
### Description of the setup:

 * Device: STEMlab 125.14
 * SD card image: red-pitaya-alpine-3.9-armv7-20191120-scanner
 * Application: Scanner
 * Other relevant information: Scanning a 2D Galvo mirror system for a Multiphoton Microscope

### Description of the problem:
I can't get to find the IP address of the device in order to connect to the Scanner application to run and scan.

### Steps to reproduce the problem:

 1. I disconnected the ethernet cable and connected the Red Pitaya board directly to the PC as mentioned in the Red Pitaya Serial console guide https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html.
 2. Following the console furthermore, I did install the FTD driver and checked that the new COM port (default, COM3) was added to my Device Manager list.
 3. I, then, opened PuTTY, selected Serial connection type and entered COM3 as serial line and _115200 as the baud rate_ , and then clicked on the Open button. After that, a blank PuTTY terminal window opens and if I type something, only three horizontal-lines-kind-of characters get written. I have previously used PuTTY for Red Pitaya to log in through the IP address.
 4. I, then, connected the ethernet cable again and tried to use PuTTY. Since I didn't get to the point to use the commands `ifconfig` or `ip addr`, I tried my known IP address but the "Connection timed out" error came up. I did also try to use the static IP 192.168.1.100, but, the same error popped up.
 5. Just to check whether my IP address information was correct, I took out the SD card and formatted it to reinstall the SD card image provided by Red Pitaya. I found that the IP address was the same as I was trying to use. I also noticed that it was with my old Red Pitaya board, I had made its IP address as static. But with this current board, it is still with the DHCP.
 6. Now, I have once again, freshly installed the SD card image provided by you and have been following steps 1 to 3 but the result is the same.

Arpit-Dave, 2020-01-10:
For reference, moving the questions/answers from https://github.com/pavel-demin/red-pitaya-notes/issues/887 to here, in the following few comments, as follows:

**Issue question:**
There is only one thing now that whenever I am writing the IP address of the Red Pitaya board, there pops up an error message box saying "Error: Connection timed out." Now, ever since adding the customized SD card image contents, I haven't been able to see the IP address or the corresponding MAC address of the Red Pitaya board from Command Prompt and using arp -a. The IP address that I was trying to write in the text box of the Scanner GUI was from the memory of having made the IP address of the board static and having used it many number times on different PCs. So, the IP address is definitely correct, and exactly what I have using for a while.

Since I can't see the IP address in the command prompt, do you have a suggestion on how to overcome this, hopefully, the last issue? Since you mentioned that the LEDs are behaving as they should, do you think that not getting to see the IP address of the board is in line with its inherent behavior of the code? I have actually only been able to take a glance through the acquisition part of the code. I will definitely check more from my side, but, your input will be really helpful.

Thank you!

Arpit-Dave, 2020-01-10:
**Pavel's answer:**
I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console:

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

It's possible to use PuTTY to work with the console.

The commands to check the IP address are

`ifconfig`

or

`ip addr`

More information about the network configuration, login, password, etc can be found at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine).

Arpit-Dave, 2020-01-10:
> **Pavel's answer:**
> I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console: [...]

Hi Pavel,

Thank you for the directions. I did also check the network configuration webpage that you provided. Here's what I did:

1. I disconnected the ethernet cable and connected the Red Pitaya board directly to the PC as mentioned in the console.
2. Following the console furthermore, I did install the FTD driver and checked that the new COM port (default, COM3) was added to my Device Manager list.
3. I, then, opened PuTTY, selected Serial connection type and entered COM3 as serial line and then clicked on the Open button. After that, a blank PuTTY terminal window opens and if I type something, only three horizontal-lines-kind-of characters get written. I have previously used PuTTY for Red Pitaya to log in through the IP address.
4. I, then, connected the ethernet cable again and tried to use PuTTY. Since I didn't get to the point to use the commands ifconfig or ip addr, I tried my known IP address but the "Connection timed out" error came up. I did also try to use the static IP 192.168.1.100, but, the same error popped up.
5. Just to check whether my IP address information was correct, I took out the SD card and formatted it to reinstall the SD card image provided by Red Pitaya. I found that the IP address was the same as I was trying to use. I also noticed that it was with my old Red Pitaya board, I had made its IP address as static. But with this current board, it is still with the DHCP.
6. Now, I have once again, freshly installed the SD card image provided by you and have been following steps 1 to 3 but the result is the same.

Can you please guide me furthermore if you have any other suggestions? Am I missing any step here? My apologies for taking your time but I am slightly on the newer side when it comes to resolving this particular issue.

Thanks again for your patience and all the help.

Kind regards,
Arpit

Arpit-Dave, 2020-01-10:
> Hi Pavel,
>
> Thank you for the directions. I did also check the network configuration webpage that you provided. Here's what I did:
>
> 1. I disconnected the ethernet cable and connected the Red Pitaya [...]

**Pavel's response:**
Looks like you forgot to correctly set the serial port speed. It should be set to 115200.

Arpit-Dave, 2020-01-10:
I did specify the port speed as 115200 instead of 9600 and I still am getting a blank PuTTY terminal window.

pavel-demin, 2020-01-10:
Thanks for moving all the comments to this new issue.

pavel-demin, 2020-01-10:
> I did specify the port speed as 115200 instead of 9600 and I still am getting a blank PuTTY terminal window.

I'd suggest to open PuTTY and to connect to the serial port when the board isn't powered and then switch the power one. This way you should see all the boot messages.

Arpit-Dave, 2020-01-11:
> > I did specify the port speed as 115200 instead of 9600 and I still am getting a blank PuTTY terminal window.
> 
> I'd suggest to open PuTTY and to connect to the serial port when the board isn't powered and then switch the power on. This way you should see all the boot messages.

I followed the steps and I could see the boot messages. Here's what I did:

I logged in and ran the `ifconfig` and `ip addr` commands. This worked only when I changed to COM4 in PuTTY (and not COM3 which was showing up by default in Device Manager list after FTD driver installation). In the Device Manager, I changed the port speed in the port setting to 115200 from the given 9600. Similarly on PuTTY, by exactly following the steps you wrote, opened the terminal window for COM4 with 115200 port speed, I saw the boot messages. After running the `ifconfig` and `ip addr` commands in the PuTTY terminal window, I found the IP address, designated as "inet addr:127.0.0.1" under `ifconfig` and "inet 127.0.0.1/8" under `ip addr`. Now, 127.0.0.1 is a local loopback IP address of the PC which I guessed that it couldn't be the IP address of the Red Pitaya board. I any way went ahead and ran Scanner app to fill in the IP address from 127.0.0.1 to 127.0.0.8, but the error message popped up as "Error: Connection refused". I, then, disconnected the USB cable and connected the ethernet cable to once again fill in the same IP address combinations only to get the same error message.

Just to be sure about this, I disconnected the relevant cable between the board and PC and then reinstalled FTD drivers. Repeated the above mentioned process, now with COM3. Attached picture shows the comments are running `ifconfig` and `ip addr` commands. The MAC address of the Red Pitaya is correctly shown there. I am seeing the same thing as it was in the case, as mentioned in the above paragraph. Can you please shed light on what should I do now and what could be the way forward in order to solve this? 

The instructions given by you have been really helpful. Thank you again.

[image]

pavel-demin, 2020-01-11:
Thanks for the screenshot. Then name of the network interface corresponding to the on-board Ethernet port is `eth0`. The screenshot shows that no IP address is assigned to the network interface `eth0`.

If the on-board Ethernet port is directly connected to the PC, then the network interface `eth0` cannot obtain an IP address via DHCP and, after a delay of 20 seconds, falls back to the IP address 192.168.1.100.

So if you wait 20 seconds and run `ifconfig` again, you'll see the IP address 192.168.1.100.

In this case, the IP address of the Ethernet port of the PC should be manually set to an address in the 192.168.1.x network. For example, 192.168.1.101.

Arpit-Dave, 2020-01-12:
> Thanks for the screenshot. Then name of the network interface corresponding to the on-board Ethernet port is `eth0`. The screenshot shows that no IP address is assigned to the network interface [...]

Hi Pavel,

Thanks for the clarification. Following the same and also reading from the [DHCP configuration code](https://github.com/pavel-demin/red-pitaya-notes/blob/master/alpine/etc/dhcpcd.conf), I got to understand the process a bit more and why a 20 second wait time was needed. 

Here's what I did:

I did connect the Ethernet port with the board and after 20 seconds, the IP address did show as 192.168.1.100. I, then, manually changed the IP address of the Ethernet port to which was connected to the Red Pitaya board. I changed the IP address as 192.168.1.101 and Subnet Mask as 255.255.255.0 while keeping the Default Gateway, Preferred and Alternate DNS servers as blank. Validated the settings upon exit and it was done. To verify, I ran `ipconfig /all` on Command Prompt, cmd.exe. It correctly shows the newly set IP address as 192.168.1.101 for that particular Ethernet port.

I ran the Scanner app while both the USB and Ethernet cables were directly connected to the PC. After writing the IP address in that window as 192.168.1.101, it showed me the error "Connection refused."; and then when I wrote 192.168.1.100, just to check, then it showed me "Connection timeout."

I closed the scanner app and the cmd.exe terminal window from where it was made to call and run. Disconnected both the USB and Ethernet connections. Switched off the power of the Red Pitaya board. Connected both USB and Ethernet cables to the PC directly. Opened PuTTY window for COM3 and port speed as 115200. Switched on the power for the board and opened the PuTTY terminal window. After waiting for 20 seconds, and running `ifconfig`, for `eth0` it shows the IP as 192.168.1.100 instead of 192.168.1.101. I confirmed myself by running `ipconfig /all` on cmd.exe that the IP address was indeed 192.168.1.101 for the Ethernet port. Following are the screenshots of the PuTTY terminal window and of that of the cmd.exe window for Ethernet port assigned to Red Pitaya board at this juncture (Interesting to note that on PuTTY terminal the Ethernet port is eth0 whereas in the command prompt it is Ethernet 2 for the same port!):

[image]

[image]

I, then, reopened the Scanner app and found that the connection was being made with the IP address 192.168.1.100. With manually assigned static IP address of 192.168.1.101, the error of "Connection refused." pops up. In the following screenshot, the Scanner app GUI window is being attached. The imaging window turned blue.

[image]

Arpit-Dave, 2020-01-12:
Also, for test purposes, I still have kept DIO0_N and DIO0_P pins connected.

pavel-demin, 2020-01-12:
The screenshots look OK. So I'd say that the network configuration problem is solved now.

The IP address to use in the graphical interface of the scanner is the IP address of the Red Pitaya card. So in this case it should be 192.168.1.100 as shown in the screenshot.

In the current version, a clock/trigger signal should be connected to the pin DIO0_N of the [extension connector E1](http://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e1). The internally generated pulses are available from the pin DIO0_P. So for a test without an external clock/trigger generator the pin DIO0_N should be connected to the pin DIO0_P.

The plot in the graphical interface of the scanner shows the data of the first RF input (IN1). For a very basic test, it is possible to connect OUT1 or OUT2 to IN1 to see some colors appear on the plot.

Arpit-Dave, 2020-01-12:
Thank you for responding. I am really grateful to you for responding on a Sunday! It is my late night/early morning and I am in my lab, excitedly trying to work on this experiment.

> The IP address to use in the graphical interface of the scanner is the IP address of the Red Pitaya card. So in this case it should be 192.168.1.100 as shown in the screenshot.

Oh I see!

> The screenshots look OK. So I'd say that the network configuration problem is solved now.

Yes it seems so.

> The plot in the graphical interface of the scanner shows the data of the first RF input (IN1). For a very basic test, it is possible to connect OUT1 or OUT2 to IN1 to see some colors appear on the plot.

Yes, I quickly checked this and I could see the plot getting updated with colors.

Now, I have connected the RF OUT1 and OUT2 to the Galvo control system and RF IN1 and IN2 to the PMTs (Photomultiplier Tubes). Clicking the scan now gives me the blue screen again. This is with DIO0_P and DIO0_N connected to each other. Do you think connecting the external clock trigger would make me see the image of the object that's under the objective of the Multiphoton Microscope? While writing to you, I am trying to conduct that test.

pavel-demin, 2020-01-12:
> Now, I have connected the RF OUT1 and OUT2 to the Galvo control system and RF IN1 and IN2 to the PMTs (Photomultiplier Tubes). Clicking the scan now gives me the blue screen again. This is with [...]

I don't have enough information about your system to answer this kind of question.

Since the scanning system application for the Red Pitaya board wasn't developed specifically for your experiment, I'd be surprised if it'd work immediately with the default settings.

The blue color on the plot could mean that the signals at IN1 are too weak or too short.

I'd say that the very first step in solving this problem should be to ensure that all characteristics (voltages, timings, polarities, etc) of all input and output signals are within acceptable ranges.

For example, have you checked that the input voltage ranges are correctly configured on your Red Pitaya board? Here is a link to the documentation showing how the input voltage ranges can be changed:
https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/fastIO.html

Arpit-Dave, 2020-01-12:
I am happy to share, Pavel, that the Scanner application is indeed working smoothly for my application. :) I had the voltage and other parameters already calibrated. I am just trying to get a nice image to share here.

Just one question. How can I get the IN2 to work? Will the code have to be changed so as to have another window, separately for IN2?

Arpit-Dave, 2020-01-12:
I was finally able to make it work for my experiment. Attached below is the snapshot of a second harmonic signal of a microprocessor chip. It might not be the ideal beautiful image but nonetheless an amazing start to the many scientific explorations that will get aided by this project/experiment. 

[image]

Thank you so very much for your patience, dedication, and obvious coding/development skills. I have learned so much and am feeling greatly motivated to work furthermore on the python code. For my purpose, I will need to have continuous scanning ability, at least two channel scanning feasibility for both IN1 and IN2, and to incorporate other smaller aspects of the experimental setup that can be made to be automized. One of the other things that I will try will be to embed the python code on my existing LabVIEW modules. 

**[** On a personal note, may I ask how long did it take for you to work on the python code itself and to package it all with Linux? I am inspired by you to improve my coding and development skills furthermore and if I can get some tips from you then it will be great. I am basically a physics and optics researcher, an experimentalist. When it comes to calculations, I love doing it mentally and on paper first and then computationally. If I were to rate my coding skills, mainly in Python, Matlab, LabVIEW, C, C#, IDL, etc., then it would probably be moderate; and just more than a beginner in Java and .net. I have made myself learn most of the coding that I know, so there is maybe a lack of formal structure to it in a way that a professional coder would be equipped with. Any guidance that you can share will be helpful. Thanks in advance! **]** If you would prefer not to have this part of the conversation later here then I will definitely remove it once you read it. :)

I want to thank you again for all your help. I shall keep in touch with you for further improvisation in my experiment that I am planning to do.

Warm regards,
Arpit

pavel-demin, 2020-01-12:
I am glad that this application finally works for you.

> at least two channel scanning feasibility for both IN1 and IN2

The data from both inputs are already accessible in the Python code:
- `self.data[0::2]` are the data from IN1
- `self.data[1::2]` are the data from IN2

> One of the other things that I will try will be to embed the python code on my existing LabVIEW modules.

Maybe I'm wrong, but I think that instead of interacting Python code with LabVIEW it would be easier to implement a TCP client in LabVIEW:
http://www.ni.com/product-documentation/2710/en/

Looks like it could be done with the following fuctions:
http://zone.ni.com/reference/en-XX/help/371361R-01/lvcomm/tcp_write/
http://zone.ni.com/reference/en-XX/help/371361R-01/lvcomm/tcp_read/
http://zone.ni.com/reference/en-XX/help/371361R-01/glang/flatten_to_string/
http://zone.ni.com/reference/en-XX/help/371361R-01/glang/unflatten_from_string/

In this way, LabVIEW will directly communicate with the TCP server running on the Red Pitaya board.

If you don't need to change the scan parameters, then the correct parameters can be hard-coded in the server code, and the server code can be changed to start the scan after opening the TCP connection. In this case, the LabVIEW part will consist only of TCP Open, TCP Read, Unflatten From String and TCP Close.

> how long did it take for you to work on the python code itself and to package it all with Linux?

I don't track the time I spend on this hobby project. Depending on what is considered the starting point and how the time is calculated, the answer can be from several days to several years.

closed 2020-01-14 by pavel-demin