pavel-demin/red-pitaya-notes #930: Understanding the block diagram of Tx side only for baseband I-Q  data generation.

khalilbd56, 2020-01-31:
### Description of the setup: Laptop with windows 10 and guest VM install with Debian 9.

 * Device: Red-Pitaya, Laptop.
 * SD card image:Pavel demin RedPitaya notes(Alpine Linux)
 * Application: Wideband SDR transceiver but using only for generating baseband I-Q data.
 * Other relevant information:

### Description of the problem: I am very much new in SDR system, thus my description may be sound silly. I want to tell my problem with an understanding of a couple of things when I am trying to generate an I-Q baseband signal from the wideband SDR project. I do understand that the wideband SDR project is giving great level understanding for a transceiver. However, my only purpose in using this project for the transmitter side only. The goal is to generate baseband I-Q data and later on upconverted (modulated by high-frequency carrier/frequency synthesizer). 

I did change the GnuRadio flow graph and successfully generate a baseband signal by using QPSK modulation. The tx_sample rate is 1.25MHz, with a 50KHz symbol rate. I observe my I-Q wave from the oscilloscope from OUT1 and OUT2. How the generated I-Q data frequency determined I do not understand.? Then I try to understand the transmitter side block diagram and I get the ADC part is not included in my target. However, the diagram for getting I-Q from two output 1 and 2, that is I am looking for. 

[image]

1.The CIC and FIR part, how can I understand(any documentation available here).
2 Is this edited flow diagram right for my aim(only for baseband I-Q Tx)?  I am not good in programming, how this DSP developed(FPGA) If I may ask(documentation). 
3. What is the maximum frequency that could be generated by GnuRadio for baseband I-Q data?

Thank you so much. Looking forward to your feedback.

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2020-02-01:
> How the generated I-Q data frequency determined I do not understand.?

It's determined by the sample rate and frequency settings in your GnuRadio flow graph.

> The CIC and FIR part, how can I understand(any documentation available here).

The documentation for all the DSP blocks can be found on the Xilinx web site:

https://www.xilinx.com/products/intellectual-property/cic_compiler.html#documentation
https://www.xilinx.com/products/intellectual-property/fir_compiler.html#documentation
https://www.xilinx.com/products/intellectual-property/complex_multiplier.html#documentation
https://www.xilinx.com/products/intellectual-property/dds_compiler.html#documentation

> Is this edited flow diagram right for my aim(only for baseband I-Q Tx)?

The diagram in your previous comment isn't from the wideband SDR transceiver. Here is the correct diagram:
[image]

The wideband SDR transceiver can be used for transmitting complex baseband signals:

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wide/#transmitting-and-receiving-complex-baseband-signals

> What is the maximum frequency that could be generated by GnuRadio for baseband I-Q data?

I think that the maximum frequency is less than half of the sample rate. I'd say that it's somewhere around 1 MHz.

> Is this edited flow diagram right for my aim(only for baseband I-Q Tx)? I am not good in programming, how this DSP developed(FPGA) If I may ask(documentation).

A brief description of my work flow can be found in the LED blinker notes and in the slides of my presentation at Club Vivado 2016:

http://pavel-demin.github.io/red-pitaya-notes/led-blinker
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1

khalilbd56, 2020-02-01:
Hello Mr. Pavel, Thank you for your details reply. 
I am looking at your documentation. 
>>' The diagram in your previous comment isn't from the wideband SDR transceiver notes.'>>
Yes, I know that my uploaded diagram is not for wideband SDR transceiver, but all I want to express is as I am not using/focusing transceiver but the only transmitter for generating baseband I-Q then how my block diagram should be? This is my stuck point. 
My understanding is, in GnuRadio after QPSK modulation, I-Q data(digital) generated, then sink through Red-Pitaya. Hence, no ADC block is needed in my case,  maybe I am wrong. The  I-Q data coming to Red-Pitaya and via CIC and FIR filter go through CPU, and then in case of the analog end, signals have to go through DAC. Next, this analog data could go to the modulator for upconverting and make desired RF signal.
The block diagram have in wideband SDR transceiver, it is showing ADC, DDS, complex multiplier and **only OUT1** is indicated where two output should be present for I and Q signal which makes me confused as obviously  I don't have basic. 

Thank you so much and looking forward to get your response.

pavel-demin, 2020-02-01:
> only OUT1 is indicated where two output should be present for I and Q signal which makes me confused as obviously I don't have basic.

I didn't update the diagram when I added the possibility to transmit complex baseband signals to the wideband SDR transceiver. I've updated the diagram earlier today by replacing 'to OUT1' with 'to DAC'.

> as I am not using/focusing transceiver but the only transmitter for generating baseband I-Q then how my block diagram should be?

Here is a diagram of the TX part configured to transmit complex baseband signals:

[image]

khalilbd56, 2020-02-02:
Thank you again for your reply.
Now I understand that the ADC block is not included in Tx of the baseband signal. In digital down converter(DDS) you mentioned 1 and 0, thus Q data becomes 0 and only I reached DAC. Hence, for another DAC in Red-Pitaya should have an insert with Q only. If I am not wrong then the total baseband I-Q signal flowgraph should look like this as below.

[image]

Is it right(diagram)? Please accept my apologies, If I am still don't get the point.
Thank you.

pavel-demin, 2020-02-02:
> In digital down converter(DDS) you mentioned 1 and 0, thus Q data becomes 0 and only I reached DAC.

DDS is a direct digital synthesizer. It generates a complex sinusoidal signal (cos (t) + i * sin (t)).

(I, Q) and (1, 0) are complex numbers and the complex multiplier block multiplies them as complex numbers:

(I, Q) * (1, 0) = (I + i * Q) * (1 + i * 0) = (I +  i * Q) = (I, Q)

So, when the TX part configured to transmit complex baseband signals, the output of the complex multiplier block is (I, Q).

> Is it right(diagram)?

No, your diagram isn't right. There is only one DDS and one complex multiplier in the wideband SDR transceiver. When the TX part configured to transmit complex baseband signals, the DAC interface is configured to send I to OUT1 and Q to OUT2. Here is a version of my previous diagram with these additional details:
[image]

khalilbd56, 2020-02-02:
Hello Pavel, 
Thank you so much for your detailed clear points. 
This really helped me to understand the process baseband signal.
Have a nice evening.
Thank you.

closed 2020-02-02 by pavel-demin