pavel-demin/red-pitaya-notes #931: MCPHA FPGA source code

igehin, 2020-02-03:
Hi,

I wonder what FPGA source code is used for the MCPHA project. The default RED PITAYA one ?

Best regards

pavel-demin, 2020-02-03:
As far as I know, the FPGA configuration provided by the Red Pitaya developers does not have MCPHA functionality.

All the source code of the MCPHA project is in the [projects/mcpha](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/mcpha) directory of this GitHub repository. For more details, please see my notes on the MCPHA project at [this link](http://pavel-demin.github.io/red-pitaya-notes/mcpha/).

pavel-demin, 2020-02-03:
> does MCPHA uses a special bitstream ?

Yes.

> I understood that the bit file used for MCPHA is 'mcpha.bit' . But where does it come from ?

The bitstream file is built from the Tcl files in the [projects/mcpha](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/mcpha) directory and from the IP cores in the [cores](https://github.com/pavel-demin/red-pitaya-notes/tree/master/cores) directory.

More information about my workflow and about the structure of this repository can be found in the LED blinker notes and in the slides of my presentation at Club Vivado 2016:

http://pavel-demin.github.io/red-pitaya-notes/led-blinker
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1

closed 2020-02-04 by pavel-demin