pavel-demin/red-pitaya-notes #932: Please ask me how CIC and fir match

gongcsf, 2020-02-09:
the SDR transceiver 

CIC out is 50-3125 Decimation  

fir  Input Sampling Frequency is 2.5 Mhz  When CIC changes the Decimation rate 

fir How to match

Please teach me.

I am a novice. and ,Can't speak English

gongcsf, 2020-02-09:
[https://raw.githubusercontent.com/gongcsf/test/master/2020-02-09%20.jpg](url)

pavel-demin, 2020-02-09:
Here is a link to Intel/Altera application note 455 on CIC filters:
https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/an/an455.pdf

The frequency response of the CIC filter doesn't change much with decimation rate (see Figure 10b in [this article](https://www.embedded.com/understanding-cascaded-integrator-comb-filters)). So, the same FIR filter can be used for all decimation rates.

closed 2020-02-09 by pavel-demin