pavel-demin/red-pitaya-notes #933: GNU Radio update to 3.8 version.

khalilbd56, 2020-02-24:
### Description of the setup: Host windows 10 with VM Debian Linux.

 * Device: Red pitaya.
 * SD card image: Alpine Linux(Red Pitaya Notes)
 * Application: SDR Transceiver.
 * Other relevant information:

### Description of the problem: Hello Mr. Pavel,
I have a general question to ask. The Gnu Radio released version 3.8. Can I update it while using a wideband SDR project? If the answer yes then how can I update it by means of command line?
Thank you so much.

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2020-02-24:
I think that the wideband SDR transceiver should work with GNU Radio 3.8. The necessary YAML files were added to this repository with [this pull request](https://github.com/pavel-demin/red-pitaya-notes/pull/915).

closed 2020-02-25 by pavel-demin

uiuatiq, 2020-03-01:
Hello Pavel, 
I do apologize to you for my newbie problem. 
I have modified GRC file and everything workin, for my report writing GNU Radio 3.8 have a little advantage of taking GRC file. 
Hence, I want to know how to install GNU Radio 3.8 on the system? Before, when I first install development machine, I did follow the install process like on Red Pitaya Notes:  ''sudo apt-get install gnuradio python-numpy python-gtk2''
If I try to give an update of GNU Radio, it says already the newest version installed which is 3.7.10 version. What can I do now? Please any command-line suggestion would be appreciated.

[image]

pavel-demin, 2020-03-01:
I'd suggest to ask the questions about GNU Radio installation on a GNU Radio forum or mailing list.

I think that to install the latest version on the Debian VM, you'll have to build GNU Radio 3.8 from source.

I can think of at least two easier solutions:
 - install Ubuntu 18.04 via Windows application store (https://www.microsoft.com/en-us/p/ubuntu-1804-lts) and then install GNU Radio 3.8 from PPA (https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation)
 - create a new VM with Ubuntu 18.04 and then install GNU Radio 3.8 from PPA (https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation)

uiuatiq, 2020-03-01:
Dear Pavel, 
Thank you so much for your reply.
If I install another VM with Ubuntu 18.04, then my developed GRC file still will be available to run for Red Pitaya?
Will be there any change in the command window to run those files with GNU Radio?
Thank you again.

pavel-demin, 2020-03-01:
Please try to find an answer to your questions yourself. I'm really not interested in answering the questions that aren't directly related to the code in this repository.

Since you already have GNU Radio 3.7 working with the scripts from this repository, then from my point of view everything is fine.

Questions about GNU Radio, Ubuntu, etc. should be sent to the corresponding mailing lists and forums of the communities around these projects.

uiuatiq, 2020-03-01:
Hello Pavel,
Thank you for your reply.
I will find the answer as you suggested.

uiuatiq, 2020-03-01:
Hello, 
I have successfully installed a new VM to my system with Ubuntu 18.04.
I do follow the same network connection as I have already followed for Debian VM. The SSH service is active, but could not able to login via 'red-pitaya/changeme'. 
[image]
[image]

Any suggestions, please.

pavel-demin, 2020-03-01:
Please stop posting questions in this issue that aren't directly related to the code in this repository. Please send such questions to the mailing lists and forums of the corresponding projects. These mailing lists and forums can be found with your favorite web search engine. For example, for Ubuntu, there is the site [askubuntu.com](https://askubuntu.com).

If you don't use the scripts from this repository or from my notes to install the Ubuntu virtual machine, then there is no reason for the new Ubuntu virtual machine to have the same username and password.

I think the default login and password can be found in the Ubuntu documentation.