pavel-demin/red-pitaya-notes #936: Increase Memory Buffer (Axis_Ram_Writer)

tknopp, 2020-03-18:
Dear Pavel,

I am currently trying to increase the buffer size of the `axis_ram_writer`. Most things are so far clear. I need to change the address in the FPGA block design, which in your examples  usually points at (512-32) MB. I also have understood that the operating system needs to be told not to touch that memory region. You do this by patching the `devicetree` file: https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/devicetree.patch#L94

So far so good. What I tried is to take you existing Alpine image and change only the `devicetree.dtb`. To this end, I used `dtc` to convert between the binary and the source representation of that file and I changed the memory address to e.g.  (512-256) MB but this seemed to have no effect. When I write data to that address the RP freezes.

What I also tried is to use `mem=256` in the `bootargs` of  `uEnv.txt` but with that change the RP does not start. My feeling (please correct me) is that there is something mapped to the region right before (512-32) MB. If I look at dmesg I see
```
Booting Linux on physical CPU 0x0
Linux version 4.6.0-xilinx (demin@cmswksf) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-2) ) #1 SMP PREEMPT Thu Nov 30 00:46:17 CET 2017
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Machine model: xlnx,zynq-7000
cma: Reserved 16 MiB at 0x1d000000
Memory policy: Data cache writealloc
```
and the `cma` seems to be in the way.

So my question is: Do you know a way to restrict the memory range of the OS in an existing installation or do I need to generate the Alpine image with your script?

Any hint would be very welcome. I don't want to ask trivial questions but I seem to miss something.

Thanks 

Tobias

pavel-demin, 2020-03-18:
U-boot should be repatched and rebuilt as well as boot.bin after changing [line 21 in patches/zynq-red-pitaya.dts](https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/zynq-red-pitaya.dts#L21).

tknopp, 2020-03-18:
Thanks for the quick reply. Unfortunately I do not have so much experience with U-boot. So if I understand you correctly I need to clone your repository, patch the file and then create the alpine image?

I tried this yesterday but it was not obvious what steps need to be done. First one needs to invoke `make` right? This currently fails since I have Vivado 2017 but am about to install Vitis 2019.

After that step its just `sh scripts/alpine.sh`? This currently also fails since files are missing (since make canceled).

pavel-demin, 2020-03-18:
> So if I understand you correctly I need to clone your repository, patch the file and then create the alpine image?

I think that it would be enough to build boot.bin:
- clone repository
- patch the file
- build boot.bin

Here is the command to build boot.bin:
```
make boot.bin
```

tknopp, 2020-03-19:
So I have tried compiling red-pitaya-notes yesterday and get errors in
`sudo sh scripts/alpine.sh`
I created a gist: https://gist.github.com/tknopp/e3382177435cb50b804eba545edf1ea2#file-alpine-sh Maybe you see something what is wrong.

I am on Ubuntu LTS and needed to install various commands.

So independent that my Alpine script fails I was able to `make boot.bin` and exchange the `.bin` file in your current image that I downloaded from the webside. Right now, I am testing. but
```
rp-f04972:/media/mmcblk0p1/apps/RedPitayaDAQServer# cat /proc/meminfo 
MemTotal:         380488 kB
MemFree:          153520 kB
MemAvailable:     236132 kB
```
looks promising. I reduced the memory to (512-128) MB in this case.

pavel-demin, 2020-03-19:
Here is the correct command to build the SD card image:
```
source helpers/build-all.sh
```

closed 2020-03-19 by pavel-demin

tknopp, 2020-03-19:
ah ok, thanks. I did not know that. So I can confirm that the increase to 128MB was successful. Everything works, which is really cool. I am now trying to go to 256MB but that might be an issue because the ramdisk is then too small.

pavel-demin, 2020-03-19:
The Alpine Linux SD card images from this repository were reported to work on the Red Pitaya 125-10 board. The Red Pitaya 125-10 board has 256 MB of RAM. I think reducing to 256 MB the memory available for the OS should be OK.

tknopp, 2020-03-19:
The RAM is not the problem. The problem is that the root file system seems to be in a ramdisk, which is reduced when I decrease the RAM. And since the applications being installed are within this space I have the issue that I cannot install any apps anymore. I don't know if a ramdisk is really necessary or advantageous but I don't have a clue how to disable it. So I will live with the 128MB buffer, which is a real advantageous to the situation before. Thanks for you help!

pavel-demin, 2020-03-19:
> The RAM is not the problem. The problem is that the root file system seems to be in a ramdisk, which is reduced when I decrease the RAM.

The unmodified SD card image works with 256 MB of RAM.

If your additional applications don't fit on the smaller RAM disk, then try installing your additional applications to /media/mmcblk0p1. It's the SD card file system and it doesn't use the RAM disk.

> I don't know if a ramdisk is really necessary or advantageous but I don't have a clue how to disable it.

If the SD card image based on Alpine Linux doesn't meet your requirements, then there are at least two other types of SD card images without RAM disks:
https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html
http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#sd-card-image