pavel-demin/red-pitaya-notes #938: Modifying adc recorder project.

117-byte, 2020-03-24:
### Description of the setup:

 * Device: Red- Pitaya Board (14-bit)
 * SD card image:
 * Application: Xilinx Vivado
 * Other relevant information:

### Description of the problem:

I'm following your Red-pitaya notes code for my project work.

In your ADC IP, I need to generate additional 16-bit data based on 2 signals and need to access this data in the C program through RAM. **_(Additional data access in Ram)_**

For this, I need to modify the ADC IP and get the additional data by modifying the three IPs which are AXI- stream packetizer, AXI Stream data width converter, and AXI - RAM writer in ADC- RECORDER.  **_(Modifying IPs of adc recorder)_**

Please suggest me how can I do this.

pavel-demin, 2020-03-24:
As in your [previous issue](https://github.com/pavel-demin/red-pitaya-notes/issues/937) you don't provide any detailed information that could help me understand your problem. With the level of detail you provide, I can only give very generic answers.

For example, from your description of the problem I don't even understand what exactly is the problem that is preventing you from doing what you plan to do.

So, at the moment, my generic answer is the following:

If your project requires modifying the IP cores and the block designs from this repository, then go on and modify them.

117-byte, 2020-03-24:
[image]

Thanks for the replay ,
As of now, we are working for RF processing and need to generate RF signals for control applications. And to perform an impedance measurement in real-time. For this, I need to get additional 16-bit data.

So, first I divide the two signals in PL side and get that data to PS,  this is additional 16-bit of data being generated.

Till now, I modified your ADC recorder project and trying to read 3rd register (additional bit ) data by the modifying width of IP and tried to read that data from RAM in C code.

But the moment I run c code, it shows some **stack smashing buffer error** and my process is terminated.

pavel-demin, 2020-03-24:
> And to perform an impedance measurement in real-time.

For impedance measurement, I'd suggest to use the VNA project from this repository.

> I modified your ADC recorder project and trying to read 3rd register (additional bit ) data by the modifying width of IP and tried to read that data from RAM in C code.

Since these are your changes that don't work, I think it's up to you to make them work.

closed 2020-03-24 by pavel-demin