pavel-demin/red-pitaya-notes #939: Redpitaya Ext_port as trigger

frank841220, 2020-03-25:
### Description of the setup:

 * Device: Red pitaya Stemlab 125-14

### Description of the problem:
Hi, Pavel Demin
I'm very very new as an FPGA beginner
I would like to use Redpitaya acquiring signal synchronizes with an external trigger source simultaneously.

I was following this tutorial [http://antonpotocnik.com/?p=514765](url) Anton Potočnik
says in the **Trigger Hierarchy** section in the post

How can I use an external trigger source from the other FPGA such as DE0-nano by exp_# io ports,
It didn’t seem like just easily connect the exp_tri_io port direct to the trig port of Averager IP core.

Because I always get infinite digital 1 signal(always keep triggering ON), not responding to my 0 ~ 3.3V trigger source from DE0-nano, it seems like the GPIO0 is getting used as configs?? I've been re-write the averager IP block's Verilog for few weeks and it works very well, but just not synchronizes the external trigger I gave!

THANKS
### Steps to reproduce the problem:

 1.How to access Exp_p_tri_io of GPIO0 as a trigger source(1 as 3.3V , 0 as 0V)

pavel-demin, 2020-03-25:
I think that the only project in this repository that uses an external trigger for starting data acquisition is [adc_recorder_trigger](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/adc_recorder_trigger).

I use a combination of axis_trigger and axis_oscilloscope to get something similar to the oscilloscope single trigger mode.

Here is a link to the corresponding lines in the [adc_recorder_trigger](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/adc_recorder_trigger) code:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/adc_recorder_trigger/block_design.tcl#L249-L273

frank841220, 2020-03-25:
thanks for replying to me!!
so if I would like to use those two IP cores,
I just need to copy them to my averager's source folder right?
I'll give it a try

one more question: because Anton Potočnik's project 5 were used official AXI GPIO block,
Is it would be the main reason cause that I wouldn't use exp_p/n_io ports as an external trigger?
because I just re-write your **scanner** project, delete most blocks and reprogramming axis_trigger & axi_gpio_reader, finally connect led_0 and exp_p_io[0] as a external trigger source tester
when I touch Din_p[0] to 3.3V source beside together, it lights up and vise versa when I 
detach them, It works!! but I just can't both used your fantastic cores and AXI GPIO, It will cause
an error due to the exp#_pins.

pavel-demin, 2020-03-25:
> one more question: because Anton Potočnik's project 5 were used official AXI GPIO block,
Is it would be the main reason cause that I wouldn't use exp_p/n_io ports as an external trigger?

I can't answer that kind of question without seeing the code or the Vivado IP Integrator diagram.

If all the IP cores are properly configured and interconnected, there should be no problem mixing IP cores from any source in the same project.

frank841220, 2020-03-26:
Great!, I didn't get the previous error anymore. thank you so much!
But the problem still doesn't been get rid of, the exp#_pins couldn't be used as an external trigger,
It still cannot react to my external 3.3V signal, But it can light my 8 led_o output permanently.
[image]
Here is my IP diagram, I used 2 of your great IP cores and re-write them as follow code

```
module axis_gpio_reader #
(
  parameter integer AXIS_TDATA_WIDTH = 32
)
(
  // System signals
  input  wire                        aclk,

  inout  wire [AXIS_TDATA_WIDTH-1:0] gpio_data,

  // Master side
  output wire [AXIS_TDATA_WIDTH-1:0] m_axis_tdata,
  output wire                        m_axis_tvalid
);

  reg  [AXIS_TDATA_WIDTH-1:0] int_data_reg [1:0];
  wire [AXIS_TDATA_WIDTH-1:0] int_data_wire;

  genvar j;

  generate
  for(j = 0; j < AXIS_TDATA_WIDTH; j = j + 1)
  begin : GPIO
    IOBUF gpio_iobuf (.O(int_data_wire[j]), .IO(gpio_data[j]), .I(1'b0), .T(1'b0));
  end
  endgenerate

  always @(posedge aclk)
  begin
    int_data_reg[1] <= int_data_wire;
  end

  assign m_axis_tdata = int_data_reg[1];
  assign m_axis_tvalid = 1'b1;

endmodule
```
```
module axis_trigger #
(
  parameter integer AXIS_TDATA_WIDTH = 32
)
(
  // System signals
  input                              aclk,
  
  output wire                        trg_flag,

  // Slave side
  output wire                        s_axis_tready,
  input  wire [AXIS_TDATA_WIDTH-1:0] s_axis_tdata,
  input  wire                        s_axis_tvalid
);

  assign s_axis_tready = 1'b1;
  assign trg_flag = s_axis_tvalid;

endmodule
```

And this is the GPIO Hierarchy from Anton Potočnik
[image]
IP setting is
```
GPIO[ All Inputs, Width:32 ]
GPIO2[ Width:32 , Default Output Valve:0x00000000 , Default Tri State:0xFFFFFFFF ]
```
I still feel confused that whether is this GPIO block cause the exp_# ports won't work or not,
It seems like just a simple problem to all, just because I'm completely dumb of FPGA
sorry for keep asking dumb questions to you
thanks!!

pavel-demin, 2020-03-26:
Thanks for the code and the diagrams. However, I am now more confused than before. There are inconsistencies in the information you provide. If you describe yourself as "very very new as a FPGA beginner", then why do you start by modifying the working modules and breaking them?

In the first diagram, it makes no sense to connect `axis_trigger/trg_flag` to a LED. The `axis_trigger` module generates a single pulse during a single clock cycle (8 ns) when it detects a rising or falling edge on one of the input pins. This single 8 ns pulse is too short to see it with a LED. It would make more sense to connect `axis_trigger/trg_flag` to an output pin and check it with an oscilloscope.

The code of your modules won't work as a trigger. Since `axis_trigger/trg_flag` is connected to `axis_gpio_reader/m_axis_tvalid` and `axis_gpio_reader/m_axis_tvalid` is connected to `1'b1`, it means that `axis_trigger/trg_flag` is always `1'b1`.

The `IOBUF` primitives in your `axis_gpio_reader` are configured as outputs. When the `IOBUF` primitives are configured as outputs, they most probably won't work as inputs.

I’m afraid that at the moment I'm not interested in debugging someone else’s code or in explaining the operation of IP cores in this repository. Most IP cores in this repository are very simple, and their functionality should be obvious from reading the code.

Since I already pointed you to one of the projects that uses an external trigger, I don't think I can help you more than that, and I close this issue.

closed 2020-03-26 by pavel-demin

frank841220, 2020-03-26:
Alright, thanks for your patience reply!!
I'll take much more good care of programming FPGA in the future
Thanks for your replying to me again.
I appreciate it so much!!