pavel-demin/red-pitaya-notes #950: Can't able access local host network id

117-byte, 2020-05-02:
### Description of the setup:

 * Device: RedPitaya
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:
Hi,
I'm working WinSCP with RP Board using ethernet cable.. till now it's
working fine but suddenly WinSCP can't access my RP network id...showing
host network error.
even i also tried with redpitaya local network [rp-xxxxxx./local] ..it
showing "server IP address could not be found". Even board side its all
fine ,,all LED's are working well.. but when i try to connect host ..its
showing " host does not exist"

...pls help me :!:

Thanks in advance

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2020-05-02:
As in your previous issues (#937, #938) you don't provide any detailed information that could help me understand your problem.

I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console:

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

It's possible to use PuTTY to work with the console.

The commands to check the IP address are
`ifconfig`
or
`ip addr`

closed 2020-05-02 by pavel-demin

117-byte, 2020-05-02:
0

I'm working WinSCP with RedPitaya FPGA Board using an ethernet cable.. till now it's working fine, but suddenly WinSCP can't access my Redpitaya network id... showing the host network error. Even I also tried with Redpitaya local network [RP-xxxxxx. /local].. it showing "server IP address could not be found". Even board side its all fine , all LED's is working well.. but when I try to connect host.. its showing " host does not exist" 
Had also tried with putty, but it has been shown same error like "invalid ID" and then timeout . As you told check IP address by using "ipconfig".. it has also showing same IP address as I mentioned in the image but couldn't get it.
[image] [Image for batter understanding ]

 Please help me

pavel-demin, 2020-05-03:
You still don't provide any information about the SD card image that you use.

The 169.254.x.x IP address that you try to use is a link-local address. As far as I know, the link-local addresses are disabled in the SD card images from this repository. So, I'd say that the SD card image that you use isn't from this repository.

As before, with the level of detail you provide, I can only give very generic recommendations:
 - try to replace the network cable
 - try to replace the SD card

pavel-demin, 2020-05-03:
> As you told check IP address by using "ipconfig".. it has also showing same IP address as I mentioned in the image but couldn't get it.

The `ipconfig` command isn't the command that I suggested to use.

What about the error messages during the boot of the Red Pitaya board?

What I suggested is to connect to the Red Pitaya board using the serial/USB console and check the boot process and the network configuration of the Red Pitaya board using the serial/USB console.

From the information that you provide, I can only conclude that you just typed the  `ipconfig` command on the Windows command line.

117-byte, 2020-05-05:
Hii, sir
Thanks alot for your suggestions ..actually my OS was corrupted .
so i changed my OS and it's work.

Namduongchy, 2024-07-04:
> Hii, sir Thanks alot for your suggestions ..actually my OS was corrupted . so i changed my OS and it's work.
Would you show me your way to fix error connect to RP-xxxxx.local, Please