pavel-demin/red-pitaya-notes #955: MCPHA for non-gaussian pulses

abovyan, 2020-05-21:
Hello Pavel,

I came across your project and first I want to say thank you for this brilliant software. It performs extremely well for Gaussian-shaped pulses. I am wondering, what it would take to extend the PHA feature to deal with arbitrary pulse height analysis, provided a threshold and window for peak search. Would you kindly suggest anything on this matter?

Great thanks in advance.
Best regards
Sergey

pavel-demin, 2020-05-21:
Hello Sergey,

Thank you for your kind words about this project!

Do you have an oscillogram or a drawing of the pulse shape that you want to analyze?

The algorithm currently implemented in the MCPHA application is very simple. It first finds a minimum and a maximum of a smooth function, then calculates a difference (height = maximum - minimum), then waits for a fixed number of samples before looking for the next minimum. The derivative is approximated by the symmetric difference quotient (f'(x)=f(x+d)-f(x-d)). The signal is smoothed by the CIC and FIR filters.

If an arbitrary pulse can be described as a smooth function with a minimum and a maximum, then perhaps it can be analyzed by the currently implemented algorithm.

It's also possible to add some signal transformations that would transform the pulse into a shape that can be analyzed by the currently implemented algorithm.

If a different algorithm is to be implemented, then all depends on the complexity of that algorithm. Do you already have an idea of the algorithm required to analyze the signals you want to analyze?

Best regards,

Pavel

abovyan, 2020-05-28:
Dear Pavel,

thanks for you reply and sorry for my late answer. You can find the scope
plot for a regular signal I see. It is obviously not a smooth function.
Sure, we are already thinking about trying to pass signal through
trans-impedance amplifier to bring the shape closer to Gaussian.
On more thing I could think of is the decimation factor and its lower limit
- 4. Would it make sense to approach without decimation factor (1)?

Great thanks for your support.

Best regards,
[image: SCR.BMP]
Sergey

pavel-demin, 2020-05-28:
It seems that images cannot be sent by email. Could you please add it via the web interface?

abovyan, 2020-05-28:
It was because of image format. BMP is not accepted. I updated previous
post.

pavel-demin, 2020-05-28:
Thank you for the image. I can see the pulse shape now. I also put a copy of the image to this comment for better visibility.
[image]

pavel-demin, 2020-05-28:
The pulse shape looks OK. The currently implemented algorithm can measure the height of this type of pulse.

However, the pulse duration is too short. I think the rise time should be 10 times longer.

It would be interesting to check how this pulse is seen by the oscilloscope integrated into the MCPHA application. It would be great if you could send a .dat file.

I agree that removing the decimation could help manage the short pulse duration, but it'll also double the noise. The easiest way to remove the decimation is to just remove the CIC filter.

If it's a current pulse coming directly from a detector, I think passing it through a charge-sensitive amplifier and a pulse shaping amplifier would improve the noise performance of your system.

abovyan, 2020-06-01:
Dear Pavel, 

I attach three .dat files which I recorded by the oscilloscope integrated into the MCPHA. My impression is that oscilloscope actually sees the peak of the pulses correctly. Please find them attached.
About CIC filter, is it possible to remotely disable it?

Great thanks for your help.

[oscillogram_20200601_155001.zip](https://github.com/pavel-demin/red-pitaya-notes/files/4711148/oscillogram_20200601_155001.zip)

pavel-demin, 2020-06-01:
Thank you for the .dat files. The pulses look OK. The main problem is that they are bipolar. I'd try to analyze them with enabled `negative IN1` and `manual baseline mode`.

To disable the CIC filter, the only solution for the moment is to edit the source code and to rebuild `mcpha.bit`. Here is a link to a pre-built .bit file without CIC filter:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADO_iiCdWNY2mEW7afPHgaUa/mcpha/mcpha_no_cic.bit?dl=1

Just copy it to `apps/mcpha/mcpha.bit` and restart the MCPHA application on the Red Pitaya board.

closed 2021-06-22 by pavel-demin