pavel-demin/red-pitaya-notes #965: How to use GUN radio to successful connect the Red Pitaya?

friedrich1993322, 2020-06-15:
I installed your official APP "SDR transceiver". But when I run your example program, the program told me "socket.error: [Errno 111] Connection refused" (shown in Figure 1) The IP address is right (shown in Figure 2). Could you tell me, where is wrong. And how to modify it? 

### Description of the setup:

 * Device:Red Pitaya
 * SD card image:
 * Application:SDR transceiver
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1. install SDR transceiver
 2.run the example programm
 3.

pavel-demin, 2020-06-15:
Unfortunately, I do not see figures 1 and 2. It seems that there was a problem with attaching them to your comment.

I'd say that one step is missing in your list:
1. install SDR transceiver
2. run SDR transceiver
3. run the example program

Please check if the SDR transceiver application is running on the Red Pitaya board.

closed 2020-06-15 by pavel-demin

pavel-demin, 2020-06-15:
I received the screenshots by e-mail.

The port parameter of the Red Pitaya Source block is set to an incorrect value (5k).

The correct value for the port parameter is 1001.