pavel-demin/red-pitaya-notes #969: Features on MCPHA

bryanmhayes, 2020-07-23:
Hello Pavel. 

Your MCPHA code is very nice and well thought out. I work in a radiation physics laboratory and when we learned about the Red Pitaya we immediately saw the opportunity to make a program just like yours. Great job on this project.

I understand this is a free product, but some simple features on this program would make it comparable to code that companies charge major $$$$ for. These features include:

1) Dead Time + Live Time: Live time is the amount of time the MCPHA + RP are not processing signals. Dead time is quantified as the time that the RP is busy analyzing signals, and thus is not able to measure another signal in this time. The set time of each measurement should correspond the amount of live time that has passed. Knowing the dead time of each measurement allows the comparison of different measurements and lets the user know if their detector is being overwhelmed or not. This would also give insights to how powerful the RP is and what applications it would be best suited for (at particle accelerators, nuclear reactors, etc).

2) Variable Voltage Input Range: The HV jumper setting has a range of 20V, but the many spectroscopy amplifiers have a maximum output voltage of 10V. This means only half of the ADC values will be used in the HV jumper setting.

3) Bin Numbers: 16K bins is great, but sometimes is too many (especially for students). The option for 1024, 2048, 8192 bins would be great.

I have more things to add, but I'll keep them to myself unless you wish to hear them. Really I came here to tell you that this program is on the cusp of making the RP a very powerful tool. I very much admire what you have done and hope you continue to develop these great programs.

Cheers,
Bryan

pavel-demin, 2020-07-23:
Hello Bryan,

Thank you for your suggestions.

There is no dead time in the MCPHA application. I think most multichannel analyzers have zero dead time these days.

I don't see how point 2 can be addressed in software. It's a limitation of the Red Pitaya board. Normally, I use the LV jumper setting and reduce the amplification. Another solution could be to replace the 10 MOhm resistors R158 and R159 with 5 MOhm resistors. The input schematics can be found at [this link](https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/fastIO.html).

The MCPHA application only does the data acquisition. It doesn't have any data analysis functionality. I think analysis should be done by dedicated tools like, for example, [InterSpec](https://github.com/sandialabs/InterSpec). Normally, I do rebinning from 16k to a lower number of bins in my analysis scripts in [ROOT](https://root.cern.ch). I think [InterSpec](https://github.com/sandialabs/InterSpec) also does rebinning.

There are  several alternative control programs:
 - web interface by Will Thompson
    https://github.com/sefffal/redpitaya-multichannelanalyser
 - command line program that records timestamped pulse heights
    https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/mcpha/client/pha-start.c

The web interface by Will Thompson does rebinning.

Best regards,

Pavel

bryanmhayes, 2020-07-24:
Thank you for the reply Pavel.

> There is no dead time in the MCPHA application.

Can you elaborate? Are you really measuring every single pulse? I don't see how this is possible. There will invariably be losses, some of which come from the limitation of the software.

Thank you for the idea of replacing the 10 MOhm resistors at the RP inputs. I will pursue this and report back.

pavel-demin, 2020-07-24:
> Can you elaborate? Are you really measuring every single pulse? I don't see how this is possible. There will invariably be losses, some of which come from the limitation of the software.

The peak finding algorithm is running in the FPGA not in the software. The algorithm is very simple. The samples received from the ADC are processed one by one. They are smoothed by the CIC and FIR filters. The peak finding algorithm looks for minima and maxima of a smooth function. The derivative is approximated by the symmetric difference quotient (f'(x)=f(x+d)-f(x-d)). The peak height is calculated as maximum - minimum.

The only losses are due to overlapping peaks. As far as I know, these losses are not considered as dead time.

bryanmhayes, 2020-07-24:
I see. I understand now. Thank you.

closed 2020-07-24 by bryanmhayes