pavel-demin/red-pitaya-notes #970: Matlab program for VNA application on Redpitaya

MxQL, 2020-08-11:
Dear Pavel,

Some years ago, you published a Matlab code for controlling the Red Pitaya (https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/vna/matlab/vna.m). I wonder if it still works with the current SD card image on the STEM lab site. There is also a VNA application among those provided by STEM lab.

Thanks

Maxim

### Description of the setup:

 * Device:
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2020-08-11:
Dear Maxim,

The SD card images provided by the Red Pitaya developers come with a copy of an outdated version of my VNA application. A newer version of the VNA application can be installed via the application marketplace. The version installed via the application marketplace is compatible with the Matlab code.

Best regards,

Pavel

pavel-demin, 2020-08-11:
I've just checked the code of the VNA application in the Red Pitaya repository and I think that it is also compatible with the Matlab code.

MxQL, 2020-08-11:
Dear Pavel,

Thank you very much for your prompt response.

I am trying to translate the Matlab code in C. I managed to send the data
to the Red Pitaya but haven't received data back yet. That's why I asked my
question.

Please let the thread open, I may still have some questions.

Best regards

Maxim

pavel-demin, 2020-08-11:
There is a client code in C for the MCPHA application:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/mcpha/client/pha-start.c

The protocol is relatively similar to the protocol of the VNA application. You'll need to change the type of the `command` array to uint32_t and change the command codes.

MxQL, 2020-08-11:
Dear Pavel,

Thank you very much. This will help me a lot.

Best regards

Maxim

MxQL, 2020-08-16:
The Matlab code works with your old version of SD image, but not with the
new one on the market place. There may be some little changes.

I managed to make a C code that runs with the old version. I would be very
interested to make it working with the latest version of SD image.

Thank you very much for your support.

Maxim

pavel-demin, 2020-08-16:
This is strange.

Here are the links to the server code for both versions:
 - new version:
   https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/vna/server/vna.c
 - old version:
   https://raw.githubusercontent.com/RedPitaya/RedPitaya/master/apps-free/stemlab_vna/server/vna.c

Here is the difference between two versions:
```
# diff vna.c.old vna.c
186c186
<             freq = start + (stop - start) * i / (size - 1);
---
>             freq = i > 0 ? start + (stop - start) * i / (size - 1) : start;
```

So the server code is almost the same in both versions. The only difference is the division by zero bug fixed in the new version.

If the Matlab code works with the old version, I can't see why it doesn't work with the new version. The protocol is exactly the same in both versions.

If you send me your C code I could try to figure out why it is not working with the new version.

MxQL, 2020-08-28:
Dear Pavel,

Attached is the file in C for interfacing with the old version VNA. I hope
you will find the difference.

Another question: I am using this old version of SD card image of the VNA
app. I cannot see the Wifi hotspot network of the Redpitaya with IP address
of 192.168.42.1. Maybe because the old version doesn't have this feature?

Thanks

Maxim
  client.c
<https://drive.google.com/file/d/1bSKCygxCe-jzfZx4J6P-6ROyPV5iYei1/view?usp=drivesdk>

pavel-demin, 2020-08-28:
> Attached is the file in C for interfacing with the old version VNA.

I can't access the file.

> I am using this old version of SD card image of the VNA app.

Please remind me what exactly SD card image you're using.

MxQL, 2020-08-28:
Sorry for the access issue. I send you again the client.c file, and the
VNA.c file of the SD card image I am using. I can't recall the exact
version.

Best regards

Maxim

closed 2020-12-14 by pavel-demin