pavel-demin/red-pitaya-notes #971: sdr_transceiver_wspr_122_88

f4goh, 2020-08-15:
### Description of the setup:

 * Device: Redpitaya
 * SD card image:https://github.com/pavel-demin/red-pitaya-notes/releases/download/20200628/red-pitaya-alpine-3.12-armv7-20200628.zip
 * Application: sdr_transceiver_wspr_122_88
 * Other relevant information:

### Description of the problem:
what is the step by step commands from start to configure WSPR receiver and to start it automaticly.
Sorry, but your description is confused and some explanation about the exact order of the command lines must be clarified.
```
login as: root
root@rp-f04391.local's password:
Welcome to Alpine!
The Alpine Wiki contains a large amount of how-to guides and general
information about administrating Alpine systems.
See <http://wiki.alpinelinux.org/>.
You can setup the system with the command: setup-alpine

You may change this message by editing /etc/motd.

rp-f04391:~# please complete command line step by step
rp-f04391:~#
rp-f04391:~#
rp-f04391:~#
rp-f04391:~#
rp-f04391:~#
rp-f04391:~#
rp-f04391:~#
```

Thanks for help

pavel-demin, 2020-08-15:
I don't know if I understand what is confusing for you. So, please explain in detail what is confusing and how you propose to improve it.

All the steps in my notes can be done before inserting the micro SD card into the Red Pitaya board. So there is no need to do anything from the command line.

Since you know how to get to the command line, you are an advanced user and in my opinion you don't need any instructions.

My notes show some examples and indicate which files need to be modified/copied/started. The exact sequence is not important. All the steps are optional and can be done in any order.

Some useful commands can be found at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine/). This link can be found in all the getting started instructions.

For example, if you want to change the `CALL` and `GRID` variables in [decode-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_transceiver_wspr_122_88/app/decode-wspr.sh#L4-L5), then it can be done with the following commands:
```
# make SD card writable
rw

# edit decode-wspr.sh
nano apps/sdr_transceiver_wspr_122_88/decode-wspr.sh

# make SD card read-only
ro
```

f4goh, 2020-08-15:
OK to configure it before inserting card, this is the only thing i understand.
When you say "Optionally, to start the application automatically at boot time, copy its start.sh file from apps/sdr_transceiver_wspr to the topmost directory on the SD card."
Is it on root directory ?
Because i don't understand how to run wspr decoding.
because when i launch putty  when  sd card in inserted the file start.sh doesn't does not appear in root directory.
How i can check wspr decoder was running ?
Does Antenna must be on 1st HF input ?
Thanks for answers.

pavel-demin, 2020-08-15:
> Is it on root directory ?

No, the root (`/`) directory and the topmost directory on the SD card are two different directories. When you're connected to the Red Pitaya command line, the path to the topmost directory on the SD card is `/media/mmcblk0p1`.

If you want to copy `start.sh` to the topmost directory on the SD card using the command line, then this can be done with the following commands:
```
# make SD card writable
rw

# copy start.sh
cp apps/sdr_transceiver_wspr_122_88/start.sh /media/mmcblk0p1/

# make SD card read-only
ro
```
> How i can check wspr decoder was running ?

When the decoder is running, the decoded data is accumulated in `/dev/shm/ALL_WSPR.TXT`. So if the file is present, the decoder is running.

> Does Antenna must be on 1st HF input ?

With default settings all the receivers are connected to IN1. They can be connected to IN2 by setting `chan` to 2 in [write-c2-files.cfg](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr_122_88/app/write-c2-files.cfg).

f4goh, 2020-08-16:
Hello,
thanks for answer. I try it.
'''
rw

# edit decode-wspr.sh
nano apps/sdr_transceiver_wspr_122_88/decode-wspr.sh
cp apps/sdr_transceiver_wspr_122_88/start.sh /media/mmcblk0p1/

# make SD card read-only
ro
'''
Reboot

when i take a look to this directory, il found decode-wspr.log     write-c2-files.cfg
'''
rp-f04391:~# ls /dev/shm/
decode-wspr.log     write-c2-files.cfg
rp-f04391:~#
'''
then after about 1 minute. the ssh connection crash and when i ping redpitaya the destination host unreachable.
i deconnect and reconnect power supply. ping work. i connect with ssh, and after 1 minute ssh crash again.
So i think WSPR spot can be done.
I never see ALL_WSPR.TXT in /dev/shm/ directory.

What is the difference between sdr_transceiver_wspr_122_88 and sdr_transceiver_wspr directory ?

If i retry the installation from the begining under windows
I add my callsign and the locator in the decode-wspr.sh file with notepad++, ok.
and what is the next operation ?
Perhaps copy start.sh with Windows explorer ... and what is the destination ? or something else ?

Sorry for the inconvenience.

pavel-demin, 2020-08-16:
When you opened this issue, you did not provide the type of Red Pitaya board you are using. Since you tried using sdr_transceiver_wspr_122_88 I thought your board was STEMlab 122.88-16 SDR and added it to your description of the setup.

If your board stops working when running sdr_transceiver_wspr_122_88, it most probably means that your board is not STEMlab 122.88-16 SDR.

So please indicate the type of Red Pitaya board you are using and change the description in the first post.

To find the instructions for your board, go to the list of my notes at [this link](http://pavel-demin.github.io/red-pitaya-notes/) and select the link for the application you want to use in the section corresponding to your board.

I have the impression that your confusion is due to your expectation of very complicated instructions and the simplicity of the instructions confuses you.

Please don't try to find overly complicated steps hidden in my instructions. There are not any. If it is written "edit file", just edit it. Notepad++ is a very good editor for this. If it is written "copy a file", just copy it with any tool that works for you.

> What is the difference between sdr_transceiver_wspr_122_88 and sdr_transceiver_wspr directory ?

sdr_transceiver_wspr_122_88 is for Red Pitaya STEMlab 122.88-16 SDR
sdr_transceiver_wspr is for for Red Pitaya STEMlab 125-14

> Perhaps copy start.sh with Windows explorer ... and what is the destination ?

To the topmost directory on the SD card (exactly as it's written if the getting started instructions). In Windows this should be something like Z: or whatever letter is assigned to your SD card.

> you say that "Applications can be started from the web interface". but how ?
> just click on application menu and if i see something like "The multiband FT8 transceiver is ready".
"ready" means "running" ?

Yes, clicking on the application name is enough to start it.

I don't think that "running" is the correct word to describe the status of the multiband FT8 transceiver. It is not running all the time. It is started at zero seconds every minute by cron.

pavel-demin, 2020-08-16:
> Ps : i have the STEMlab 125-10 V1.0 may be not compatible ?

OK. Thank you for this information. I've just updated your first post.

pavel-demin, 2020-08-16:
The correct application for STEMlab 125-10 is sdr_transceiver_wspr. The corresponding notes are at [this link](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wspr/).

I don't have STEMlab 125-10 but I have seen reports that most of my applications work with this card.

f4goh, 2020-08-16:
Thanks for anwsers,

But after launch sdr_transceiver_wspr in web browser. After 2 minutes STEMlab 125-10 crash. ssh closed. redpitaya Web browser not respond. The only way is to power off, and power on redpitaya, but this is the same.

pavel-demin, 2020-08-16:
Please send me a screenshot of the web interface. It's possible that the web interface in the latest release ([20200628](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20200628)) doesn't work correctly on the STEMlab 125-10 board.

I can think of two possible workarounds:
 - try the previous release ([20191120](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20191120))
 - try to start sdr_transceiver_wspr from the Red Pitaya command line using the following command:
```
~/apps/sdr_transceiver_wspr/start.sh
```

f4goh, 2020-08-16:
The previous release (20191120) works with STEMlab 125-10. 
No crash since 30 minutes. 
I let power on STEMlab 125-10 all the night.
Reports are ok to Wsprnet.
the configuration is easy once you understand and there is no crash.
I post the screenshot tomorrow because i delete 20200628 and replace it by 20191120.
I need to create a new msdcard with 20200628.

Thanks for help.

```
rp-f04391:~# ls /dev/shm/
decode-wspr.log     write-c2-files.cfg  wspr_timer.out
hashtable.txt       wspr_spots.txt      wspr_wisdom.dat
rp-f04391:~#

rp-f04391:~# cat /dev/shm/ALL_WSPR.TXT
200816 1818   1 -27 -3.09   1.8381312  EO4RFA ML74 33          0  1449  -32    1 -155
200816 1818   2 -23  0.32   7.0400210  DL2ALE JO50 37          4     1    0    1  421
200816 1818   2 -25  0.28   7.0400264  M5TXJ IO84 23           0     1    0    1   84
200816 1818   6  -5  0.28   7.0400458  G4CAO IO91 27           2     1    0    1  659
200816 1818   4 -20  0.24   7.0400778  OZ7IT JO65 37           2     1    0    1  497
200816 1818   4 -14 -0.23   7.0401294  GM3RVL IO85 20          3     1    0    1  360
200816 1818   6 -14 -0.10   7.0401452  EA4URA IN80 27          3     1    0    1  608
200816 1818   5 -11  0.45   7.0401620  M0GUC IO82 23           3     1    0    1  491
200816 1818   6 -12  0.45   7.0401823  DL1BGE JO43 23          2     1    0    1  605
200816 1818   6 -14  0.24  10.1401526  G0MBA JO01 23           4     1    0    1  512
200816 1818   2 -25  0.32  10.1401637  PA0RBC JO22 10          4     2    0    1   26
200816 1818   6 -16  0.28  10.1402871  OZ7IT JO65 37           4     1    0    1  667
200816 1818   2 -26  0.45  10.1403097  PA0SLT JO33 20          4     1    0    1  208
200816 1818   3 -21  0.11  14.0970516  PD0KT JO33 10           4     2    0    1  104
200816 1818   2 -25  0.28  14.0971524  MM/M0VIK 23             4     1    0    1  104

rp-f04391:~# cat /dev/shm/decode-wspr.log
Sun Aug 16 18:17:01 UTC 2020
Sleeping ...
Sun Aug 16 18:18:00 UTC 2020
Recording ...
Sun Aug 16 18:19:01 UTC 2020
Sleeping ...
Decoding ...
Sun Aug 16 18:20:00 UTC 2020
Recording ...
1818 -27 -3.1   1.838131  0  EO4RFA ML74 33
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
1818 -23  0.3   7.040021  4  DL2ALE JO50 37
1818 -25  0.3   7.040026  0  M5TXJ IO84 23
1818  -5  0.3   7.040046  2  G4CAO IO91 27
1818 -20  0.2   7.040078  2  OZ7IT JO65 37
1818 -14 -0.2   7.040129  3  GM3RVL IO85 20
1818 -14 -0.1   7.040145  3  EA4URA IN80 27
1818 -11  0.5   7.040162  3  M0GUC IO82 23
1818 -12  0.5   7.040182  2  DL1BGE JO43 23
<DecodeFinished>
<DecodeFinished>
1818 -21  0.1  14.097052  4  PD0KT JO33 10
1818 -25  0.3  14.097152  4  MM/M0VIK 23
<DecodeFinished>
<DecodeFinished>
1818 -14  0.2  10.140153  4  G0MBA JO01 23
1818 -25  0.3  10.140164  4  PA0RBC JO22 10
1818 -16  0.3  10.140287  4  OZ7IT JO65 37
1818 -26  0.5  10.140310  4  PA0SLT JO33 20
<DecodeFinished>
Uploading ...
Sun Aug 16 18:21:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
1820 -20  0.2  10.140123  2  DL0PBS JO33 23
1820  -6  0.5  10.140140  2  DK8JP JO31 30
1820 -26  0.2  10.140158  1  PA0RBC JO22 10
1820 -15 -0.2  10.140168  2  DL6NL JO50 20
1820  -5  0.2  10.140193  2  DJ2TS JO40 30
1820 -11  0.2  10.140202  3  ON7KB JO21 37
1820 -14  0.2  10.140208  3  PA3DSC JO21 30
1820   2  0.2  10.140219  3  ON7KB JO21 37
1820 -17  0.2  10.140222  0  PA0SLT/2 20
1820  -6  0.2  10.140246  2  DL4MFC JN58 37
1820 -17  0.7  10.140277  0  PA0WV JO22 37
<DecodeFinished>
Sun Aug 16 18:22:00 UTC 2020
Recording ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
1820 -11  0.2   3.570056  0  <...> JN48SA 33
1820  -8  0.2   3.570104  1  GW1CLH IO81 33
1820 -20  0.3   3.570106  0  DC4CP JO30 30
<DecodeFinished>
1820 -20  0.3   7.040047  0  OZ1KVB JO44 27
1820 -18  0.4   7.040056  1  DK2DB JN48 37
1820 -27  0.2   7.040066  0  DK9MS JO40 23
1820 -16  0.2   7.040098  0  <...> KN87UX 37
1820   6  0.3   7.040114  2  F4FVW JN29 33
1820 -20 -0.4   7.040132  2  GM3RVL IO85 20
1820 -17 -0.3   7.040168  2  5P2A JO56 20
1820 -26  1.3   7.040190  1  G7UJY IO93 23
<DecodeFinished>
1820 -25  1.6  14.097047  2  DJ2NL JO32 23
1820 -16  1.2  14.097090  3  DL6NL JO50 27
1820 -22  1.7  14.097112  4  DB9OH JO52 23
1820 -21 -1.0  14.097123  3  G0EKQ IO83 37
1820 -22  2.4  14.097163  3  EI8ETB IO51 20
1820 -22  0.2  14.097179  3  RA9WE LO74 37
<DecodeFinished>
Uploading ...
rp-f04391:~#
```

f4goh, 2020-08-17:
here is the  screenshot 
[image]

pavel-demin, 2020-08-17:
Thank you for the screenshot. The web interface is OK and it starts the correct application.

I cannot reproduce this problem on STEMlab 125-14.

It is still not very clear to me which part of the application is causing this problem on STEMlab 125-10.

Could you do the following test:
```
cd /dev/shm

/media/mmcblk0p1/apps/stop.sh

cat /media/mmcblk0p1/apps/sdr_transceiver_wspr/sdr_transceiver_wspr.bit > /dev/xdevcfg

/media/mmcblk0p1/apps/sdr_transceiver_wspr/write-c2-files /media/mmcblk0p1/apps/sdr_transceiver_wspr/write-c2-files.cfg
```

The last command reads data from the FPGA and writes it to .c2 files. It is possible that this command is causing the problem. However, if it works, it will run for two minutes and output .c2 files.

f4goh, 2020-08-18:
```
rp-f04391:~# cd /dev/shm
rp-f04391:/dev/shm# /media/mmcblk0p1/apps/stop.sh
ip: can't find device 'mvl0'
rp-f04391:/dev/shm# cat /media/mmcblk0p1/apps/sdr_transceiver_wspr/sdr_transceiv
er_wspr.bit > /dev/xdevcfg
rp-f04391:/dev/shm# /media/mmcblk0p1/apps/sdr_transceiver_wspr/write-c2-files /m
edia/mmcblk0p1/apps/sdr_transceiver_wspr/write-c2-files.cfg
rp-f04391:/dev/shm# ls
wspr_0_1836599_1_200818_0700.c2   wspr_4_10138699_1_200818_0700.c2
wspr_1_3568599_1_200818_0700.c2   wspr_5_14095599_1_200818_0700.c2
wspr_2_5287199_1_200818_0700.c2   wspr_6_18104599_1_200818_0700.c2
wspr_3_7038599_1_200818_0700.c2   wspr_7_21094599_1_200818_0700.c2
rp-f04391:/dev/shm#
```
No crash

pavel-demin, 2020-08-18:
Thank you for the test.

So it is not `write-c2-files` that crashes in the latest version ([20200628](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20200628)). It is good news.

The only remaining component is the WSPR decoder. I think the easiest test would be to comment out the lines that are running the decoder and check if the system hangs or not.

Could you please do the following test with the [20200628](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20200628) SD card image?
- comment out [lines 44-51 in apps/sdr_transceiver_wspr/decode-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/decode-wspr.sh#L44-L51)
- start the WSPR transceiver application via the web interface
- check if the system hangs after two minutes

pavel-demin, 2020-08-18:
One more question. Does the system hangs while running the FT8 transceiver application from the [20200628](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20200628) SD card image?

f4goh, 2020-08-18:
Hello Pavel,

I comment these lines in decode-wspr.sh
```
#for file in wspr_*_$TIMESTAMP.c2
#do
#  while [ `pgrep $DECODER | wc -l` -ge $JOBS ]
#  do
#    sleep 1
#  done
#  nice -n $NICE $DECODER -JC 5000 $file &
#done

```
and show decode-wspr.log
```
rp-f04391:~# cat /dev/shm/decode-wspr.log
Tue Aug 18 19:11:01 UTC 2020
Sleeping ...
Tue Aug 18 19:12:00 UTC 2020
Recording ...
Tue Aug 18 19:13:01 UTC 2020
Sleeping ...
Decoding ...
Tue Aug 18 19:14:00 UTC 2020
Recording ...
```
Seem no crash for wspr with line in comment.

For ft8, crash after 2 minutes.
73

pavel-demin, 2020-08-18:
I have checked the memory usage of the old and new version of the WSPR decoder and it looks like the new version is consuming less memory:
  - new
```
  Maximum resident set size (kbytes): 7748
```
  - old
```
  Maximum resident set size (kbytes): 8688
```

pavel-demin, 2020-08-18:
I'm almost out of ideas.

Could you try to uninstall some packages from the 20200628 SD card image, uncomment the previously commented lines in decode-wspr.sh and run the WSPR transceiver application?

I think the following commands should uninstall some larger packages:
```
apk del fftw-* alsa-utils
lbu commit -d 
reboot
```
Since all packages are installed on a RAM disk, uninstalling some packages should free up RAM.

f4goh, 2020-08-19:
Hello Pavel,
Now WSPR works 
```
rp-f04391:~# ls /dev/shm
ALL_WSPR.TXT                      wspr_1_3568599_1_200819_0822.c2   wspr_6_18104599_1_200819_0822.c2
decode-wspr.log                   wspr_2_5287199_1_200819_0822.c2   wspr_7_21094599_1_200819_0822.c2
hashtable.txt                     wspr_3_7038599_1_200819_0822.c2   wspr_spots.txt
write-c2-files.cfg                wspr_4_10138699_1_200819_0822.c2  wspr_timer.out
wspr_0_1836599_1_200819_0822.c2   wspr_5_14095599_1_200819_0822.c2
rp-f04391:~#
```
but not ft8 : system hangs
73

pavel-demin, 2020-08-19:
Thank you for this new test.

So the 125-10 board does not have enough RAM to work with the latest version of the SD card image.

If the 20191120 version of the SD card image works, then I think the only solution is to use the 20191120 version of the SD card image.

f4goh, 2020-08-19:
Thanks for help. 
Perhaps a short note on http://pavel-demin.github.io/red-pitaya-notes/
for STEMlab 125-10 V1.0 users shouls be welome with 20191120 link.
73.

pavel-demin, 2020-08-19:
I prefer not to mention the 125-10 board in my notes. Otherwise some people would misunderstand that my applications are designed for the 125-10 board. However, this is not the case. I only have applications for 125-14 and 122.88-16 boards.

closed 2020-08-19 by pavel-demin