pavel-demin/red-pitaya-notes #972: ADC recorder trigger C code not running

jesusabilopez, 2020-08-19:
### Description of the setup:

 * Device: Red Pitaya 125-14
 * SD card image: red-pitaya-alpine-3.6-armhf-20180523
 * Application: adc_recorder_trigger
 * Other relevant information:

### Description of the problem:

Hi Pavel,

I've started to work with your adc recorder project, I've added a test signal using a DDS compiler and your DAC IP.

I've loaded the bitstream to the board and compiled the C code using the _" gcc -o adc_recorder_trigger adc_recorder_trigger.c "_ command, and the type the _"./adc_recorder_trigger"_ command, but the promt just does nothing, does the C code requires any input arguments? 

Also, I would like to know what's the extension port used for in the block desing, do I need aditional hardware to this project?

Thanks in advance

pavel-demin, 2020-08-19:
It's normal that it doesn't do anything. It waits for a trigger edge on the DIO0_P pin.

jesusabilopez, 2020-08-20:
Ohhh, so I need to use a push button or something like that to get it started?

pavel-demin, 2020-08-20:
Yes, a push button can be used for a very basic test. If you don't need a trigger, there's adc_recorder.