pavel-demin/red-pitaya-notes #973: SDR Transceiver Embedded

little-tesla, 2020-08-30:
Dear Pavel

First of all I wanted to say thank you for spending your time and developing such a great software ecosystem around the RedPitaya board!
I'm currently trying out your sdr_transceiver_emb project and have a working system. When analyzing the code I noticed that the calculated FFT from the FPGA isn't used at all. I would like to try and change the software to use the calculated FFT from the FPGA rather then software and was wondering how this has to be done. Especially the interface and control to the averager isn't clear to me since my verilog knowledge is very basic. Could you maybe give me a few hints regarding how this should be done?
A summary of my interpretation is as follows:

sp_rst --> Reset spectrum module, similar as for RX/TX
fft_data --> Spectrum data, are they float or uint32? Am I correct that data is the sum of the FFT bins?
sp_cntr --> FFT data counter, similar as RX/TX
sp_rate --> FFT CIC rate 125 - 1250 (500 - 50kSps)
sp_phase --> FFT frequency, similar as RX/TX

Averager
sp_pre -->  Samples to skip before averaging?
sp_tot --> Total samples to average?

Any further information or even some sample code is greatly appreciated.

Many thanks
Dimce

pavel-demin, 2020-08-30:
Dear Dimce,

Thank you for your interest in the sdr_transceiver_emb project. It would be great if you could use the spectrum calculation.

This part of the FPGA configuration was taken from the [sdr_transceiver_fft](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_transceiver_fft) project. The corresponding [server code](https://github.com/pavel-demin/MiniTRX/blob/master/server/server.cpp) can be found in the MiniTRX project. This combination worked and I had a spectrum displayed on the screen.

The last time I tested the spectrum calculation code in the sdr_transceiver_emb project, it didn't work. Then I fixed a bug (wrong const_1 value) with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/f582e6fc387497b7b37dc959b3447cd854c7009b#diff-51ff95e4247f331a1c6be948a036bbca) but did not test if the spectrum calculation started working after this fix.

> sp_rst --> Reset spectrum module, similar as for RX/TX

The spectrum module shares the reset register with the RX module. The SP module uses the second and third bits.

> fft_data --> Spectrum data, are they float or uint32? Am I correct that data is the sum of the FFT bins (referring to the averager) ?

They are int32.

The idea is to average each of the bins over several FFT results. It's calculated as sum(ln(re^2+im^2))). To convert it to dB, it should be multiplied by 5*log10(exp(1))/N or something like this.

There is also win_data that should be loaded with a windowing function.

> Averager
> sp_pre --> Samples to skip before averaging?
> sp_tot --> Total samples to average?

I think sp_pre is the number of FFT bins minus one (511) and sp_tot is the number of FFT bins multiplied by the number of values to average minus one (511, 1023, 2047, etc).

Best regards,

Pavel

little-tesla, 2020-08-31:
Dear Pavel,

Thank you very much for your valuable inputs!
They help a lot and I'll give it a try soon.

Best regards,

Dimce

little-tesla, 2020-09-01:
Dear Pavel,

I have the source code for testing the FFT almost ready, but a few more things came up which are not fully clear to me:

To calculate the relative power I would have expected something like this:
bin_n= r^2 + i^2
sum_n =sum(bin_n) --> averager
P_dB = 10*log10(sum_n/n) --> in software

Current implementation is:
bin_n = ln(re^2+im^2) --> FFT (I red somewhere the ln() is used to "compress" the range)
sum_n=sum(bin_n) --> averager

p_dB=sum_n*5*log10(exp(1))/N --> in software, as you suggested
Should this not be more like this?
p_dB=10*log10(exp(sum_n/N))

Am I right that the fft_data output buffer is 1024Byte?

Best regards,

Dimce

pavel-demin, 2020-09-01:
I agree, the logarithm calculation needs to be moved after the averager. Then the result will be ln(sum(bin_n)).

ln(sum(bin_n)) = ln(average*N) = ln(average) + ln(N) = log10(average)/log10(exp(1)) + ln(N)

log10(average) = (ln(sum(bin_n)) - ln(N))*log10(exp(1))

P_dB = 10*log10(average) = 10*(ln(sum(bin_n)) - ln(N))*log10(exp(1))

little-tesla, 2020-09-02:
Dear Pavel,

Thanks for the explanation! Now it's clear to me why ln() is used in the calculation.
I tried the FFT for quite some time and I seem not to be able to get the FFT outputting any data.
Steps I did:
1. Load FPGA with bitstream (cat sdr_transceiver_emb.bit > /dev/xdevcfg, blue LED on)
2. Replace sdr-transceiver-emb.c with the simple attached c-program and compile
3. Run the program

Main observation is that the sp_cntr doesn't seem to be increased in any way. fft_data is always zero. I tried different reset orders of the fft with no luck.

Is there something obviously wrong I'm doing or is still something wrong with the fft module? What problem did you experience when you did your testing?

[sdr-transceiver-emb.txt](https://github.com/pavel-demin/red-pitaya-notes/files/5164831/sdr-transceiver-emb.txt)

pavel-demin, 2020-09-02:
I've just checked the code in [sp.tcl](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_emb/sp.tcl) and found a few bugs. It will take me some time to fix them and test the code. I'll let you know when it works.

F1EYG, 2020-11-24:
Dear Pavel 
 I have several questions about the StemLab. It is more powerful than the pitaya therefore:
- do you think it would be possible to run two RxSDRs, one on each ADC with two IPs (ethernet and Wifi for example or with an additional ethernet dongle?).
- do you think it would be possible to recompile your FT8 decoder in deep3 with a limited number of bands (8 for example instead of 11)?
- sparkSDR is limited to 8 bands with the pitaya but also with the stemlab where does this limitation come from?
Tks again for your application :-)
73
F1EYG

pavel-demin, 2020-11-24:
> do you think it would be possible to run two RxSDRs, one on each ADC with two IPs (ethernet and Wifi for example or with an additional ethernet dongle?).

It's already the case. It's possible to run two instances of [SDR receiver compatible with HPSDR](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver-hpsdr-122-88/) with two different IP addresses.

> do you think it would be possible to recompile your FT8 decoder in deep3 with a limited number of bands (8 for example instead of 11)?

The CPU on the Red Pitaya boards isn't powerful enough to run the FT8 decoder with higher depth settings. To run  the FT8 decoder with higher depth settings, I'd suggest running it on a computer with a more powerful CPU.

> sparkSDR is limited to 8 bands with the pitaya but also with the stemlab where does this limitation come from?

No idea.

If you have any other questions, please create a new issue. Please don't post questions in unrelated issues.

F1EYG, 2020-11-25:
Tks Pavel
Sorry for the mistake on GitHub
OK i am sorry i do not read completely the note for StemLab (i use both ;-)
I starting new test from beginning , i hope share StemLab between one
SkimmerServer and One SparkSDR for FT4
tks for your answer
73 Sylvain
F1EYG

pavel-demin, 2021-03-14:
The spectrum module is fixed with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/62a175ffc009b511d0c140bdcde596dc8d71a721).

Here is an example of its output:
[image]

pavel-demin, 2021-03-26:
The fixed version of the spectrum calculations is now in the [this release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20210326).

closed 2021-03-26 by pavel-demin