pavel-demin/red-pitaya-notes #975: Question: how to address the 8th receiver frequency?

f4exb, 2020-09-05:
Hello,

I am currently developing support in [SDRangel](https://github.com/f4exb/sdrangel) for the HPSDR/Metis emulation in the Red Pitaya developed here. [This issue refers](https://github.com/f4exb/sdrangel/issues/344). I see that the [SDR receiver compatible with HPSDR](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver-hpsdr-122-88/) has 2 x 8 receivers in two separate virtual HPSDR devices. I would like to support all 8 receivers however the [offciial documentation](https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Documentation/USB_protocol_V1.58.doc) for the command protocol has provision to control the frequency of just the 7 first receivers.

So I am wondering what is the command index to change the frequency of the 8th receiver. The [Hermes-Lite2 extended protocol](https://github.com/softerhardware/Hermes-Lite2/wiki/Protocol) has provision to address 12 receivers. I tried to address the 8th receiver accordingly using command 0x12 in bits [6:1] of C0 but this does not seem to work.

Can you please help me with the command corresponding to your implementation of frequency control of the 8th receiver?

pavel-demin, 2020-09-05:
It's the same command as in the HL2 protocol. The corresponding code can be found on [lines 271-277 in sdr-receiver-hpsdr.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_receiver_hpsdr_122_88/server/sdr-receiver-hpsdr.c#L271-L277).

f4exb, 2020-09-05:
OK thanks for your prompt answer so 0x12 (times 2 is 36 decimal) should work and my issue is elsewhere.

f4exb, 2020-09-08:
Issue was on my side with a wrong loop boundary missing command 36 in the "round robin". Thanks for confirming the command code anyway. This issue can be closed.

closed 2020-09-08 by f4exb