pavel-demin/red-pitaya-notes #976: LED0 no longer flashing on startup

w1njc, 2020-09-10:
Hi Pavel,

I noticed that on the latest release (red-pitaya-alpine-3.12-armv7-20200628.zip) LED0 no longer flashes on power-up while the system is loading.  On the previous release (red-pitaya-alpine-3.9-armv7-20191120.zip) LED0 would flash and then go dark when loading was complete.  I have integrated this into my project as a front panel indicator to show when the system is ready.  

So my questions are:  Why does it no longer flash, and is there a way to bring this "feature" back?

Thanks,
Nick W1NJC

pavel-demin, 2020-09-13:
Hi Nick,

> Why does it no longer flash

In previous versions, the FPGA was configured with the LED blinker configuration before booting Linux. In the new version, the FPGA remains unconfigured until an application is launched via the web interface or a start.sh script.

This change was necessary to make the SD card image compatible with both STEMlab 125-14 and STEMlab 122.88-16 SDR.

> is there a way to bring this "feature" back?

A possible workaround would be to configure the FPGA from u-boot. For example, this can be done by adding the following lines to `common/main.c` of u-boot:
```
       if (zynq_slcr_get_idcode() == 7)
               run_command("load mmc 0 0x2000000 apps/led_blinker_122_88/led_blinker_122_88.bit && fpga loadb 0 0x2000000 ${filesize}", 0);
       else
               run_command("load mmc 0 0x2000000 apps/led_blinker/led_blinker.bit && fpga loadb 0 0x2000000 ${filesize}", 0);
```

Cheers,

Pavel

w1njc, 2020-09-14:
Pavel,

Thanks for the response!

Nick

startup (#976)

closed 2021-06-22 by pavel-demin