pavel-demin/red-pitaya-notes #977: Question about the PLL clock generator

vahishta31, 2020-09-18:
Hello,

I am wondering why the clock generated for the ddr_clk in projects like radar, are -90 phase shifted? and how can one make assure that the DAC and ADC are completely synchronized and the clock jitter is at the lowest?

thanks in advance.

pavel-demin, 2020-09-18:
If you see a phase shift of -90 degrees, then you're looking at an old version.

In the current version, ddr_clk is phase shifted by -112.5 degrees and wrt_clk is phase shifted by -67.5 degrees.

These clocks are configured this way to meet [AD9767](https://www.analog.com/en/products/ad9767.html) timing requirements. Timing specifications are shown in figure 64 on page 25 in the [AD9767](https://www.analog.com/en/products/ad9767.html) data sheet.

vahishta31, 2020-09-18:
thank you. This was very helpful.

closed 2020-09-18 by pavel-demin

vahishta31, 2020-09-18:
Is this the same as the model DAC1401D125? I had this DAC as RP's DAC in mind.

pavel-demin, 2020-09-18:
Yes, they are very similar.

BTW. The clk_wiz module has an option to minimize the output clock jitter:
```
JITTER_SEL Min_O_Jitter
```

According to the Clocking Wizard summary, the peak-to-peak jitter is expected to be around 80-90 ps.

vahishta31, 2020-09-18:
I am using the RP inside an specific radar concept and the Rx and Tx samples need to be absolutely sync. I decided to compare my block design with your radar design and I realized this phase shift that you have chosen. With my design I am getting 40 dB SNR for a single target, which could be much better, but it seems that there are lots of spurious effects in my signal which could be because of the clock configurations and I am not sure if the PLL would make it worse or better compared to the case where I don't use one.

pavel-demin, 2020-09-18:
Have a look at the [VNA project](http://pavel-demin.github.io/red-pitaya-notes/vna/). Its RX and TX parts are synchronized.

vahishta31, 2020-09-18:
that's great to know. thanks for your time.