pavel-demin/red-pitaya-notes #978: Difficulties with STEMLab 125-14 ADC Format

TestingAccount4, 2020-09-22:
### Description of the setup:

 * Device: Red Pitaya Stemlab 125-14
 * SD card image: 0.98-696
 * Application: Custom (see attached)
 * Other relevant information: Project based on Anton Potočnik's Frequency Counter project.

### Description of the problem:
Hi Pavel,

First of all I'd like to say thank you so much for everything you've done with the Red Pitaya, it has made learning FPGA programming and using the Red Pitaya tremendously easier. I'm reaching out with this problem since I have tried and double checked everything all possible solutions that came to mind, and things still aren't working as we hoped- if this is not an appropriate place for this question please let me know.

My custom project is built off of Anton Potočnik's frequency counter project. The idea is to sample a signal from a superconducting nanowire single photon detector (SNSPD) at 125 MHz, and to detect the number of pulses that occur in one minute. The code I wrote and included in the project (shown below) takes in user defined threshold values, and records the number of times the first recorded data point is below the threshold, and the second recorded data point (sampled after the first data point) is above the threshold. I have tested this algorithm using SNSPD data, and it works as expected.

I believe the problem we are having has to do with the way a misunderstanding about the ADC's data format. Following the ADC data sheet present on your website, I typically set the variables "RP_Thresh_1" and "RP_Thresh_2" to have what I believe to be a value of 0.05V and 0.2 V, respectively. My thought process for obtaining a value that corresponds to 0.05 V is as follows:

    Desired Value=0.05 V
    (0.05 V)/(0.000122 V)  ≈ 410

    Convert 410 to binary (110011010)
    Pad the binary result to be 14 bits wide: 00000110011010

    Since 0.05V is positive, set the MSB equal to one: 10000110011010

However, when I input the values according to the above procedure by using the monitor command, I consistently observe zero peaks being recorded, although these thresholds worked perfectly on recorded SNSPD data w/ a python script.

### Steps to reproduce the problem:
 1. Configure the program as follows (see screenshot below). I can find a way to generate a TCL script if needed.
 2. Generate bitstream/copy to SD card/ssh into the RP/execute "cat example.bit > /dev/xdevcfg"
 3. Set the threshold values appropriately using the process described above.
 4. Read the number of recorded peaks after sixty seconds has passed.
[SNSPD_Output_Counter.txt](https://github.com/pavel-demin/red-pitaya-notes/files/5260272/SNSPD_Output_Counter.txt)
[image]
[image]
[image]

pavel-demin, 2020-09-22:
> Since 0.05V is positive, set the MSB equal to one: 10000110011010

The ADC interface module from this repository outputs positive 16-bit values for positive input voltages.

So if you need a threshold value of 409, just set it to 409.

closed 2020-09-22 by pavel-demin

TestingAccount4, 2020-10-29:
Hi Pavel,

Sorry to re-open this, but I tried what you did and I still don't understand the format. I modified another one of Anton's projects so that the data acquisition goes straight to GPIO, where I read the values coming out of the data acquisition module (see below
[image]
). When I observe those values via the "monitor" command (output in hexadecimal), convert them to decimal, and multiply by the step size in the ADC document on your website (0.000122 V), I am not getting the expected values. I have tested this with a known signal from a function generator and it still hasn't been working.

I know from issue #479 the output of your axi4-Stream Red Pitaya ADC is the input signal in signed two's complement form. However, I am still stuck- if you could elaborate further on exactly how the signal is processed from output to ADC to inverting amplifier to FPGA (and so on) that would be very helpful. Thank you for your time.

pavel-demin, 2020-10-29:
> Sorry to re-open this, but I tried what you did and I still don't understand the format.

What exactly is it that you don't understand?

> I have tested this with a known signal from a function generator and it still hasn't been working.

I'd suggest starting with a a simpler test:
 - connect -1V to one of the board's inputs
 - observe -8192 at the output of the axis_red_pitaya_adc module
 - connect +1V to one of the board's inputs
 - observe 8191 at the output of the axis_red_pitaya_adc module

TestingAccount4, 2020-10-29:
I don't understand how to convert the output I'm seeing with the monitor command to the voltage that is input to the adc. From my understanding the flow goes: Input Signal=>ADC (convert using the ADC data sheet, is 14 bits, not sure what the output form is here) => Inverting amplifier (just flips the sign) =>Your RP ADC IP (extends the sign bit so that the first three msb are all the same value as the sign bit, and negate the remainder of the data. At this point the data is in its 2's complement form ) => GPIO That is monitored in my program. Also, do the ADC and DAC use the same data format? 

From my understanding of your first response, if I want to set a value of 0.05 V in my fpga program from the SoC, I just should write a value of 0.05/0.000122 = 410 (approx) to the threshold, and that will work with the data being read in from the adc. However, when I apply the logic in reverse (i.e. input a voltage of 0.05 V and read the signal using the monitor command) I do not get 410. 

> I'd suggest starting with a simpler test:
> 
> * connect -1V to one of the board's inputs
> * observe -8192 at the output of the axis_red_pitaya_adc module
> * connect +1V to one of the board's inputs
> * observe 8191 at the output of the axis_red_pitaya_adc module

Thanks for the suggestion, I'll try it asap

pavel-demin, 2020-10-29:
An inverting amplifier is an analog component. It is in front of the ADC.

The ADC chip on the Red Pitaya board outputs samples in the offset binary format (not two's complement).

My ADC IP core converts offset binary to two's complement, extends sign and inverts the ADC samples.

> Also, do the ADC and DAC use the same data format?

The ADC and DAC chips on the Red Pitaya board use different data formats.

The axis_red_pitaya_adc and axis_red_pitaya_dac modules from this repository use the same data format. If the axis_red_pitaya_adc and axis_red_pitaya_dac modules are connected directly, the signal at the output of the Red Pitaya board (OUT1 or OUT2) should be very similar to the signal at the input of the Red Pitaya board (IN1 or IN2).

> However, when I apply the logic in reverse (i.e. input a voltage of 0.05 V and read the signal using the monitor command) I do not get 410.

What do you get?

TestingAccount4, 2020-10-30:
I see- thank you for elaborating. 

> What do you get?

My data is below. The value in square brackets is the converted voltage. When the measured value was negative, I used the formula: Bracket_Voltage= -1+x*0.000122, where x is the absolute value of the measured value in decimal. I did this to mimic the 2's complement section of the ADC data sheet on page 25. When the measured value was positive, I just multiplied it by 0.000122 to get the value in brackets. 

My procedure was: 
1. Use the monitor command to obtain the value of the signal from the data acquisition module (as shown above) in hexadecimal.
2. Convert the hex number to a 16 bit string, noting bits 15-13 should be the sign bit, where 1=>negative and 0=>positive. 
3. Delete the leading sign bit copies (and the sign bit) and pad with leading zeros appropriately so that the string is 13 bits. I know the leading zeros won't change the value of the bit string but I was just trying to visualize how I would process it similar to the aforementioned 2's complement column.
4. Convert the 13 bit value calculated in 3 to decimal, and apply the appropriate voltage formula (see first paragraph below "What do you get" for reference). 

Channel 1: 
Nothing plugged in:  -8312  [-0.007896 V]

Input: -0.05 V, Output: -7772 [-0.051816 V]
Input: 0.05 V, Output: 312 [0.038064 V]

Input: -1 V, Output: -900 [-0.8902 V]
Input: 1 V, Output: 7202 [0.878 V]

Channel 2: 

Nothing Plugged in: Output: -8030 [-0.02034 V]

Input: -1 V, Output: -770 [-0.90545 V]
Input: 1 V, Output: 7079 [0.863 V]

Input: -0.05 V, Output: -7670 [-0.0642 V]
Input: 0.05 V, Output: 201 [0.0245 V]

Thanks for your help.

pavel-demin, 2020-10-30:
> Use the monitor command to obtain the value of the signal from the data acquisition module (as shown above) in hexadecimal.

Please provide these values.

pavel-demin, 2020-10-30:
To conclude. The ADC interface module from this repository outputs signed 16-bit values. The sign of these values is the same as the sign of the input voltage. These 16-bit signed values can be used directly in calculations. No additional conversion is necessary.

TestingAccount4, 2020-10-30:
> Please provide these values.

I will as soon as I am able.

> > Delete the leading sign bit copies (and the sign bit) and pad with leading zeros appropriately so that the string is 13 bits.
> 
> Why 13 bits? The ADC chip on the Red Pitaya board outputs 14-bit samples.
>

I did this so that I could follow the conversion process I mentioned above. The output of your adc module (from what I understand) is ({three sign bits, 13 bits that correspond to the absolute value of the measured value I presented above}). In taking off all three of the sign bits (and remembering the sign it corresponds to) I was trying to get an accurate decimal number to multiply by 0.000122 as appropriate. 
 
> > Convert the 13 bit value calculated in 3 to decimal
> > Nothing plugged in: -8312 [-0.007896 V]
> 
> -8312 isn't a 13 bit value. Looks like something is wrong with your conversion.

See above response. 

> The correct conversion procedure is simpler:
> 
> * obtain a 16-bit value
> * interpret it as a 16-bit signed value
> * multiply it by 1/8192 (0.000122)

If the values are signed 16 bit values (that you then multiply by 0.000122 V), wouldn't the binary strings corresponding to +/- 1V be: 

0001 1111 1111 1111 (corresponding to a value of >1V, as per the ADC data sheet. This converts to a value of 8192 in decimal, as expected from your previous post).

1110 0000 0000 0000 (corresponding to a value of =< -1V, as per the ADC data sheet. This converts to a value of -24576 in decimal, which is not what I would expect). 

> The ADC and DAC chips on the Red Pitaya board use different data formats.
>
> The axis_red_pitaya_adc and axis_red_pitaya_dac modules from this repository use the same data format. If the [...]

If I understand your comment (and the axis_red_pitaya_dac code) correctly, it takes in data in the signed 16 bit integer format we are currently discussing, and takes care of the appropriate conversion to get the data sent to the dac to correspond to 0.000122*(signed 16 bit integer). It then uses a memory buffer to ensure signals are being output at the desired frequency (125 MHz in Anton's project). Is this the case?  Thank you for your help.

TestingAccount4, 2020-10-30:
> This is wrong. This is not how the two's complement form works.
> 
> Please do some reading about the two's complement form.

Thank you for explaining as much as you have to me, and thank you for your patience- I have only done a few projects and am still fairly new.  

Using the conversion from two's complement to decimal I got from Wikipedia, I am getting correct values for the voltages I tested (see below) except +50 mV. Do you have any idea why this is occurring? Thank you for your time.

Voltage: +50 mV, Hex: 0x0000013b, Decimal: 315 [Should be around 410]
Voltage: -50 mV, Hex: 0x0000fe59, Decimal: -423 [Around what I would expect]

Voltage: +200 mV, Hex: 0.00000581, Decimal: 1409 [Around what I expect]
Voltage: -200 mV, Hex:  0x0000fa22, Decimal: -1502 [Around what I would expect]

pavel-demin, 2020-10-30:
> Do you have any idea why this is occurring? 

Noise, input offset voltage.