pavel-demin/red-pitaya-notes #979: frequency range and data rate of the RP122.88-16 transceiver application

hanrod97, 2020-10-05:
Hello Pavel, 

i am currently using a red pitaya 122.88-16 with your SD-Card-Content "red-pitaya-alpine-3.12-armv7-20200628" and i'm pretty new to the whole topic using SDRs, programming FPGAs, GitHub itself and stuff like that. 
Anyways, thanks for providing this nice piece of software and your active community-support.

A few questions came across my way in the past weeks and maybe you can help me with them, since i don't find help in the other issues or application notes:

1. In your Application-Notes for the sdr-transceiver-hpsdr-122-88 (http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr-122-88/) you wrote "The tunable frequency range covers from 0 Hz to 61.44 MHz.", which kind of confuses me since the RP122.88-16 is AC coupled and has a Lowest -3dB RF input corner frequency of 300kHz and on the other hand can receive up to 550 MHz if we use undersampling (which we do anyway if i got the schematic right?). Is there an easy way to open the frequency range above the mentioned 61.44MHz, or is there another reason for this boundary?

2. Why are there only so little I/Q-Data Rates? If i look into other projects for the RP125-14 i see data rates up to 1250kSPS (http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/) but not in this one. 

Thank you for support and best regards,

Hannes

pavel-demin, 2020-10-05:
Hello Hannes,

You're right, low frequencies are very likely unusable with the 122.88-16 board.

To use undersampling, no modification of the software running on the Red Pitaya board is required. I think in PowerSDR mRX PS the transverter settings can be used to configure VHF bands corresponding to higher Nyquist zones.

The HPSDR/Metis communication protocol supports only four rates. This is why only four rates are implemented in sdr-transceiver-hpsdr-122-88.

Best regards,

Pavel

hanrod97, 2020-10-06:
Hello Pavel,

thanks for your fast response.
regarding the undersampling-thing: that's a good hint, so at least i know it's possible. Unfortunately I'm using GNU Radio with the gr-hpsdr package, so i'll see what i can do with that piece of software and if they can help me in some forums.

to the HPSDR-thing: ok that's a pity. Do you have any idea how else i could connect the Red Pitaya 122.88-16 to GNU Radio? I feel like all the current solutions are only for the 125-10 and 125-14.

Best regards,

Hannes

pavel-demin, 2020-10-06:
Out of curiosity, what kind of system are you trying to build and what are the requirements?

> Unfortunately I'm using GNU Radio with the gr-hpsdr package, so i'll see what i can do with that piece of software and if they can help me in some forums.

With GNU Radio, undersampling is very easy. If you want to tune to `n * 122.88 + x`, you set the RX frequency to `x` and that's it. Frequency conversion calculations can be easily implemented in your code without any changes to gr-hpsdr.

hanrod97, 2020-10-06:
I'm trying to measure capacitance with RF for an university project. So it'll be basically an VNA, but i want to cover a higher range of frequency and also different measurement techniques (so i need "2 VNA-Ports"), that's why i decided to use GNU Radio and the SDR-Software, not the VNA one. Requirements are 2 Inputs (that's why i got the RP122.88-16), a frequency range from about 500kHz to 300MHz (you see the specs of the RP again) and to export the measured data. So i basically just need to measure phase shift between two CW-Signals with various frequency.

Oh thanks, so easy yet i didn't come up with that yet. I guess i'll also have to turn that around somehow between the Nyquist 61.44 and the full 122.88, but i'll see in some tests, thank you for the hint!

pavel-demin, 2020-10-06:
> Requirements are 2 Inputs (that's why i got the RP122.88-16), a frequency range from about 500kHz to 300MHz (you see the specs of the RP again) and to export the measured data.

In the VNA application, two ports can be used. The data from the second input aren't used in the GUI client, but are available.

If the frequency range from 500 kHz to 300 MHz is required to be continuous, then I don't think that this can be done using undersampling.

hanrod97, 2020-10-06:
Good to know, maybe i can try it like that if my current solution doesnt work.

It's not meant to be continuous, more various stages to get different data from different frequencies to compare, which should work.... or not?

pavel-demin, 2020-10-06:
If it's not continuous, then maybe it'll work. However, you'll need a set of switchable bandpass filters to filter the frequency ranges in each of the Nyquist zones that you want to analyze.

hanrod97, 2020-10-06:
ok thank you very much for your help, i'll see what i can do :)

Have a nice day!

closed 2020-10-06 by hanrod97