pavel-demin/red-pitaya-notes #980: Suggestion about the mode of data transmission of the VNA program

MxQL, 2020-10-07:
### Description of the setup:

 * Device: RedPitaya VNA
 * SD card image:  the last version of your VNA server program
 * Application: VNA
 * Other relevant information:

### Description of the problem:

Currently, the vna.c program has to receive one command containing the 10 parameters then return the sweep  data collected at the received frequency then stops. It needs to receive a new command of 10 parameters and send data again.

In the case of unchanged parameters, would it be possible to let the RedPitaya to continue to send data, until a cancel parameter  is received? That way, it seems to me that the the RedPitaya's operation would be smoother and data rate would be higher.

Thanks

Maxim

pavel-demin, 2020-10-07:
I think that what you propose will not solve any problem and will only add new problems.

> It needs to receive a new command of 10 parameters and send data again.

If the parameters don't change, there is no need to resend them.

If you want to repeat the frequency sweep with exactly the same parameters, you can simply send the sweep command.

> In the case of unchanged parameters, would it be possible to let the RedPitaya to continue to send data, until a cancel parameter is received?

When the frequency sweep is complete, there is no other data to send.

What data should the VNA application send continuously?

If the data are sent continuously, then how to find where is the beginning and the end of a single frequency sweep?

> That way, it seems to me that the the RedPitaya's operation would be smoother and data rate would be higher.

What is wrong with current data rate?

Do you have test results confirming that your approach performs better?

MxQL, 2020-10-08:
Thank you for your reply. I didn't see that one can simply send the sweep command to resume.

I often work with only one frequency, may be continuous sending makes sense? I need to acquire as much samples as possible because the phenomena to observe is fast. Actually, for my application, the best is to get R & X data of the signal at 1 ksamples/sec. All this is done at only one frequency. For the current operation mode, the RedPitaya has to wait for external commands, so I think it is less rapid than the continuous mode.

Thanks

pavel-demin, 2020-10-08:
> I often work with only one frequency, may be continuous sending makes sense?

I see. This is a pretty specific use case.

> Actually, for my application, the best is to get R & X data of the signal at 1 ksamples/sec.

I think that if you set the sweep size to 32768 and the rate to 45, you'll obtain a continuous stream of samples at 1 ksamples/sec during 32 seconds.

MxQL, 2020-10-09:
Dear Pavel,

Thank you very much for your advice. Your solution suits me perfectly,
indeed.

Best wishes

Maxim

closed 2020-10-09 by pavel-demin