pavel-demin/red-pitaya-notes #981: DAC update rate

lely475, 2020-10-19:
Hi Pavel,

I have a question regarding the update rate of the red pitaya DAC.
Ideally I want to record and downsample a signal from 125 MHz to 31.25MHz:
ADC -> CIC decimation filter (1:4) (-> maybe add FIR filter for removing unwanted signal comopnents)
and then playing the signal again from 31.25MHz to 125MHz:
(maybe add FIR filter here ->) CIC interpolation filter (4:1) -> DAC

However the DAC output is looking like this:
[image]
instead of this:
[image]

Then I tried to record the signal at 125MHz with no filtering after, to make sure it is not related to that and just play it straight from the ADC output. However the results were almost identical to the above.

Next I added your interpolation core in the player project just to check if it would make a difference, and voilà with an interpolation factor of around 20 I can finally get a more desired output signal:
[image]
But since I'm now basically holding each sample for 20 clk cycles the signal is a lot longer than the originally recorded one, which poses a big issue in my application.

Now my question is:
Do you have any idea why that is the case? Can the DAC not handle a faster signal change? (Is this related to the DAC settling time?)
And is it necessary for the 14Bit values that go into the DAC to stay constant for a certain amount of clk cycles (of 125MHz clk)? Or did I just miss some basic understanding here?

Thank you already,
Lydia

pavel-demin, 2020-10-19:
Hi Lydia,

I think all the setups you describe should work.

You can confirm that the problem is not with the DAC by directly connecting the ADC and DAC interfaces and checking the input and output signals.

I think the problem is in the part of your setup where samples are transferred from RAM to the FPGA. At a sample rate of 31.2 MSPS, this should be done with DMA.

If you are not using DMA while playing the signal, you are most probably not transferring samples fast enough.

Cheers,

Pavel

lely475, 2020-10-20:
Thanks a lot, yes I am currently not using DMA. I will look into it!

closed 2020-10-20 by lely475