pavel-demin/red-pitaya-notes #984: wspr trx uploads

jeffforber, 2020-10-23:


jeffforber, 2020-10-23:
!--Hello Pavel,,,i just  purchased a second RED-pitaya to run WSPR trx on ..the first one has been monitoring FT8 for a year or so and is very good in fact some times better than my SDRplays with WSJT-X  ...but the new one running WSPR trx doesnt up load every 2mins it seems be ok for 30 mins then it may go 10mins before it uploads again or it uploads every 4mins it is very erratic  i just checked pskreporter it didnt upload for 30mins.... iam monitoring on same antenna with  sdrplay and wsjt-x it is ok on there never drops out  ...i have tried 3 different SD cards 
i have swapped over the REDPITAYAS the problem is still there ...please can you advise ...i am not very up on linux ...iam  ok copy and paste as my memory is not  so good these days,, thank you in advance jeff  G8SCU
ps the call sign iam using on WSPR.net is TA4/G8SCU2

pavel-demin, 2020-10-23:
I've just tried to connect to the WSPRnet website and it seems to be very slow at the moment. So I think the problem is on the WSPRnet website's side.

jeffforber, 2020-10-23:
i just checked wspr again it looks like it has stopped uploading

jeffforber, 2020-10-23:
wsjt-x is ok no problem 
the call sign for wsjt-x  is TA4/G8SCU

jeffforber, 2020-10-23:
wsprnet is always like this very slow  but never misses uploading from wsjt-x

jeffforber, 2020-10-23:
i have rebooted the RP

jeffforber, 2020-10-23:
it is back on  and reporting ok atm but it wont last long .....  i have been trying to get it working for 3 days it is ok for a bit then starts dropping uploads

jeffforber, 2020-10-23:
and then stopping all together

jeffforber, 2020-10-23:
i have fitted fans to both RPs as i thought it would help

pavel-demin, 2020-10-23:
What Red Pitaya model you're using?

jeffforber, 2020-10-23:
125-14

jeffforber, 2020-10-23:
its stopped uploading again

jeffforber, 2020-10-24:
hello Pavel ....today i have connected both my REDPITS to same antenna running WSPR on both ...call signs TA4/G8SCU1 & TA4/G8SCU2  my SDRplay & WSJT_X  are on same antenna call sign TA4/G8SCU  the REDS are uploading different times to each other & about half as many spots as WSJT_X iam very disapointed as i purchased the second REDPITAYA to do antenna comparisons ..   regards jeff

pavel-demin, 2020-10-25:
I've just checked what URL is used in the WSJT-X code to contact WSPRnet. I've found it in [this file](https://sourceforge.net/p/wsjt/wsjtx/ci/master/tree/Network/wsprnet.cpp#l23). It's the same URL that is used in [my script](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/decode-wspr.sh#L65). So I'm assuming the problem isn't in the way my script contacts WSPRnet.

Could you please check the following:
 - When decodes are not uploaded to WSPRnet, are there any error messages in /dev/shm/decode-wspr.log?
 - When decodes are not uploaded to WSPRnet, are there any decodes in /dev/shm/ALL_WSPR.TXT?

jeffforber, 2020-10-25:
where do i find these files ...sorry i need step by step instructions

pavel-demin, 2020-10-25:
These files are present on the Red Pitaya board when the WSPR application is running.

You'll need to connect to the Red Pitaya command line using SSH and check the files with the following commands:
```
cat /dev/shm/decode-wspr.log
cat /dev/shm/ALL_WSPR.TXT
```

Some information about IP address, login and password can be found at [this link](https://pavel-demin.github.io/red-pitaya-notes/alpine).

jeffforber, 2020-10-25:
rp-f06a8e:~# cat /dev/shm/decode-wspr.log
Sun Oct 25 13:41:01 UTC 2020
Sleeping ...
Sun Oct 25 13:42:00 UTC 2020
Recording ...
Sun Oct 25 13:43:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 13:44:00 UTC 2020
Recording ...
<DecodeFinished>
<DecodeFinished>
1342 -22 -1.4  18.106020  0  IW2DMO JN45 37
<DecodeFinished>
Uploading ...
rp-f06a8e:~# cat /dev/shm/ALL_WSPR.TXT
cat: can't open '/dev/shm/ALL_WSPR.TXT': No such file or directory
rp-f06a8e:~#

jeffforber, 2020-10-25:
1322 -15  0.6  14.097124  0  ON7KO JO21 37
1322 -13  0.3  14.097134  0  M0AFZ JO01 33
1322 -20 -1.1  14.097137  0  DB1NN JN59 30
1322 -25  0.3  14.097144  0  ON4WS JO20 30
1322 -21 -0.0  14.097168  0  DG8GAD JN48 27
1322 -13  0.3  14.097170  0  G0CCL JO02 23
1322 -18 -1.1  14.097174  0  OH3FR KP20 33
1322 -14  0.5  14.097208  0  OE7XZB JN57 23
<DecodeFinished>
Uploading ...
curl: (28) Connection timed out after 30001 milliseconds
Sun Oct 25 13:25:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 13:26:00 UTC 2020
Recording ...
1324 -26  0.3  18.106099 -1  DL1DUZ JO61 23
1324 -29  0.3  18.106127  0  DG8ACD JO52 30
<DecodeFinished>
1324 -23 -4.2   7.040050  0  DG8DP JN58 30
1324  -8  0.6   7.040097 -2  YO4DFT KN44 30
1324 -16 -1.5   7.040130  0  UR4UPQ KO50 30
1324 -28  0.3   7.040156  0  IZ3WXD JN55 27
<DecodeFinished>
1324 -31  0.9  21.096105  0  G3SXH IO80 30
<DecodeFinished>
1324 -23 -0.6  14.097016 -4  HB9BAX JN36 30
1324 -16  0.2  14.097037  0  DF7II JO62 30
1324 -14  0.7  14.097046  0  DJ7KA JN48 33
1324 -25  0.2  14.097055  0  DL4KBB JN59 33
1324 -17  1.6  14.097057  0  G8MCD IO91 30
1324 -17  0.2  14.097061  0  <PE0FK/P> JO21WI 37
1324 -21 -0.4  14.097070  0  IW5BNL JN53 27
1324 -13  0.8  14.097076  3  DJ6LB JN59 20
1324  -3  1.6  14.097124  0  G4UGD IO83 37
1324 -22  0.2  14.097141  0  B0SUE JN18 37
1324 -22 -0.1  14.097146  0  DL1FX JN49 20
1324 -10  0.3  14.097152  0  G0CCL JO02 37
1324  -8  1.4  14.097166  0  RD4HU LO53 37
1324 -23 -0.4  14.097168  0  DG8GAD JN48 27
1324 -20  0.1  14.097174 -1  M7ANJ IO92 27
1324 -16  1.1  14.097194 -1  DJ4FF JN47 37
<DecodeFinished>
Uploading ...
curl: (56) Recv failure: Connection reset by peer
Sun Oct 25 13:27:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 13:28:00 UTC 2020
Recording ...
1326 -22 -0.7   7.040017  0  SP5OSF KO01 37
1326 -25  0.3   7.040051  0  DK2DB JN48 37
1326 -21  1.0   7.040072  0  UR5MLG KN99 23
1326 -27  0.2   7.040092  0  IZ3WXD JN55 27
1326 -18  1.0   7.040104  0  I0UVN JN61 30
1326 -27  0.3   7.040106  0  ON7KB JO21 37
1326 -20  0.4   7.040173  0  DG7RJ JN58 37
<DecodeFinished>
<DecodeFinished>
1326 -17  1.2  18.106055  0  LY3EC KO24 37
1326 -15  0.2  18.106113  0  OZ7IT JO65 37
1326 -22  0.3  18.106153  0  DL4MFC JN58 23
1326 -25  0.2  18.106167  0  F4WCW JN05 20
[... 1919 lines cut ...]
-ash: 1328: not found
rp-f06e25:~# 1328 -15  0.3  14.097170  0  G0CCL JO02 23
-ash: 1328: not found
rp-f06e25:~# 1328 -19  0.1  14.097177  0  PD0KT JO33 10
-ash: 1328: not found
rp-f06e25:~# <DecodeFinished>
-ash: syntax error: unexpected newline
rp-f06e25:~# Uploading ...
-ash: Uploading: not found
rp-f06e25:~# curl: (28) Connection timed out after 30000 milliseconds
-ash: syntax error: unexpected word (expecting ")")
-ash: syntax error: unexpected ")"
rp-f06e25:~# Sun Oct 25 13:31:01 UTC 2020
-ash: Sun: not found
rp-f06e25:~# Sleeping ...
-ash: Sleeping: not found
rp-f06e25:~# Decoding ...
-ash: Decoding: not found
rp-f06e25:~# <DecodeFinished>
-ash: syntax error: unexpected newline
rp-f06e25:~# <DecodeFinished>
-ash: syntax error: unexpected newline
rp-f06e25:~# <DecodeFinished>
-ash: syntax error: unexpected newline
rp-f06e25:~#

pavel-demin, 2020-10-25:
Thanks for the logs. The following error messages mean that the connection with the WSPRnet website is very poor:
```
curl: (56) Recv failure: Connection reset by peer
curl: (28) Operation timed out after 30001 milliseconds with 0 bytes received
```

As I already mentioned, I have the same problem when trying to connect to WSPRnet  with a web browser. Sometimes it's just very slow, sometimes I get the 503 error.

So the problem is clearly on the WSPRnet side and I don't think I can do anything to fix it.

jeffforber, 2020-10-25:
why is it perfect  with WSJT-X  it never misses a  time

jeffforber, 2020-10-25:
looks like i wasted my money on red pitaya 
i just connected RED PITAYA  HPsdr to WSJT-x  with a vac it was decoding more spots than my SDRplay  & wsjt-x so the red pitaya rx is very good...but i have just reconnected wspr trx on red pitaya with call sign TA4/G8SCU2 7 at it not working ......so back to the drawing board

jeffforber, 2020-10-25:
so how can i monitor 8 bands at once with red pitaya & wsjt-x as it is not compatible with SDRUNO or SDRconsole   any advice would be welcome ...thank you for your time

jeffforber, 2020-10-25:
i was just wondering can bad micro sd  cards cause this do think

jeffforber, 2020-10-25:
what minimum size do you recommend

pavel-demin, 2020-10-25:
> i just purchased a second RED-pitaya to run WSPR trx on ..the first one has been monitoring FT8 for a year or so
> looks like i wasted my money on red pitaya

You should have tested the WSPR application with the first Red Pitaya board before buying the second one.

> i was just wondering can bad micro sd cards cause this do think

If the OS boots and the WPSR application runs, then the micro SD card is OK.

> what minimum size do you recommend

256 MB

I'd suggest checking your local network (cables, connections, router configuration, etc) and maybe trying one of the older versions of the SD card image:

https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20190527
https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20191120

jeffforber, 2020-10-25:
i did test wspr trx on first RP  but not deep enough ...it wasnt until i had 2 RPs running together that i see this problem giving very different results .i have tried older releases but get the same problems .....you say its wsptnet is the problem ...but when i am running 12 instances of WSJT_X with 2 sdrplays i dont have any drops outs ... you just need a very powerful pc ...using huge amounts off electricity in comparison with a RED-PITAYA

jeffforber, 2020-10-25:
is there away i can use HPsdr with 6 rxs ... to feed 6 wsjt-xs

dc7ds, 2020-10-25:
Try sparksdr http://www.ihopper.org/radio/ ....using sparksdr and a red pitaya, you can decode on 8 bands in parallel. Wspr, FT8 and FT4 are supported. You might need a strong decoding machine, I testet 4 instances of FT8 decoding on a Raspberry pi 4, that was quite the limit there

jeffforber, 2020-10-25:
> Try sparksdr http://www.ihopper.org/radio/ ....using sparksdr and a red pitaya, you can decode on 8 bands in parallel. Wspr, FT8 and FT4 are supported. You might need a strong decoding machine, I testet 4 instances of FT8 decoding on a Raspberry pi 4, that was quite the limit there

thank you STEFAN for the info i will try

pavel-demin, 2020-10-25:
When trying to access WSPRnet from a web browser, I have the impression that http**s**://wsprnet.org works much better than http://wsprnet.org.

Could you try to replace `http` with `https` on line 65 in decode-wspr.sh? It's the same file where you specified your call sign.

jeffforber, 2020-10-25:
thank you for that advice PAVEL

jeffforber, 2020-10-25:
did great for 10 mins more deodes than wsjt-x

jeffforber, 2020-10-25:
rp-f06a8e:~# cat /dev/shm/decode-wspr.log
Sun Oct 25 21:03:01 UTC 2020
Sleeping ...
Sun Oct 25 21:04:00 UTC 2020
Recording ...
Sun Oct 25 21:05:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:06:00 UTC 2020
Recording ...
Sun Oct 25 21:07:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:08:00 UTC 2020
Recording ...
Sun Oct 25 21:09:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
2108 -22 -1.5   7.040120  0  UR4UPQ KO50 30
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Uploading ...
Sun Oct 25 21:10:00 UTC 2020
Recording ...
Sun Oct 25 21:11:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:12:00 UTC 2020
Recording ...
<DecodeFinished>
<DecodeFinished>
2110  -9  0.5   3.570094  0  SV1CIF KM17 30
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Uploading ...
Sun Oct 25 21:13:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
Sun Oct 25 21:14:00 UTC 2020
Recording ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
2112 -15  0.3   3.570041  0  DK2DB JN48 37
2112 -25  0.2   3.570062  0  RX3AOM KO86 30
2112 -26 -0.3   3.570105  0  ON7KO JO21 37
<DecodeFinished>
2112 -19  0.2  10.140204  1  G0LJU IO91 43
<DecodeFinished>
2112 -23  2.1   5.288776  0  G4ZFQ IO90 30
<DecodeFinished>
Uploading ...
Sun Oct 25 21:15:01 UTC 2020
Sleeping ...
Decoding ...
Sun Oct 25 21:16:00 UTC 2020
Recording ...
2114 -20  1.1   3.569994  0  F4FRQ JN37 30
2114 -25  0.2   3.570039  0  DJ6DK JN48 37
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
2114 -22  0.2   5.288620  0  G4GCI IO90 37
<DecodeFinished>
Uploading ...
Sun Oct 25 21:17:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:18:00 UTC 2020
Recording ...
<DecodeFinished>
2116 -29  0.2   7.040071  0  R2ARX KO96 37
<DecodeFinished>
2116 -25  0.3  10.140134  0  DL1BGE JO43 23
2116 -26  1.2  10.140167  0  DL2BBC JO32 23
2116 -24  0.3  10.140200  0  DJ2TS JO40 30
<DecodeFinished>
<DecodeFinished>
Uploading ...
Sun Oct 25 21:19:01 UTC 2020
Sleeping ...
Decoding ...
2118 -22  0.4   3.570063  0  RX3AOM KO86 30
2118  -5  0.4   3.570095  0  SV1CIF KM17 30
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:20:00 UTC 2020
Recording ...
2118 -25  0.2  10.140133  0  G0MBA JO01 23
2118 -25  0.2  10.140154  0  DF1DR JO31 33
<DecodeFinished>
2118 -21  0.3   5.288620  0  <G4GCI> IO90GU 37
<DecodeFinished>
Uploading ...
Sun Oct 25 21:21:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:22:00 UTC 2020
Recording ...
Sun Oct 25 21:23:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:24:00 UTC 2020
Recording ...
Sun Oct 25 21:25:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:26:00 UTC 2020
Recording ...
Sun Oct 25 21:27:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:28:00 UTC 2020
Recording ...
Sun Oct 25 21:29:01 UTC 2020
Sleeping ...
rp-f06a8e:~#

jeffforber, 2020-10-25:
p-f06a8e:~# cat /dev/shm/decode-wspr.log
Sun Oct 25 21:03:01 UTC 2020
Sleeping ...
Sun Oct 25 21:04:00 UTC 2020
Recording ...
Sun Oct 25 21:05:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:06:00 UTC 2020
Recording ...
Sun Oct 25 21:07:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:08:00 UTC 2020
Recording ...
Sun Oct 25 21:09:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
2108 -22 -1.5   7.040120  0  UR4UPQ KO50 30
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Uploading ...
Sun Oct 25 21:10:00 UTC 2020
Recording ...
Sun Oct 25 21:11:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
<DecodeFinished>
Sun Oct 25 21:12:00 UTC 2020
Recording ...
<DecodeFinished>
<DecodeFinished>
2110  -9  0.5   3.570094  0  SV1CIF KM17 30
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
Uploading ...
Sun Oct 25 21:13:01 UTC 2020
Sleeping ...
Decoding ...
<DecodeFinished>
Sun Oct 25 21:14:00 UTC 2020
Recording ...
<DecodeFinished>
<DecodeFinished>
<DecodeFinished>
[... 481 lines cut ...]
rp-f06a8e:~# <DecodeFinished>
-ash: syntax error: unexpected newline
rp-f06a8e:~# <DecodeFinished>
-ash: syntax error: unexpected newline
rp-f06a8e:~# <DecodeFinished>
-ash: syntax error: unexpected newline
rp-f06a8e:~# Sun Oct 25 21:28:00 UTC 2020
-ash: Sun: not found
rp-f06a8e:~# Recording ...
-ash: Recording: not found
rp-f06a8e:~# Sun Oct 25 21:29:01 UTC 2020
-ash: Sun: not found
rp-f06a8e:~# Sleeping ...
-ash: Sleeping: not found
rp-f06a8e:~# rp-f06a8e:~#
-ash: rp-f06a8e:~#: not found
rp-f06a8e:~# cat /dev/shm/ALL_WSPR.TXT
201025 2130   2 -25  0.32   3.5700572  RX3AOM KO86 30          0     1    0    1   44  0
201025 2130   2 -21  0.24  10.1401883  G0LJU IO91 43           0     1    0    1  273  0
201025 2130   2 -21  1.26   7.0399931  7X2UV JM16 37           0   292    0    2    1  0
201025 2130   2 -24  0.75   7.0400767  YO4DFT KN44 30         -3     1    0    1  227  0
201025 2130   3 -22  1.05   7.0400892  PD0AV JO21 27           0     1    0    1  359  0
201025 2130   1 -27  0.19   7.0401396  G4GVZ IO81 37           0   158    0    1  -47  0
201025 2130   3 -23 -0.02   7.0401712  LZ1UBO KN12 37          0     1    0    1  467  0
rp-f06a8e:~#

pavel-demin, 2020-10-25:
Does it work better now? What have you changed?

jeffforber, 2020-10-26:
http...https  it is still no good

jeffforber, 2020-10-26:
thank you for your . time i have given up now and trying Spark sdr  with the red pitaya hpsdr   kind regards jeff

closed 2020-10-26 by jeffforber