pavel-demin/red-pitaya-notes #985: FT8 .c2 example files

softerhardware, 2020-10-25:
Hi,

I'm trying to get the hermes-lite to work with the ft8d .c2 4000Hz decoder used on the red-pitaya. I've recompiled it for x86. I've updated the hermes-lite to generate 4000Hz complex data. I can't get ft8d to decode my data. I'd like to check if there is a problem with my .c2 files. Are there any example working ft8 .c2 files available that I can use for testing?

73,

Steve
kf7o

softerhardware, 2020-10-25:
I figured out that I need to invert Q from the IQ stream. I now have good decodes from ft8d. I noticed that the code has depth set to 1 as default. Is that what is used by most users? When I change depth to 3, I do see more spots (34 vs 29) and the time is still very quick on my ryzen system.

dc7ds, 2020-10-25:
At our skimming station DL0PF we are using depth=5. Sysload on odroid N2 at peak times when decoding 16 .c2 files in parallel: https://www.youtube.com/embed/t3cN5uIc7l0   :-)

pavel-demin, 2020-10-25:
Hi Steve,

Many .c2 files were provided by Andy K1RA in [this comment](https://github.com/pavel-demin/red-pitaya-notes/issues/689#issuecomment-446850894). Here is a link to his dropbox folder:
https://www.dropbox.com/sh/5w15ghvh9gi1ryw/AAAmETbU5BRDKYBGRbQjreDFa?dl=0

The depth is set to 1 because the CPU on the Red Pitaya boards isn't powerful enough to run the FT8 decoder with higher depth settings.

Best regards,

Pavel

softerhardware, 2020-10-26:
Hi Pavel,

Thanks for the link. I confirmed that my .c2 files are the same. I am able to generate .c2 files with the HL2 and decode with your ft8d. I've found your ft8d to be the fastest. Thanks for sharing it. Below is some data I collected. It appears that depth 3 takes the longest and produces the most spots. I saw the same behavior with .c2 files from K1RA. What is the difference between depth 3 and 5? Depth 3 and 5 behave differently for ft8d versus jt9. I've been changing ndepth directly in ft8d.f90. Is that the expected place to change it?

I am working on connecting two HL2s with coherent clock to do multidirection beam forming softeware. For example, decode FT8 from multiple "beam" directions by adding up complex IQ from RX1 and RX2 with RX2 rotated by some angle in real time to increase spot counts. This should also be possible with the 2 ADCs on the red pitaya. Has someone tried this with the red pitaya?

ft8d (Optimized, loops unrolled and using processor AVX2, -O3 -funroll-loops -march=znver2 -mtune=znver2 -fopt-info)
0.2 seconds for 41 spots at depth 1
2.2 seconds for 61 spots at depth 3
0.9 seconds for 53 spots at depth 5

jt9 (Unoptimized, built straight from the source with default settings)
0.6 seconds for 18 spots at depth 1
6.8 seconds for 30 spots at depth 3
7.8 seconds for 30 spots at depth 5

jt9 (Optimized, loops unrolled and using processor AVX2, -O3 -funroll-loops -march=znver2 -mtune=znver2 -fopt-info)
0.2 seconds for 18 spots at depth 1
2.0 seconds for 30 spots at depth 3
2.1 seconds for 30 spots at depth 5

73,

Steve
kf7o

dc7ds, 2020-10-26:
sorry i was wrong, we're using ndepth=3 in ft8d which gave the best results in our case. Your optimizations look promising, have you any hint, which of them should/could work on an ARM-based odroid N2 board?

73,
Stefan DC7DS

closed 2020-10-29 by pavel-demin