pavel-demin/red-pitaya-notes #986: Question on write-c2-files bandwidth

pskerrett, 2020-11-17:
HI Pavel,

Thanks again for your amazing software. This has revolutionized tracking beacons and reporting spots for me.. I'm having more fun with your wspr & ft8 decoders than I have had in ham radio for years. Thanks again..

In an ideal world, I would love to be able to spot and report both FT8 and WSPR signals at the same time, without compromising 8 band capabilities. 

Would it be possible to say extend the sampling bandwidth of the resulting C2 files such that we could feed a single C2 file to both the WSPR and the FT8 decoder?

IE WSPR for 30 meters is on 10.1387Mhz and FT8 is 10.1360Mhz. If each typically has a 3Khz passband, then a file that samples from 10.136 - 10.1417 should cover both segments, and allow us to feed both wsprd and ft8d for decoding from the same file.

- Would this even be possible?
- Is this just a dumb idea?

Thanks again Pavel,

73, W9PDS.

pavel-demin, 2020-11-17:
Hi Patrick,

I'm glad that the FT8 and WSPR receivers are working well for you.

I don't think the Red Pitaya board has enough CPU resources to run both FT8 and WSPR decoders simultaneously.

I think [SDR receiver compatible with HPSDR](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver-hpsdr/) has a bandwidth that covers FT8 and WSPR bands and can be used with [Spark SDR](http://www.ihopper.org/radio) to decode FT8 and WSPR signals.

Cheers,

Pavel

closed 2020-11-20 by pavel-demin

pskerrett, 2020-11-21:
Thanks Pavel,

I decided to take advantage of the current sale going on and purchase a second RedPitaya.. I will put a low noise pre-amp in front of the split and feed both units. One will do FT8 and one will do WSPR. 

Problem solved.

73, W9PDS.