pavel-demin/red-pitaya-notes #987: Link going to 404

waterwin, 2020-11-20:
http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver-wide-122-88/

Referred to from 
http://rp-f0xxxx.local/sdr_receiver_wide_122_88

Goes to 404

The Alpine image on the SD card is from here : https://github.com/pavel-demin/red-pitaya-notes/releases/download/20200628/red-pitaya-alpine-3.12-armv7-20200628.zip
The Red Pitaya is a SDRlab 122-16

pavel-demin, 2020-11-20:
I have nothing to write about the sdr_receiver_wide_122_88 application. This application was developed for @johnretired and he knows how to use it.

closed 2020-11-20 by pavel-demin

waterwin, 2020-11-21:
Is it same as the 125-14 wideband receiver but brought to 122-16 or completely different?

pavel-demin, 2020-11-21:
It's completely different. It's very wide-band (6-7 MHz). I don't have any usable client code that works with this application.