pavel-demin/red-pitaya-notes #989: Relationship between sweep size, decimation rate, acquisition duration and data out rate?

MxQL, 2020-12-14:
Hi Pavel,

In a previous discussion, I asked about the possibility to let the VNA app run continuously at only 1 frequency. You advised me a configuration where sweep size= 32768, rate= 45 in order to obtain a continuous stream of samples at 1ksamples/sec during 32 seconds ([pavel-demin/red-pitaya-notes] Suggestion about the mode of data transmission of the VNA program (#339 ).

It works, indeed. However, I still not understand the relationship between these parameters.

Recently, at a single output frequency of 52 kHz, I set sweep size= 10000, rate= 1000 and the stream lasts several minutes. I don't know what is the data acquisition rate in this case.

Besides this, each time I restart the data acquisition again, there is an outlier data that results in a spike in my curve. I wonder if it is possible to slightly modify the vna.c app on the Redpitaya in order to make it stream out data continuously for single frequency operations, or this depends on the FPGA design and cannot be changed by programming.

Thank you for your help.

Maxim

pavel-demin, 2020-12-14:
Hi Maxim,

The expression for the sample rate is `50000 / (rate + 5)`. The five are added so that the first five potentially noisy samples can be skipped.

Continuous operation would require changes in both server code and FPGA configuration.

Cheers,

Pavel

MxQL, 2020-12-15:
Dear Pavel,
The continuous operation mode is very useful for many application, especially in sensor measurement. Would you plan to develop a new version for this mode in a near future?
Best regards
Maxim

pavel-demin, 2020-12-15:
I've already tried to describe the issues I see when combining continuous mode with the VNA application in https://github.com/pavel-demin/red-pitaya-notes/issues/980#issuecomment-705137170.

I think that your requirements:
  - single frequency
  - continuous operation

are incompatible with VNA requirements:
  - several frequencies
  - discontinuous operation

I think it doesn't make sense to mix these incompatible requirements into one application, and it would make more sense to create a separate application that would meet your requirements.

Am I missing something?

MxQL, 2020-12-15:
> mode with the VNA application in #980 (comment)

No. You don't miss any thing. It would be nice if you plan to develop this
application in a near future.
I belong to a company and can make money with your application. It's great.

pavel-demin, 2020-12-15:
I do not plan to develop this application because at the moment it is of no use to me.

pavel-demin, 2020-12-15:
> I belong to a company and can make money with your application. It's great.

Don't you think that asking me to develop this application for your company for free isn't very fair.

MxQL, 2020-12-16:
Dear Pavel. I let you know that I belong to a company because I want you to
cooperate with my company. The development won't be free.
I think a private development will take you less time for support than a
public one. Moreover, your help will be really important for us. It will
allow us to develop interesting things.
Could we discuss about this?

Thanks

pavel-demin, 2020-12-16:
> Could we discuss about this?

Yes, let's discuss it by e-mail. You can find my e-mail address in [my profile](https://github.com/pavel-demin).

closed 2020-12-16 by pavel-demin