pavel-demin/red-pitaya-notes #991: Spurious Rejection/STEMlab 122.88-16 SDR

Koichi-Koyama, 2021-01-22:
There are relatively strong spurious signals observed around 73MHz on 6m band and 94MHz on 10m band transmissions. 
It seems that these spurious are related to 122.88MHz clock signal based on the calculation below. 

6m: 122.88 - 50.1 = 72.78MHz (Photo attached)
10m: 122.88 - 28.5 = 94.38MHz
Other HF bands also have similar spurious signals with the same calculation, but more than 100MHz. 

I am wondering whether these spurious can be rejected by amending software if possible, that may be more smart solution. 
Is this possible?
	
For daily operation, I have inserted homemade LPF/Notch Filter between OUT1 and RF amplifier to reduce the spurious to permissible level. 

Koichi Koyama
7N1ICA

[image]

pavel-demin, 2021-01-22:
As far as I know, what you are observing is one of the main features of the STEMlab 122.88-16 SDR board. This board doesn't have any filters on its RF outputs and can be used for undersampling reception and transmission.

Signals 2, 3, 4 on the spectrogram are the images of the 50.1 MHz signal in the higher order Nyquist zones.

I think that Red Pitaya boards with the DAC clocked at 122.88 and 125 MHz are not suitable for transmission in the 6m band because they always generate a signal image very close to the main signal and difficult to filter out. For transmission in the 6m band, the DAC must operate at a higher sample rate. For example, the integrated transceiver chips like [AFE7225](http://www.ti.com/product/AFE7225) have the DAC sample rate twice higher (250 MSPS) than the ADC sample rate (125 MSPS).

Koichi-Koyama, 2021-01-22:
Dear Pavel-san,

Thank you very much for your prompt response and noted.

Best regards,

Koichi Koyama

closed 2021-01-26 by pavel-demin