pavel-demin/red-pitaya-notes #992: 122-88 is not listening to 16 bands in parallel but to 11 - documentation typo

waterwin, 2021-02-03:
### Description of the problem:
documentation says:
simultaneously record FT8 signals from sixteen bands

### Correct according to write-c2-files.cfg file
simultaneously record FT8 signals from eleven bands

There are no more HF bands fitting in the frequency range of the Red Pitaya.

pavel-demin, 2021-02-03:
The notes explain the technical side of the project. The project has 16 receivers and they simultaneously record FT8 signals from sixteen bands.

For example, it is possible to connect two different antennas to two RF inputs and configure the same 8 bands for both inputs. This configuration can be used to compare antennas.

In addition, the table at [this link](https://www.sigidwiki.com/wiki/FT8) lists 20 FT8 bands in the 0-50 MHz frequency range. There are therefore more than 11 FT8 bands.

I since Red Pitaya STEMlab 122.88-16

Since the RF inputs of the Red Pitaya STEMlab 122.88-16 board can be used for undersampling reception, I think it can receive FT8 signals in some VHF and UHF bands and it is not limited to HF bands.

closed 2021-02-03 by pavel-demin

dc7ds, 2021-02-03:
There is an ongoing discussion to assign additional 2.7KHz FT8 slots on some bands to address current congestion so that could be a potential future use-case. Meanwhile, you could use the surplus slots to receive FT4 or WSPR on some bands and do manual/external decoding via a csdr chain and a suitable decoder. 

documentation could state:
simultaneously record FT8 signals from up to sixteen independant, configurable receive slots

waterwin, 2021-02-03:
Dear Pavel,
The standard configuration with the delivered config file only listens to 11 bands in parallel, even though the technical setup could listen to 16 in parallel.You might want to add 1 line to the explanation:The write-c2-files.cfg has 11 bands active, you can add more by editing this file to the maximum of 16.
And still formally, there are no more than 11 HF bands, even though you can listen to more parts in the same band or listen to other parts of the frequency spectrum for instance by using higher Nyquist zones as the Charly25 team is experimenting with.
Years ago we had some communication on using both antenna inputs at the same time which I had solved by editing a set of the .sh scripts in combination with the cron file as you might remember, and you addressed that very well in the later releases.
Another question cropped up for me: has the StemLab enough resources to also work with more bands in parallel than the current 8? Or was this only possible due to the SDRlab version having the Z-7020 FPGA? Instead of the Z-7010.

Kind regards,
Erwin
PE3ES / F4VTQ