pavel-demin/red-pitaya-notes #993: SDR receiver compatible with HPSDR

jeffforber, 2021-02-10:
Hello Pavel ....i was wondering if you can help me ?  iam running your SDR receiver compatible with HPSDR-with SPARKSDR it is working very well and  iam getting really good reception of FT8 FT4 & WSPR but there is one thing i would like to do is change the antenna input from in1 to in2 so i can compare different input transformers etc etc  my computer skills are very limited  copy and paste iam ok with   so if you can give me advice it will be much  appreciated ..thank you in advance jeff G8SCU
-->

### Description of the setup:

 * Device:
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2021-02-11:
Here is how it can be done:
 - extract the micro SD card from Red Pitaya
 - insert this micro SD card into a PC
 - find apps/sdr_receiver_hpsdr/start.sh on this micro SD
 - open start.sh in a good text editor like for example [Notepad++](https://notepad-plus-plus.org)
 - on the last line in start.sh, replace the digits '1' with the digits '2' for the bands you want to use IN2
 - save the file

jeffforber, 2021-02-11:
thank you Pavel i will try this later

jeffforber, 2021-02-11:
hello Pavel ...it is all working ok now ...thankyou  for your time 

regards jeff g8scu

closed 2021-02-11 by jeffforber