pavel-demin/red-pitaya-notes #995: Trouble with creating a bootable SD card image for Red Pitaya

fdloh, 2021-02-25:
Hi Pavel,

I am using a Red Pitaya 125-14 SDR kit, and I have been trying to create a bootable SD card image using the directions that you provided in the following links: 
https://www.scivision.dev/red-pitaya-gnuradio-setup
https://pavel-demin.github.io/red-pitaya-notes/alpine

I am able to download the .zip file that you provided (red-pitaya-alpine-3.12-armv7-20200628.zip) and extract it. However, shouldn't the extracted file be a .img file? Instead, it is a directory with files such as boot.bin and subdirectories like "apps" and "cache".

Anyway, I followed your instructions in the first link (https://www.scivision.dev/red-pitaya-gnuradio-setup) and extracted the .zip file to my SD card. However, when I try to boot my Red Pitaya device with this SD card, the device fails to boot, and only the green LED lights up. I have verified that my device works with the .img file provided by Red Pitaya.

Obviously, I am missing something here. Thanks in advance for any clarification you can provide.

pavel-demin, 2021-02-25:
I didn't write the instructions at https://www.scivision.dev/red-pitaya-gnuradio-setup.

> However, shouldn't the extracted file be a .img file?

No.

> Anyway, I followed your instructions in the first link (https://www.scivision.dev/red-pitaya-gnuradio-setup) and extracted the .zip file to my SD card.

This is correct.

> However, when I try to boot my Red Pitaya device with this SD card, the device fails to boot, and only the green LED lights up.

Why do you think that the device fails to boot?

The solid green LED is OK.

It was once requested to disable the blinking LEDs in my applications to reduce the noise. So, it's OK that there is no blinking LEDs.

BTW. It was recently suggested to disable the LEDs on the Ethernet connector:
https://forum.redpitaya.com/viewtopic.php?f=8&t=24793

I'd suggest checking the boot process using USB/serial console: 
https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

It's possible to use [PuTTY](https://www.putty.org/) to work with the console.

fdloh, 2021-02-26:
Pavel,

Thanks for the help. Ok, I understand that my Red Pitaya device is booting fine, it's just that only the green LED on the board lights up. 

I have followed the rest of the directions (at https://www.scivision.dev/red-pitaya-gnuradio-setup) for setting up my laptop and for setting up a project, but it looks like I am still doing something wrong.

The thing is that I created a simple GNURadio flowgraph, using your blocks for the Red Pitaya Source and the Red Pitaya Sink. I kept the default values of the "Address" and "Port" fields for both blocks (192.168.1.100 and 1001 respectively). However, when I attempt to execute the flowgraph in GNURadio (using an ethernet connection to my Red Pitaya board), I get a "Connection refused" error. 

I am able to ping my Red Pitaya board (using the IP address 192.168.1.100), so I think the ethernet connection to the board is working. I have also read the relevant information at https://www.scivision.dev/red-pitaya-quick-start and at https://pavel-demin.github.io/red-pitaya-notes/alpine.

Are there any modifications I need to make to the SD card image (e.g. network settings, etc.)? My understanding is that your SD card image should work out of the box. Sorry about the trouble caused -- I'm new to this stuff. Thanks again for your help.

pavel-demin, 2021-02-26:
What is missing in your description of the problem is how you start the SDR transciever application on the Red Pitaya board. If you don't start it, then it's normal that GNU Radio can't connect it.

The easiest way is to start it via the web interface. Simply direct your web browser to the IP address of the Red Pitaya board, then click on the SDR transceiver application link.

pavel-demin, 2021-02-26:
BTW. The correct link to the GNU Radio instructions is
https://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/#getting-started-with-gnu-radio

The instructions on the scivision site seem quite outdated.

fdloh, 2021-02-26:
Hi Pavel,

Thanks for the helpful and quick response. I neglected to mention that I also tried to start a web interface to my Red Pitaya board though an ethernet cable, but I had trouble connecting to my board via my web browser (Firefox) in Ubuntu Linux. I tried this with both DHCP and the default static IP address 192.168.1.100. So anyway, it looks like the SDR transceiver application wasn't even started at all. I think it is possible that the network settings of my Ubuntu Linux OS might not be configured properly, but I am unsure of how to rectify this.

According to the link you provided, I can copy the start.sh shell script from the apps/sdr_transceiver directory to have the SDR transceiver application start automatically at boot time of my Red Pitaya board. Is this method preferable to using the web interface method? Thanks.

pavel-demin, 2021-02-26:
> Is this method preferable to using the web interface method? 

It depends on your preferences. Both methods are available and work reliably.

> I also tried to start a web interface to my Red Pitaya board though an ethernet cable, but I had trouble connecting to my board via my web browser (Firefox) in Ubuntu Linux.

If you can't access the web interface, there may be indeed a boot problem.  I'd still suggest checking the boot process using USB/serial console.

fdloh, 2021-03-01:
Hi Pavel,

I have verified the boot process with Minicom and GNU screen, and have confirmed that my Red Pitaya board is properly booting your SD card image using the USB/serial console. I was able to log in to the root account using your directions at https://pavel-demin.github.io/red-pitaya-notes/alpine. 

I have also fixed my network config settings on Ubuntu Linux, and I am now able to access my Red Pitaya board through the web interface and start the SDR transceiver application. So it looks like things are now working properly. Thanks a lot for your assistance.

closed 2021-03-01 by fdloh

dl5rmh, 2021-03-02:
Hi there, I was following this issue since I was neither able to boot the 20200628 release successfully.
On a 2 GB FAT32 formatted SD card, I can follow the boot process via the serial line.
It ends up in following error message:

`Alpine Init 3.4.5-r3
 * Loading boot drivers: ok.
 * Mounting boot media: random: fast init done
ok.
 * Loading user settings from /media/mmcblk0/red-pitaya.apkovl.tar.gz: ok.
 * Installing packages to root filesystem: WARNING: Ignoring APKINDEX.2c4ac24e.tar.gz: No such file or directory
WARNING: Ignoring APKINDEX.40a3604f.tar.gz: No such file or directory
OK: 0 MiB in 0 packages
ok.
grep: /sysroot/etc/inittab: No such file or directory
/sbin/init not found in new root. Launching emergency recovery shell
Type exit to continue boot.
sh: can't access tty; job control turned off
/ #
`
In the file system that is avaiable after this error message, the system is really basic and no network is configured.

I tried to directly extract the ZIP file to the SD card and also tried to extract it on the hard disk and copy it, but it makes no difference.

Btw, the 20191120 runs successfully.
Any ideas?

pavel-demin, 2021-03-02:
Looks like the system can't find some files.

Sometimes strange errors like this occur when there is a problem with the micro SD card.

pavel-demin, 2021-03-02:
I've just checked `red-pitaya-alpine-3.12-armv7-20200628.zip` and I see that it contains `APKINDEX.2c4ac24e.tar.gz` in the `cache` directory. If the system can't find `APKINDEX.2c4ac24e.tar.gz`, then something is indeed wrong with the micro SD card.

I think the easiest test would be to check if the file is on the micro SD card and if it can be opened.