pavel-demin/red-pitaya-notes #996: ADC Recorder readings doesn't match

jesusabilopez, 2021-03-03:
### Description of the setup:

 * Device: STEMLAB 125-14
 * SD card image: 0.96-86
 * Application: ADC Recorder

### Description of the problem:
Hi 

I've been working with the adc recorder and adc recorder trigger projects, sadly, the values displayed on console that I get back are not matching the source signal.

I'm connecting IN1 and IN2 to a DC 5V source to be sure, but I'm getting values that look like this:

 1299  1322
 1298  1320
 1297  1319
 1297  1319
 1296  1318
 1294  1317
 1295  1316
 1294  1315
 1294  1316
 1293  1316
 1294  1317
 1295  1317

I've already set the gain jumpers to HV, so Idk how to solve this 

I want to read other signals as well, and desing a Matlab interface to run either adc recorder or adc recorder trigger and display de recorded signals, but none of them seems to match on the console.

am I missing something?

### Steps to reproduce the problem:
 1. Set IN1 and IN2 gain jumpers to HV
 2. Connect a 5 V DC source to both input channels
 3. Run ADC recorder project on Red Pitaya Board
 4. Run ADC C code
 
Thanks in advane

My best regards

pavel-demin, 2021-03-03:
There is nothing wrong with the values returned by ADC being different for different inputs. The differences are most likely due to the mismatch of the parameters of the electronic components between the input connectors and the ADC.

If you want to perform precision measurements, you will need to calibrate your system to know exactly what voltage corresponds to what value returned by ADC.

closed 2021-03-03 by pavel-demin

jesusabilopez, 2021-03-03:
Is there any source or text where I can find how to do that?

pavel-demin, 2021-03-03:
Try searching for "adc calibration" with your favorite web search engine.

jesusabilopez, 2021-04-09:
Hi

I've been busy, so I couldn't get back and answer earlier

I have set up a project to check the raw output from the ADC IP

[image]

Using a GPIO and the led bar to check the bit output

Also I ran into this post on the redpitaya forums

https://forum.redpitaya.com/viewtopic.php?t=314

which says that the conversion for LV settings and HV settings are different from one another

As I understand you need to calculate the Volts per bit, so, How do i need to convert the samples obtained on the adc IP for the project to give back something that is more like the input voltage? according to the volts per bit

My Best Regards

Jesus Lopez

pavel-demin, 2021-04-10:
I'd still recommend taking a look at some of the many articles on ADC calibration. Here are some links:

https://www.ti.com/europe/downloads/f2810_12_calibration_10.pdf

https://training.ti.com/ti-precision-labs-adcs-understanding-and-calibrating-offset-and-gain-adc-systems

The most basic calibration is done by measuring two known reference values using each of the ADC channels and calculating a calibration gain and offset. Then the measured calibration gain and offset are used to convert the ADC samples to units of the reference values.

jesusabilopez, 2021-04-10:
Hi

I have used the Calibration app on Red Pitaya, and got these values:

[image]

I'm gonna check the articles that you linked to see if I'm missing something else, because using the conversión for the LV settings values are getting pretty close to the input actual value, 

By multiplying the value readed using the monitor command by 1/8192, but for the HV settings is still not working

Alos, I would like to know, Is the output format for the ADC IP on 2's complement? Could my readings be wrong because I'm not noticing that?

My best Regards

Jesus Lopez

pavel-demin, 2021-04-11:
> By multiplying the value readed using the monitor command by 1/8192, but for the HV settings is still not working

If this level of precision is OK for your application, then simply multiply the result by 20 when the jumpers are in the HV position.

ADC range is ±8191, LV range is ±1 V, HV range is ±20 V. So, 8191 corresponds very approximately to 1 V when the jumpers are in the LV position and corresponds to 20 V when the jumpers are in the HV position. 

> Is the output format for the ADC IP on 2's complement?

Yes.

pavel-demin, 2021-04-11:
I've just reread the description of the problem in your first comment.

So your goal is to use the adc_recorder application and to convert the values returned by the adc_recorder application to Volts. Is it correct?

If the previous is correct, then please try to stay on the main topic. Changing the subject to other FPGA setups or using other tools like Red Pitaya calibration doesn't help. 

You already obtained the values corresponding to 5 V returned by the adc_recorder application.

Using the values from your first comment, it's already possible to approximately calculate the gains:
```
average1 = (1299 + 1298 + 1297 + 1297 + 1296 + 1294 + 1295 + 1294 + 1294 + 1293 + 1294 + 1295) / 12 = 1295.5
gain1 = 5 / 1295.5

average2 = (1322 + 1320 + 1319 + 1319 + 1318 + 1317 + 1316 + 1315 + 1316 + 1316 + 1317 + 1317) / 12 = 1317.7
gain2 = 5 / 1317.7
```

To obtain more precise gains and offsets, the average values corresponding to 15 V returned by the adc_recorder application should be obtained. Then the gains and offsets can be calculated as explained in the articles on ADC calibration.