pavel-demin/red-pitaya-notes #998: multi band wspr transceiver

jeffforber, 2021-03-11:
### Description of the setup:

 * Device: rp 14 125
 * SD card image:RED-PITAYA\red-pitaya-alpine-3.12-armv7-20200628
 * Application:
 * Other relevant informa

### Steps to reproduce the problem:

 1.
 2.
 3.

jeffforber, 2021-03-11:
using multi band wspr t/r i configure tx message "TA4/G8SCU KM56 10" its running good lots of spots with only 10mw ..but WSPRnet is reporting power as 0.001W changing the power on tx message doesnt make any difference ie   33dbm still reports as 0.001W  please advise regards jeff TA4/G8SCU

pavel-demin, 2021-03-11:
Looks like a problem with the format of your message. A description of three types of WSPR messages can be found at [this link](https://dxplorer.net/wspr/msgtypes.html).

Since you are using a compound callsign (with a slash), the correct format is "TA4/G8SCU 10".

pavel-demin, 2021-03-11:
BTW. I only tested the transmission of the type 1 messages (without slash). I didn't know about other formats.

closed 2021-03-11 by pavel-demin