pavel-demin/red-pitaya-notes #999: Passthrough Mode for SDR Transceiver

fdloh, 2021-03-25:
Hi Pavel,

I'm trying to get a better understanding of how GNURadio interfaces with your transceiver. I'm using a RedPitaya 125-14 SDR kit with your latest SD card image (red-pitaya-alpine-3.12-armv7-20200628.zip) and the "sdr_transceiver" app.

Suppose I wish to use your transceiver in a simple passthrough mode, that is, I want the signal (at some particular frequency) that is being received through the IN1 port to be outputted through the OUT1 port with no changes to the input signal. In GNURadio, to achieve this, should I directly connect a "Red Pitaya Source" block to a "Red Pitaya Sink" block in a flow graph (setting the "Center Frequency" value to the desired freq)? 

Thanks in advance for your advice.

pavel-demin, 2021-03-25:
> In GNURadio, to achieve this, should I directly connect a "Red Pitaya Source" block to a "Red Pitaya Sink" block in a flow graph (setting the "Center Frequency" value to the desired freq)?

Yes, that's exactly how I would do it.

fdloh, 2021-03-26:
Hi Pavel,

Thanks for the reply. 

Just another question: With regards to your latest SD card image (red-pitaya-alpine-3.12-armv7-20200628.zip), the RedPitaya is configured to run at the maximum clock frequency (I'm referring to the ARM Cortex A9 processors), is that correct? Thanks.

pavel-demin, 2021-03-26:
The CPU clock frequency is set to 666.6 MHz. It's the documented maximum for the speed grade -1 Zynq-7000 SoC.

closed 2021-03-26 by pavel-demin