pavel-demin/red-pitaya-notes #1000: Questions about modifying the SDR transceiver

andrewemark, 2021-04-14:
Hello, thank you for the wonderful notes and applications for the Red Pitaya. 

I would like to modify the SDR transceiver application to synchronize the oscillators for upmixing and downmixing across channels. For this particular application, it would be important to have the phases/frequencies locked across channels for purposes of continuous phase measurements. 

My knowledge of FPGA development is superficial. However, I took a detailed look at the server c code. Would a simple approach be to have one memory-mapped register that modifies frequency of all the DDS blocks? Is this a trivial change to make? If so, guidance would be greatly appreciated! 

I set up a development environment and can compile the existing bit file without issue.

A side comment/question: I cannot easily discern the purpose of the `tx_sync` and `rx_sync` registers in the server code.

Thank you in advance.

pavel-demin, 2021-04-14:
I don't think the SDR transceiver application is a good starting point for building a synchronous configuration. Channel synchronization has never been a requirement for this project and adding it won't be trivial.  For example, the SDR transceiver application uses separate network connections for all channels. These separate network connections aren't trivial to synchronize.

I'd start with the wideband SDR transceiver application and add additional RX and TX signal paths.

> For this particular application, it would be important to have the phases/frequencies locked across channels for purposes of continuous phase measurements.

What kind of signal you're planning to transmit and what are other requirements?

Maybe you could use the VNA application. It has all the required synchronizations and performs phase measurements. 

> Would a simple approach be to have one memory-mapped register that modifies frequency of all the DDS blocks?

It won't be enough.

> I cannot easily discern the purpose of the tx_sync and rx_sync registers in the server code.

The  purpose of the `tx_sync` and `rx_sync` registers is to set the phase to 0 when the frequency is 0.

andrewemark, 2021-04-14:
Thanks for the quick reply. 

The baseband "signal" is a DC value. The goal is to transmit on a single channel continuously and modify the transmit frequency to keep the system impedance matched. Two input channels would be used to measure impedance.

pavel-demin, 2021-04-14:
I'd say the VNA application has almost everything needed to meet the requirements of your project: 
- variable TX amplitude
- variable TX and RX frequency
- synchronized TX and RX oscillators

You'll probably need to remove the FIFO and interpolator connected to the DDS modules and replace them with a configuration register. This configuration register will be used to set the frequency.

Since impedance measurements don't need to be fast, it's even possible to perform all calculations on the on-board CPU and the system could be self-contained.

Am I missing something?

closed 2021-04-19 by pavel-demin