pavel-demin/red-pitaya-notes #1001: Question about wideband receiver on redpitaya 122.88

maxsdr, 2021-04-19:
Hello Pavel, thank you for your notes of Red Pitaya !

I have a question of wideband receiver application include in your SD Alpine card Image. 
There is no page explaining this project, I would like use HDSDR or another software but when i launch aplication with HDSDR , i think that  I/Q data rate isn't to 500 kSPS. 

Can you explain this project and/or if is possible to take technicals specifications 

Thank you in advance :)

pavel-demin, 2021-04-19:
I was once asked (#921) to make an application with two wideband receivers with a bandwidth of 6.144 MHz each. The result is the wideband receiver application. As far as I know, it works well for the person who requested this application.

I'm using this application only to test how fast I can transfer data between the Red Pitaya board and a computer.  All the code that I have is in [projects/sdr_receiver_wide_122_88](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_receiver_wide_122_88).

If you want to use this application with HDSDR, you'll need an ExtIO plug-in compatible with this application.

closed 2021-04-19 by pavel-demin

maxsdr, 2021-04-21:
Hi pavel, Thanks a lot for your response ! 
i'm interest to an application who use 2 Rx of my redpitaya 122.88 with 1.024Mhz bandwidth each int tcp or udp and change the central frequency dynamically by tcp command , my knowledge of FPGA development is most superficial, is this possible by simply modifying the fpga or not ?  If so, guidance would be greatly appreciated!

Thank a lot in advance

pavel-demin, 2021-04-21:
> i'm interest to an application who use 2 Rx of my redpitaya 122.88 with 1.024Mhz bandwidth each int tcp or udp and change the central frequency dynamically by tcp command

The wideband receiver application can do this. The only missing part is the client.

I think that the ExtIO plugin for the SDR receiver can be easily modified to work with the wideband receiver. Here is a link to the source code:
https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_receiver/ExtIO_RedPitaya

The data type on line 121 should be set to 3 (16-bit integer). The format of commands should be also changed in SetHWLO and SetRate.

pavel-demin, 2021-04-21:
What client program that supports two receivers are you planning to use?

maxsdr, 2021-04-21:
for SDR program, i'm developing one that will handle the two redpitaya receivers via tcp commands.

The wideband receiver application, actually  make with two wideband receivers but use one rf input that's right ? 

for the wideban receiver (part fpga), I looked at the project under vivado but I only see one rx , so if you want to use 2 rx, I should modify the wideband project, that's right ? 

I will look at the ExtIO plugin for the commands 

Thank a lof for your response,

pavel-demin, 2021-04-21:
I'm afraid that the answer is "no" to all the questions.

The wideband receiver application has two wideband receivers. Each receiver is connected to a separate input.

I don't think that the FPGA configuration should be modified.

Here is a link to the client program to look for the commands:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_receiver_wide_122_88/client/test-signal-continuity.c

If you develop and SDR program, then just copy code from test-signal-continuity.c to your program. All the commands are on lines 48-50:
```
  command[0] = 8; // set decimation rate
  command[1] = (1 << 28) + 10000000; // set frequency 1
  command[2] = (2 << 28) + 10000000; // set frequency 2
```

maxsdr, 2021-04-22:
Hi pavel, 

thank's a lot, 

indeed, I had not looked well in the code of the fpga... my knowledge of FPGA development is most superficial, sorry !

i thave questions again : 
 1-  I looked your client application , thank a lot for last comment but i'm not sur of my calcul to set my decimation rate : in your code , when rate decimation = 8 you have 6.144Mhz ,  so if I want 1.024Mhz , i must set rate decimation to 48  that's right ?
                 my calcu l: 8*6.144Mhz = 49.152Mhz sampling rate , is 80% of bandwith to redpitaya 122.88 ?? 
                 so 49.152/1.024 = 48 rate deciamtion

 2-  about your server application, you take I/Q data to ram and insert with function neoncopy in buf
               2.1- how to differentiate I/Q of rx1 to I/Q rx2, my idea is to send I/Q data of rx in différent socket : IQ rx1 to sock1 on port X and  IQ rx1 to sock2 on port Y
              2.2- is possible to explain your stucture to send you data in packet ?

Thank a lof for your response,

pavel-demin, 2021-04-22:
Here is how the sample rate is calculated:
```
sample rate = 122.88e6 / rate / 2
```
where `rate` is the decimation rate of the CIC filter, `2` is the decimation rate of the FIR filter.

To obtain 1024 MSPS, `rate` should be set to 60.

The I and Q samples of the two channels are stored in the form of the following C struct:
```
struct samples
{
  int16_t I1;
  int16_t Q1;
  int16_t I2;
  int16_t Q2;
}
```
In case of a flat array of int16_t, element 0 is I1, element 1 is Q1, element 2 is I2, element 3 is Q2, element 4 is I1, etc.

maxsdr, 2021-04-22:
thank you very much for this explanation and this quick answer  !

I understand better now , i go to my code now  :)

pavel-demin, 2021-04-24:
I've just added a version of the SDR transceiver application compatible with the STEMlab 122.88-16 board: 

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-122-88/

Maybe it would match your project requirements better than the wideband receiver application.

maxsdr, 2021-04-26:
thank for this new informations, 

I will look at your application !