pavel-demin/red-pitaya-notes #1002: Question: Can we reuse CIC filter for I & Q in same channel

howard0su, 2021-04-22:
### Description of the setup:

 * Device: STEMLab 125-14
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:
I noticed that we are using two CIC Filters for I and Q channels in all the implementations. I am wondering if we can connect the 245.76M clock to a CIC Filter and reuse one filter for two channels. I am wondering why we didn't do this and if there is other constraints and consideration. This will saves some LUTs and enable the board based on 7010 for additional channels. (I am working on replace 125M osc with a 122.88 Osc on STEMLAB to get accurate 96Khz samples and also remove unneed 25/24 FIR filter.)

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2021-04-22:
Yes, it's possible to connect twice faster clock to CIC filter and other modules.

However, there could be two issues:
- clock domain crossing
- some modules cannot work at 250 MHz

howard0su, 2021-04-22:
I think CIC Compiler and combiner should be fine for 250M clock. cross clock domain is too hard problem for me.

pavel-demin, 2021-04-22:
> I think CIC Compiler and combiner should be fine for 250M clock. 

Running only CIC filters at 250 MHz won't save a lot of resources. All modules that follow CIC filters must also operate at 250 MHz.
 
> cross clock domain is too hard problem for me.

You just add an asynchronous FIFO buffer (AXI4-Stream Clock Converter or AXI4-Stream Data FIFO) between ADC/DAC and all other modules.

howard0su, 2021-04-22:
I am looking at sdr_transceiver_emb, after CIC, there is a combiner in rx. and in tx, we can skip the pair of bcast and combiner. Let me try.

pavel-demin, 2021-04-22:
250 MHz is mentioned as the maximum frequency for the IP cores of the AXI4-Stream infrastructure on page 17 in [PG085](https://www.xilinx.com/support/documentation/ip_documentation/axis_infrastructure_ip_suite/v1_1/pg085-axi4stream-infrastructure.pdf).

I cannot find any information on the maximum frequency of the CIC and FIR filters provided by Xilinx.

The last time I tried to configure one of the SDR applications in this repository to run at 250 MHz, I had timing problems.

If you have the time, it would be interesting to try. Maybe a starting point could be a very simple FPGA configuration with just a few modules like for example [filter_test](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/filter_test) or [filter_test_122_88](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/filter_test_122_88). If this very simple FPGA configuration can run at 250 MHz, then we can continue this discussion.

howard0su, 2021-04-22:
You are right. I cannot make filter_test working under 250M and compiler cannot fit timing requirements.

pavel-demin, 2021-04-22:
So, the next step is to develop a signal processing chain that would run at 250 MHz.

pavel-demin, 2021-04-22:
The answer to the question in the title of this issue appears to be "no". Closing this issue.

closed 2021-04-22 by pavel-demin