pavel-demin/red-pitaya-notes #1005: axi_bram_reader.v and axi_bram_writer.v core question

TestingAccount4, 2021-05-24:
* Device: Stemlab 125-14
 * SD card image:
 * Application:
 * Other relevant information:

Hi Pavel,

Thank you for making your code publicly available, it has been very helpful. I am working on a project where I need to store a look up table consisting of 32 bit elements (the total file is approx 85 kb in size), and depending on a special set of decision rules, access a specific element of the LUT. I also want to keep track of how many times a given element was accessed. Given some reading I've found that the on chip block memory would be ideal for this, so I've read into how it is accessed and interfaced with. I worked on writing and verifying my own cores with AXI protocols (using the Create and Package New IP feature in Vivado) but got bogged down with verifying its performance with the Xilinx AXI verification IP. Upon looking for a simpler solution, I found your axi_bram_reader and axi_bram_writer cores. 

I have some questions about the functionality of both cores. From what I understand, if the reset signal is not active, there are a set of variables (all denoted by *_reg) who's values are assigned the corresponding *_next values. The change in the *_reg values causes the assign statements for the *_wire values to execute, and for the always@* block to execute. The always@* block updates the *_next registers for the next cycle, so that the cycle repeats. The assignments within the always@* block and the if statements configure the *_next variables such that the next read or write operation is triggered once the previous one has been completed. 

I'm getting confused on how a read or write is considered completed. I've tried to follow your logic by following the code, and have tried simulating but hadn't gained much from it. Any help is very much appreciated. Thanks

pavel-demin, 2021-05-25:
I am not sure I understand your question.

If the question is about the AXI part of the code, then it's based on the information from the following two links:

https://zipcpu.com/blog/2019/01/12/demoaxilite.html

https://github.com/ZipCPU/wb2axip/pull/6#issuecomment-500034278

TestingAccount4, 2021-05-25:
Thank you for those references. In short I am trying to understand how your axi_bram_reader.v and axi_bram_writer.v cores work so I can use them in one of my projects. I've read through the different types of the axi protocol and can follow it on a signal diagram, but have found it very difficult to know how to appropriately write/interface with the master side so I can drive the two cores correctly. 

In my project I plan to connect the axi_bram_writer.v core to the AXI interconnect that's connected to the PS so I can write to the BRAM from the PS, and I plan to connect the axi_bram_reader.v core to a custom IP that will provide the address of the data to be read out. This read data will then be used as the signal to the DAC. I have been having trouble with the BRAM parts of this project. To figure out which how to interact with the master signals to successfully drive your cores, I was trying to trace a cycle of read/write.  Sorry if this is a basic question, this is my first time having my code directly interact with an AXI interface.

pavel-demin, 2021-05-25:
> I plan to connect the axi_bram_reader.v core to a custom IP that will provide the address of the data to be read out. This read data will then be used as the signal to the DAC.

The correct module for this job is axi**s**_bram_reader not axi_bram_reader. The axis_bram_reader can be directly connected to the axis_red_pitaya_dac module.

An example of interconnecting axi_bram_writer and axis_bram_reader modules can be found in the iir_test project. 

> I have been having trouble with the BRAM parts of this project.

Normally, the BRAM interface is very simple. All you need to do is provide an address, then the data is available at the next clock cycle.

closed 2021-05-25 by pavel-demin

TestingAccount4, 2021-06-15:
Hi again Pavel,

I did some more work to try and simplify/pinpoint where I'm getting stuck with using your axi_bram cores- as an exercise I created a bitstream that is meant to just write values to BRAM from a C program, and read them back (again using a C program). To do this I used your axi_bram_reader and axi_bram_writer cores, and connected them directly to seperate instances of the "block memory generator" IP. The axi_bram_reader and axi_bram_writer cores are configured as follows:
AXI ADDR WIDTH: 16
AXI DATA WIDTH: 32
BRAM ADDR WIDTH: 13
BRAM DATA WIDTH: 32

and both individual instances of the block memory generator IP were set to stand alone single port RAM with a write width (& read width) of 32, and a write depth (& read depth) of 8192. I also set "enable port type" to "always enabled", and the rest of the parameters are their default values. 

After generating the bitstream, I modified the code from the following example (https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842146/Zynq-7000+AP+SoC+-+Using+BRAM+for+Additional+On-Chip+Memory+Tech+Tip). I also studied your iirc project as you mentioned to try and get a better sense of how the C code works:

    #include <stdio.h>
    #include <stdlib.h>
    #include <unistd.h>
    #include <fcntl.h>
    #include <sys/mman.h>
    #include <stdint.h>

  
    int main()
    {
       //unsigned int bram_size = 0x40000;
       //int bram_pbase = 0x43C10000; // physical base address
       //int bram_pbase_read = 0x43C00000; // physical base address
       volatile uint32_t *bram64_vptr;
       volatile uint32_t *bram64_vptr_read;
       int fd;

  
       // Map the BRAM physical address into user space getting a virtual address for it
       if ((fd = open("/dev/mem", O_RDWR | O_SYNC)) != -1) {
  
          bram64_vptr = (uint32_t *)mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0x43C10000);
  
          bram64_vptr_read = (uint32_t *)mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0x43C00000);

          // Write to the memory that was mapped, use devmem from the command line of Linux to verify it worked
          // it could be read back here also

	  int k=0;
	  for(k=0; k<10; k++) {  
              bram64_vptr[k] = 9;
          }

          int i=0;
          for(i=0;i<10;i++) {
      	      //bram64_vptr[i]=9999999;
	      printf("%d\n",bram64_vptr_read[i]);
	  }
          close(fd);
       }
     }
     
     
     
 However my print statements always print zeros. The C code compiles and runs without any issue, and is similar to what I had seen in your other projects that used the same cores. I believe i'm using your cores incorrectly, but haven't been able to isolate where the problem is. Any help would be greatly appreciated.

pavel-demin, 2021-06-15:
> connected them directly to seperate instances of the "block memory generator" IP

If you are writing to one instance of BRAM and reading from another instance of BRAM, then it is perfectly normal for the program to print zeros.

If you want to read the same values that you write to BRAM, reader and writer must be connected to the same BRAM. It is exactly how it is done in the iir_test project.

TestingAccount4, 2021-06-15:
Thank you! I changed the block memory generator instance to true dual port BRAM (still in stand alone mode) and connected the read and write cores to it- now I am able to read and write to BRAM using the C program I had above. 

I did find that the elements written to index i with the write array are accessed via index i+1 in the read array. For example, if I had the line bram64_vptr[0]=17, and I later accessed bram64_vptr_read[0], it would not be 17, but bram64_vptr_read[1] would return 17. The same pattern holds for higher indices. From other runs that I've done, the pattern still holds for higher values of i as far as I can tell, but bram64_vptr_read[0] takes on the value that would be read next if the for loop continued for another cycle (e.g. if in the C code listed above the loop read for(i=0;i<5;i++), when bram64_vptr_read[0] was printed it would return the value written at bram64_vptr[5]). Is this due to some memory offset between ports A and B of the block memory generator IP? Have you experienced this before/is this to be expected? I couldn't find documentation on this. As long as the trend of element i that is written being available at element i+1 of the read array continues, I'm not concerned about using it in my code, but I'm trying to understand it. Apologizes for being verbose

pavel-demin, 2021-06-16:
Try to disabled 'Primitives Output Register' for the BRAM port connected to axi_bram_reader.