pavel-demin/red-pitaya-notes #1006: Alpine image no booting

SiggiSorglos, 2021-05-26:
* Device: RP 125-14
 * SD card image: red-pitaya-alpine-3.12-armv7-20210427.zip
 * Application: OS
 * Other relevant information: image red-pitaya-alpine-3.9-armv7-20191120.zip is booting

No boot, green LED on, no other LEDs on.

 1. Download red-pitaya-alpine-3.12-armv7-20210427.zip from https://github.com/pavel-demin/red-pitaya-notes/releases
 2. unzip file to an sdcard (i.e. 32GB)
 3. put sdcard in RP, power up

pavel-demin, 2021-05-26:
Why do you think that the device fails to boot?

Green LED on is OK.

No other LEDs on is also OK.

Looking at your description of the problem, I would say everything is OK. 

Have you tried to connect to the SSH or web interface of the board? 

I'd suggest checking the boot process using USB/serial console:
https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

It's possible to use [PuTTY](https://www.putty.org/) to work with the console.

SiggiSorglos, 2021-05-27:
sorry, my assumption that there have to be more LED indications than the green LED was wrong.
The RP is booting fine.

closed 2021-05-27 by SiggiSorglos