pavel-demin/red-pitaya-notes #1007: HPSDR server not keying ALEX T/R in CW mode

ka6s, 2021-06-06:
### Description of the setup:

 * Device:  SDR-16
 * SD card image:  about a year old
 * Application:sdr_transceiver_hpsdr_122_88/
 * Other relevant information:  I have a real ALEX sub-system.  I've modified the server code to send I2C messages to my own conversion hardware that re-originates the data as SPI traffic to the ALEX.  So I've spent a fair amount of time getting things aligned so I know that the data over I2C is correct and controls the ALEX hardware nicely.   I'm also running PiHPSDR as my GUI.   This problem was found when trying out the system when sending CW.
 *

### Description of the problem:
The T/R relay is inside the ALEX hardware, and the keyer code never updates the T/R relay to follow cw_ptt.

### Steps to reproduce the problem:

 1. Send morse code using PiHPSDR with the settings to have the Radio manage CW (which means use the built in keyer.)
 2. Send morse code using the bulit-in keyer.

Pavel - I've added a few lines of code - and it works.  I haven't figured out how to add the server code I've modified to you through this interface.  i can send it to you as an attachment via email.  The only changes are those required to get the keyer info to Alex.

pavel-demin, 2021-06-07:
Thank you for testing the ALEX interface and for finding how to fix this problem.

No problem with sending the code by e-mail. My e-mail address is in [my profile](https://github.com/pavel-demin).

pavel-demin, 2021-06-08:
I think that [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/8882f3d29da5641b27dca3bf5196f1ee9e944535) fixes this problem.

ka6s, 2021-06-09:
Pavel - I haven't been able to confirm this yet - and likely won't get a
chance till the weekend. I DO want to thank you for your quick attention!!

Thanks again,

Steve KA6S

closed 2021-06-22 by pavel-demin