pavel-demin/red-pitaya-notes #1008: Query - Best way to 'tune' A/D levels?

darksidelemm, 2021-06-20:
Hi! Thanks for your amazing work in putting these projects together!

We're planning on setting up a FT8 skimmer at a remote receive site (where http://kiwisdr.areg.org.au:8073/ is running), using a Red Pitaya and your software.

At this site we have encountered times where the KiwiSDRs up at site (also with 14-bit ADCs) overload due to the overall HF large-signal environment. On the KiwiSDR I can at least see indications of overload.

What would be the best way to check for overload on the Red Pitaya? Ideally I'd be after a dB Full-Scale level, to give me some idea of how close to overload I'm getting, but even just a basic overload indication would be sufficient. Then I can judge if I need to add in attenuation.

Thanks,
Mark VK5QI

StefanDausP, 2021-06-20:
+1 from my side. I'm operator of DL0PF https://www.darc.de/der-club/distrikte/u/ortsverbaende/11/technik/high-performance-ft8-ft4-und-cw-skimmer-an-clubstation-dl0pf/ and I would be interested in setting up a adaptive preamp setup based on possible adc overload indicators.

BR & 73
Stefan DC7DS

Gesendet SD mobil

pavel-demin, 2021-06-20:
> Ideally I'd be after a dB Full-Scale level, to give me some idea of how close to overload I'm getting, but even just a basic overload indication would be sufficient.

Please provide an algorithm of how you would like to calculate these indicators.

darksidelemm, 2021-06-20:
I'm not sure of how you'd implement this easiest in a FPGA, but what i'd be after would be a record of the highest observed A/D level over some historical period (say, 5 seconds). This can be converted to a dBFS level by dividing it by the highest possible A/D reading (unsure what this would be - possibly 2**13-1 ?) and taking the 20log10 of it. Supplying the raw peak A/D level would be more than sufficient.

Unsure the best way to provide this to the user, but this has useful applications for all HF direct-sampling receiver applications, since gain management is one if the biggest issues with this type of SDR, especially without preselectors.

pavel-demin, 2021-06-20:
> I'm not sure of how you'd implement this easiest in a FPGA, but what i'd be after would be a record of the highest observed A/D level over some historical period (say, 5 seconds).

Thanks for the details of the algorithm.

I think that it can be done with the following (untested) module:

```
`timescale 1 ns / 1 ps

module axis_maxabs_finder #
(
  parameter integer AXIS_TDATA_WIDTH = 16,
  parameter integer CNTR_WIDTH = 32
)
(
  // System signals
  input  wire                        aclk,
  input  wire                        aresetn,

  input  wire [CNTR_WIDTH-1:0]       cfg_data,

  // Slave side
  output wire                        s_axis_tready,
  input  wire [AXIS_TDATA_WIDTH-1:0] s_axis_tdata,
  input  wire                        s_axis_tvalid,

  // Master side
  input  wire                        m_axis_tready,
  output wire [AXIS_TDATA_WIDTH-1:0] m_axis_tdata,
  output wire                        m_axis_tvalid
);

  reg [AXIS_TDATA_WIDTH-1:0] int_max_reg, int_max_next;
  reg [AXIS_TDATA_WIDTH-1:0] int_tdata_reg, int_tdata_next;
  reg [CNTR_WIDTH-1:0] int_cntr_reg, int_cntr_next;
  reg int_tvalid_reg, int_tvalid_next;

  wire int_comp_wire;

  always @(posedge aclk)
  begin
    if(~aresetn)
    begin
      int_max_reg <= {(AXIS_TDATA_WIDTH){1'b0}};
      int_tdata_reg <= {(AXIS_TDATA_WIDTH){1'b0}};
      int_cntr_reg <= {(CNTR_WIDTH){1'b0}};
      int_tvalid_reg <= 1'b0;
    end
    else
    begin
      int_max_reg <= int_max_next;
      int_tdata_reg <= int_tdata_next;
      int_cntr_reg <= int_cntr_next;
      int_tvalid_reg <= int_tvalid_next;
    end
  end

  assign int_comp_wire = int_cntr_reg < cfg_data;
  assign int_abs_wire = s_axis_tdata[AXIS_TDATA_WIDTH-1] ? ~s_axis_tdata : s_axis_tdata;

  always @*
  begin
    int_max_next = int_max_reg;
    int_tdata_next = int_tdata_reg;
    int_cntr_next = int_cntr_reg;
    int_tvalid_next = int_tvalid_reg;

    if(s_axis_tvalid & int_comp_wire)
    begin
      int_max_next = int_abs_wire > int_max_reg ? int_abs_wire : int_max_reg;
      int_cntr_next = int_cntr_reg + 1'b1;
    end

    if(s_axis_tvalid & ~int_comp_wire)
    begin
      int_max_next = {(AXIS_TDATA_WIDTH){1'b0}};
      int_tdata_next = int_max_reg;
      int_cntr_next = {(CNTR_WIDTH){1'b0}};
      int_tvalid_next = 1'b1;
    end

    if(m_axis_tready & int_tvalid_reg)
    begin
      int_tvalid_next = 1'b0;
    end
  end

  assign s_axis_tready = 1'b1;
  assign m_axis_tdata = int_tdata_reg;
  assign m_axis_tvalid = int_tvalid_reg;

endmodule
```

The output of this module can be written to a FIFO buffer in the same way as for the PPS counter.

pavel-demin, 2021-06-20:
> This has useful applications for all HF direct-sampling receiver applications, since gain management is one if the biggest issues with this type of SDR, especially without preselectors.

I don't know if this is really an issue. I have seen articles and presentations expressing the opposite opinion.

I don't have a lot of experience with variable gain devices. However, the few that I tried brought more problems (noise and distortions) than they solved. So if possible I would prefer to add a filter that would reduce the signals that overload the ADC.

When I used the FT8 and WSPR applications, I just checked the signal levels with an oscilloscope. So I'd say that currently this is the only/best way to do it. 

If KiwiSDR can detect overloads and is connected to the same antenna as Red Pitaya, perhaps the easiest solution would be to use KiwiSDR to control the gain.

darksidelemm, 2021-06-20:
Unfortunately the KiwiSDR's overload reporting functionality requires that you connect in and take up a channel, which is not ideal for long-term monitoring. Also, it only reports the presence of overloads, and doesn't report the A/D level which is far more useful. Also, the input circuitry is different, with different transformer ratios before the ADC, so it's not an ideal comparison - The transformer ratio is also something that I need to select for the Red Pitaya.

In my situation I don't need to add more gain into the system, I actually need to add slight amounts of attenuation at certain times in the evening. I regularly see the KiwiSDRs overload after evening grey-line when we see large signal levels from lots of shortwave stations in Asia propagate in. I've 'solved' the problem temporarily by putting a few dB of fixed pad in front of the KiwiSDRs, but this means i'm now limited by the KiwiSDRs A/D performance on the higher bands (15m and up).

And yes... any front-end that goes in front of the ADC needs to be carefully selected to avoid causing other performance issues. Instead of variable gain devices a better option is multiple fixed gain devices (norton amp designs work well for this), with a switching network to switch amps and attenuators in and out to achieve the required system noise figure & dynamic range.

In your code example, how often does that max value reset? It looks to be every second?

pavel-demin, 2021-06-20:
> In your code example, how often does that max value reset? It looks to be every second?

It is configurable via cfg_data. The counter increments every 8 ns and counts from zero to cfg_data. If the counter width is 32 bits, then the maximum time is approximately 30 seconds.

darksidelemm, 2021-06-23:
Just as a thought (after trying out a few other

darksidelemm, 2021-06-23:
(Apologies, pressed the wrong button!)

An extension to this idea would be to display the top few bits of the ADC on the array of LEDs down the side of the Red Pitaya. This would give a quick indication of how close the ADC is to overload without needing to SSH in to the Red Pitaya.
It would certainly give an immediate indication if the receiver was overloading...

pavel-demin, 2021-06-23:
I like the idea of using LEDs. Maybe a simple bar graph display showing full scale dB level would be a nice feature for FT8 and WSPR applications.

darksidelemm, 2021-06-23:
Yep! Very useful during setup. 
Though being able to access it remotely would still be useful... (I'l be putting my Red Pitaya at a remote site - no LED access!)

pavel-demin, 2021-06-23:
I think it's a little C program that should read maxabs values, calculate dBs, and control LEDs. It should be easy to change this C program and adapt it to other use cases. As for example adding an I2C interface to control external relays switching the attenuators.

Basically, the idea is to keep the code in this repository easy for me to test without needing any custom hardware. I can test the on-board LEDs but I can't test your attenuators.

darksidelemm, 2021-06-23:
Yep, what you suggest would be perfect - others can build out from that - for most of us it's the FPGA bits that are the hard part. Once the data is in some easily accessible form a lot more is possible...

pavel-demin, 2021-06-23:
I think the next step should be to move the code related to frequency calibration to a separate sub-project (common_tools) and add the level measurement code to it. This sub-project will be reused by at least four projects (FT8 and WSPR for two types of boards).

pavel-demin, 2021-06-24:
Here is a link to a pre-release version of the SD card image with the new level indicator:
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABe3Tv0ycs6pocu9SRjzWnba/red-pitaya-alpine-3.14-armv7-20210624.zip?dl=1

Commands to test the new level indicator:
- start the FT8 transceiver:
```
~/apps/sdr_transceiver_ft8/start.sh
```
- start the level indicator:
```
~/apps/common_tools/measure-level 125 1
```

darksidelemm, 2021-06-24:
Hmm, the ft8 cron jobs don't seem to be running. I moved the start.sh script to the top level of the SD card, and can see that the ft8 script has been added to /etc/periodic, but I'm not seeing anything starting up (no log files, nothing in 'ps x' or 'top').
This is on a RP 125-14.

pavel-demin, 2021-06-24:
Thank you for finding the problem with the FT8 transceiver. I haven't checked if the FT8 transceiver works with this SD card. There have been a lot of changes recently and I haven't had time to test all the applications yet. The main purpose of this SD card image is to test the `measure-level` program.

darksidelemm, 2021-06-24:
Sure - should I expect the measure-level code to work with no other transceiver software running though? 
If not, what would I need to do to start up something so that measure-level produces output?

pavel-demin, 2021-06-24:
The `measure-level` program only requires the FPGA to be correctly configured (done by the `sdr_transceiver_ft8/start.sh` script).

I confirm that cron jobs do not run with this SD card image. It is based on the latest version of Alpine Linux. Looks like something has changed in the way the cron service works.

darksidelemm, 2021-06-24:
Hmm interesting.. I've run start (which presumably has configured the FPGA - I also tried running the line cat sdr_transceiver_ft8.bit > /dev/xdevcfg ), but see no output when running `~/apps/common_tools/measure-level 125 1`

pavel-demin, 2021-06-24:
> I've also noted that the same 4 LEDs stay on no matter what I do at the RF input

Looks like a hardware problem with one of the inputs.

darksidelemm, 2021-06-26:
I can confirm the new image works, with signals being decoded. I'm still unsure as to the behaviour of the 4 LEDs. Are these indications of the MSBs of the ADC or something else? Hooked up to an antenna (via a 9:1 balun, input attenuation on RP bypassed), I can see the LED states fluctuating, but it's not clear what it means.

Just having a dBFS level printed out to the terminal would be valuable...

pavel-demin, 2021-06-26:
Here is a link to the source code of the measure-level program:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/common_tools/app/measure-level.c

The LEDs are configured as a simple bar graph display showing full scale dB level. There are two groups of four LEDs corresponding to two RF inputs.

darksidelemm, 2021-06-26:
Yes, I've had a look at the source code, and couldn't quite figure out what the different LED scales meant. I think it works out to roughly 6 dBFS steps?

Can we also please have the peak A/D level emitted to the terminal? This would allow reading off the level without having to look at the LEDs. I'm about to install a red pitaya into a site 50km away from my home QTH, and pointing a webcam at the LEDs probably isn't the best solution to monitoring these levels...

At this point I'm not setup to build for the Red Pitaya, and there's no build tools included in the image, else I would have just modified the measure-level.c code myself.

pavel-demin, 2021-06-26:
> there's no build tools included in the image

They are actually included. The packages are on the SD card but not installed by default. They can be installed with the following command:
```
apk add make gcc
```
If you prefer to keep them installed by default, then write the changes to the SD card with the following command:
```
lbu commit -d 
```
The commands to compile measure-level.c are
```
cd apps/common_tools
rw
make
```

darksidelemm, 2021-06-26:
Excellent, thanks for that info!

I just added in the following line:
```
printf("RX1: %.2f dBFS,  RX2: %.2f dBFS\r\n",20.0*log10(1.0*level[0]/scale),20.0*log10(1.0*level[1]/scale));
```

Which results in output like:
```
rp-f00eb9:/media/mmcblk0p1/apps/common_tools# ./measure-level 125 1
RX1: inf dBFS,  RX2: inf dBFS
RX1: -7.13 dBFS,  RX2: -37.52 dBFS
RX1: -7.25 dBFS,  RX2: -37.60 dBFS
RX1: -7.06 dBFS,  RX2: -37.68 dBFS
RX1: -7.17 dBFS,  RX2: -37.60 dBFS
```

Testing with a sig-gen, full-scale appears to be about -6 dBm with my current front-end setup.

pavel-demin, 2021-06-27:
Thank you for the tests.

The level measurement code is now in the [new release](https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20210627).

closed 2021-06-27 by pavel-demin