pavel-demin/red-pitaya-notes #1009: Antivirus (Avast) Reporting malware

PrimLem, 2021-06-20:
Description of the setup:

    Device: Red Pitaya STEMlab125-14, Windows 7 PC, both connected over Ethernet.
    SD card image: red-pitaya-alpine-3.12-armv7-20210427.zip
    Application: vna-win32-20210427.zip
    Other relevant information: Windows 7, local IPs 192.168.100.x, IP mask 255:255:255:0, Red Pitaya MAC 00:26:32:F0:0F:73

Description of the problem:

I am using the most recent VNA SD card and also the control program. The same program used to work OK on the same PC 2 weeks ago but now the antivirus is reporting that the "vna.exe" contains malware (Mde Class) and puts it into the Quarantine. Before that the OS complains that Windows don't have appropriate permissions although I am an administrator on this PC. This happens instantly and 100% repeatable.  I wouldn't like to bypass the antivirus program because I am really not sure what is going on. The Red Pitaya is working OK in the VNA mode, I can access it through the HTTP port and load the " Red Pitaya Notes" page with 11 bullets from "Led Blinker" to "Vector Network Analyzer".

Steps to reproduce the problem:

1. Unzip the freshly downloaded VNA control program into a selected folder
2. Run vna.exe.

[image]
[image]

Thanks,
Primož S53KS

pavel-demin, 2021-06-20:
I've just tested vna.exe using www.virscan.org. It also shows that avast finds something in this files. However, most of other engines don't find any problem.

You can always manually install Python, PyQt5 or PySide2 and matplotlib and then run vna.py using your Python installation.

closed 2021-06-22 by pavel-demin

PrimLem, 2021-06-23:
Thanks for the investigation from your side and the proposal for manual install. On another computer with Windows 10 and different antivirus it works without problems.